Moddable-OpenSource/moddable #520: Windows 10 Getting started compile error.

nickm324, 2020-12-17:
> Note: This template is intended to help organize your report. Feel free to modify it as you need. The most important thing is to file a clear, detailed report, not to follow the template.

> You may use this template to file a report of a security vulnerability. If you prefer to report the issue privately, please send the report by email to info@moddable.com.

**Build environment:** Windows
**Target device:** NodeMCU ESP8266

**Description**
Error 0xc0000135 when trying to execute the getting started hellowworld code sample

**Steps to Reproduce**
1. Build and install the app using this build command: mcconfig -d -m -p esp or  mcconfig -d -m -p nodemcu 
2. Get the following error: 
# cc xsAll.o (strings in flash + (not) force-l32)
NMAKE : fatal error U1077: 'C:\Users\rippe\esp\xtensa-lx106-elf\bin\xtensa-lx106-elf-gcc.EXE' : return code '0xc0000135'
Stop.

**Expected behavior**
app to compile and run.

**Other information**
Followed all the instruction in the getting started documentation, reviewed and checked several times and still cant get past this error.

andycarle, 2020-12-17:
Hi @nickm324,

I'm sorry to hear that you're having trouble getting started with ESP8266 builds. Let's see if we can sort out the issue here.

I've heard one other report of this same error in a Windows ESP8266 build. On that user's system, we found that the problem was that the [Cygwin toolchain support package](https://www.moddable.com/private/cygwin.win32.zip) bin directory was not included in the `PATH` environment variable.

To check and see if that is the problem, could you please:

1. Verify that you unzipped and copied the [Cygwin toolchain support package](https://www.moddable.com/private/cygwin.win32.zip) to `C:\Users\<your-user-name\esp\cygwin` (Step 5 of the [Windows ESP8266 Installation instructions](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp8266.md#win-instructions).)
2. Verify that you updated your `PATH` to include to folder `C:\Users\<your-user-name\esp\cygwin\bin`. Referring to it as `%BASE_DIR%\esp\cygwin\bin` is also fine, as long as you have set `BASE_DIR` to `C:\Users\<your-user-name>`.  (Step 14 of the [Windows ESP8266 Installation instructions](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp8266.md#win-instructions).)
3. Open a new "x86 Native Tools Command Prompt for VS 2019" instance to pick up the changes to your environment variables.
4. Verify that the Cygwin DLLs are in your `PATH` using the command `where cygwin1.dll`. If your `PATH` is properly configured, you should see something like this:
```console
C:\Users\andyc\Projects\moddable>where cygwin1.dll
C:\Users\andyc\esp\cygwin\bin\cygwin1.dll
```

Please give those steps a shot and let me know how it goes.

&nbsp;- Andy

nickm324, 2020-12-17:
Cygwin package was unzipped here.
[image]

PATH has Cygwin as the first entry in the path.

[image]

When I run "where cygwin1.dll" in the VS Native tools command window it cannot find it. I have rebooted after I added all the path changes and user variables.

andycarle, 2020-12-17:
Try either making `BASE_DIR` a System variable or adding that Cygwin folder to your User `PATH`. I suspect it may be a problem to have a System variable with a dependency on a User variable.

nickm324, 2020-12-17:
Ok, I will give that a try but that came directly from the Getting Started instructions, so that is what I added.

andycarle, 2020-12-17:
Understood. If that ends up being the issue, I'll definitely tweak the Getting Started guide to clarify. :)

nickm324, 2020-12-17:
That did resolve the issue, I can now use WHERE to find the dll, but now a new issue. When I ran the command "mcconfig -d -m -p esp" I get the following error.

[image]

nickm324, 2020-12-17:
I am using a nodeMCU is that correct platform (mcconfig -d -m -p esp)
 to be using to compile?

andycarle, 2020-12-17:
I'm taking a look at the abi.o issue.

`mcconfig -d -m -p esp` should be fine for `helloworld`. The blink example is at `%MODDABLE%\examples\pins\blink`. And when you want to try the nodemcu target, that looks like: `mcconfig -d -m -p esp/nodemcu`.

andycarle, 2020-12-17:
abi.o should be being built by this rule:

```
{$(CORE_DIR)\}.cpp{$(LIB_DIR)\}.o:
	@echo # cpp $(@F)
	$(CPP) $(C_DEFINES) $(C_INCLUDES) $(CPP_INCLUDES) $(CPP_FLAGS) $? -o $@
	$(AR) $(AR_OPTIONS) $(LIB_ARCHIVE) $@
```

Can you verify that abi.cpp is in `%BASE_DIR%\esp\esp8266-2.3.0\cores\esp8266`:

```console
C:\Users\andyc>cd %BASE_DIR%\esp\esp8266-2.3.0\cores\esp8266

C:\Users\andyc\esp\esp8266-2.3.0\cores\esp8266>dir abi.*
 Volume in drive C is Local Disk
 Volume Serial Number is AA17-54D9

 Directory of C:\Users\andyc\esp\esp8266-2.3.0\cores\esp8266

10/16/2020  11:42 AM             2,315 abi.cpp
               1 File(s)          2,315 bytes
               0 Dir(s)  61,131,460,608 bytes free
```

nickm324, 2020-12-17:
> mcconfig -d -m -p esp/nodemcu

Yea that is what I had been using all day and just kept getting a different error, something about make and invalid platform.
Just tried blink and got the same abi.o error.

C:\Users\rippe\Projects\moddable\examples\pins\blink>mcconfig -d -m -p esp/nodemcu
NMAKE : fatal error U1073: don't know how to make 'C:\Users\rippe\Projects\moddable\build\tmp\esp\debug\lib\abi.o'
Stop.

Just tried the to check on the abi.o issue and that directory is not found.

nickm324, 2020-12-17:
Not sure what's going on these variables have been setup all day and after closing my VS2019 native command prompt and reopening now it can find the folder.

C:\Users\rippe>cd %BASE_DIR%\esp\esp8266-2.3.0\cores\esp8266

C:\Users\rippe\esp\esp8266-2.3.0\cores\esp8266>dir abi.*
 Volume in drive C is BOOT
 Volume Serial Number is 620D-2AE7

 Directory of C:\Users\rippe\esp\esp8266-2.3.0\cores\esp8266

06/21/2016  02:06 AM             2,310 abi.cpp
               1 File(s)          2,310 bytes
               0 Dir(s)  20,331,466,752 bytes free

C:\Users\rippe\esp\esp8266-2.3.0\cores\esp8266>

andycarle, 2020-12-17:
Curious, to be sure.

Now that you have that variable working, maybe try cleaning and building again? You can delete `%MODDABLE%\build\bin\esp` and `%MODDABLE%\build\tmp\esp` to make sure you clean everything related to the ESP8266 build.

nickm324, 2020-12-17:
Ok I just compiled and ran Blink and it loaded on my nodemcu and is working!
Thanks for your help, much appreciated! I am really excited to going through the book and working the examples.

andycarle, 2020-12-17:
Excellent! I'm glad we got that sorted out!

And I've already pushed a clarification to the System vs User environment variables issue to our local repository. That will go to the GitHub repo in our next open source push. Thank you for working through the issues with me.

closed 2020-12-17 by andycarle

nickm324, 2020-12-18:
Where would we get support if we were having issues with the examples in the book?
Just tried to run the very first one from the book, Chapter 1 Getting Started and have already run into errors.

C:\Users\rippe\Projects\iot-product-dev-book\ch1-gettingstarted\helloworld>mcrun -d -m -p esp/nodemcu
C:\Users\rippe\Projects\moddable\build\tmp\esp\debug\helloworld\makefile(71) : fatal error U1036: syntax error : too many names to left of '='
Stop.

I have deleted all the folders under the build\tmp and it regenerates the esp folder when I run the above command, but it goes straight to this error.

andycarle, 2020-12-18:
We've taken care of this over on Gitter, but for the historical record:

1) The issue you saw with the mcrun makefile fragment is real and was a very recently introduced bug in the Moddable SDK. I've pushed a fix to our private repository that will make it to GitHub in the next push.

2) The potential issue you mentioned with the UPLOAD_PORT shouldn't be a problem if you have that environment variable set. The makefile will check to see if it is set and will only use "com8" if it has not been set in the environment.