Moddable-OpenSource/moddable #521: Document `staticSize` machine allocation value

chances, 2020-12-21:
https://github.com/Moddable-OpenSource/moddable/blob/40162325723fa17b1b64158aee645e46ef16ab2d/xs/includes/xs.h#L1130

See #147

phoddie, 2020-12-21:
Short answer: for code running on computers (which seems to be your interest?), the `staticSize` field is unused. 

Here's a longer answer that will be added to the XS in C documentation:

> - Some XS hosts attempt to grow the slot and chunk heaps without limit at runtime to accommodate the memory needs of the hosted scripts; others limit the maximum memory that may be allocated to the machine. For the latter, the `staticSize` defines the total number of bytes that may be allocated for the combination of chunks and slots, which includes the stack. In general, only hosts running on resource constrained devices implement `staticSize`.

chances, 2020-12-23:
@phoddie Thank you for updating the XS in C documentation.

Is this value not what `creation.static` is used for? If so, please also consider documenting that [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/manifest.md?rgh-link-date=2019-03-11T16%3A06%3A56Z#creation), too.

phoddie, 2020-12-23:
> Is this value not what creation.static is used for?

They are the same.

> If so, please also consider documenting that here, too

Sure.

phoddie, 2020-12-30:
Closing as all requests have been addressed.

closed 2020-12-30 by phoddie