Moddable-OpenSource/moddable #523: Non-standard properties found and removed

erights, 2020-12-22:
The following non-standard properties were found on the XS primordials. The SES-shim initialization package run on XS correctly does not find them on the SES whitelist and so deletes them. This issue isn't exactly a bug report since an implementation may provide extra properties that can be deleted, and we do successfully delete all of these.

Why are they present in the first place? Should we expect everything else to continue to work fine after they are deleted? Are any of these sufficiently useful and general that they should be proposed for eventual standardization?

```
ArrayBuffer.fromBigInt
ArrayBuffer.fromString
BigInt.bitLength
BigInt.fromArrayBuffer
String.fromArrayBuffer
Math.idiv
Math.idivmod
Math.imod
Math.imuldiv
Math.irem
Math.mod
Array.prototype.reverse.@@  // some symbol
Array.prototype.sort.@@  // some symbol
Array.prototype.@@isConcatSpreadable
ArrayBuffer.prototype.concat
DataView.prototype.getUint8Clamped
DataView.prototype.setUint8Clamped
Object.prototype.propertyIsScriptable
Object.prototype.@@toPrimitive
String.prototype.compare
String.prototype.toLocaleString
```

phoddie, 2020-12-22:
@erights -- interesting report, thank you. I'll address some of these immediately. 

> ArrayBuffer.fromString
> String.fromArrayBuffer

These are light versions of what browsers today provide with `TextEncoder` and `TextDecoder`. They use UTF-8. While we find them essential (converting between buffer and string in script is needlessly expensive), I doubt the committee would consider these.

> ArrayBuffer.fromBigInt

This is similar to the string/array buffer methods above.

> BigInt.bitLength

This returns the number of significant bits in a `BigInt` (ignoring leading zeroes). We modeled the API on discussions about this functionality in `BigInt` that  was deferred from the standard. We think it should be in the standard.

> Math.idiv
> Math.idivmod
> Math.imod
> Math.imuldiv
> Math.irem
> Math.mod

These are an initial implementation of our integer math proposal, currently at Stage 1. ;)

> ArrayBuffer.prototype.concat

This can be implemented in user code, but is more efficient in the native implementation and that matters to us. I don't know the committee would find it relevant given that ArrayBuffers are obscure on the web to begin with.

> Array.prototype.reverse.@@  // some symbol
> Array.prototype.sort.@@  // some symbol

I don't see these in my tests. How are you finding these?

> DataView.prototype.getUint8Clamped
> DataView.prototype.setUint8Clamped

These appear to be a mistake and should be removed.

> Object.prototype.propertyIsScriptable

This is the final vestige of a JavaScript sandbox we used some time ago, a kind of predecessor to SES. It should be removed.

> Array.prototype.@@isConcatSpreadable
> Object.prototype.@@toPrimitive
> String.prototype.compare
> String.prototype.toLocaleString

These may be removable, but a require a little more time, and likely a run of test262 for a definitive answer.

erights, 2020-12-23:
```js
// const f = Array.prototype.reverse;
const f = Array.prototype.sort;
const names = Object.getOwnPropertySymbols(f).map(String);
console.log(JSON.stringify(names));
```
prints `[]` on Node as expected, but prints `["Symbol()"]` on XS, detecting the extra symbol mentioned above.
This also happens if `f` is `Array.prototype.reverse`. I'm reporting these here rather than a separate issue because it is not technically a bug. But it is unexpected, apparently by you as well ;).

erights, 2020-12-23:
See https://github.com/Agoric/SES-shim/pull/549

phoddie, 2020-12-23:
@erights - FWIW, when I run the lines above in our helloworld example, it prints `[]` as expected for both sort and reverse. What host are you using for your test? My guess is that you are testing with xsnap. Recall that it contains an example of how to use metering on built-ins. The examples uses reverse and sort. ;) See `fxPatchBuiltIns`.

erights, 2020-12-23:
I am indeed testing with xsnap. Thanks.

phoddie, 2020-12-24:
Today's push removes the following:

- Array.prototype.@@isConcatSpreadable
- ArrayBuffer.prototype.concat
- DataView.prototype.getUint8Clamped
- DataView.prototype.setUint8Clamped
- Object.prototype.propertyIsScriptable
- Object.prototype.@@toPrimitive
- String.prototype.toLocaleString

> Note: `String.prototype.compare` remains as a stub to help ease the migration (our code used it in places). We'll remove that entirely at some point. Your shim can safely delete it.

The other APIs remain as they are useful for the purposes described above.

You'll need to do a full, clean rebuild of both XS tools and your projects after updating because this is a minor version update to XS.

erights, 2020-12-25:
Are you sure you removed `ArrayBuffer.prototype.concat` ?

On the others you list, verified.

phoddie, 2020-12-25:
My mistake. We use Array.prototype.concat so it remains. It can be safely deleted, however.

erights, 2020-12-25:
All the ones in my original list that remain can be safely deleted?

erights, 2020-12-25:
Under the assumption that all the remainder on my original list can be safely deleted, closing. Thanks!

closed 2020-12-25 by erights

phoddie, 2020-12-25:
Yes, that is correct. (If you find otherwise, just leet us know.)

erights, 2024-02-14:
More such properties seen recently by @gibson042 

```
Removing unpermitted intrinsics
Removing intrinsics.%InitialMath%.irandom
Removing intrinsics.%SharedMath%.irandom
Removing intrinsics.%AsyncIteratorPrototype%.@@asyncDispose
Removing intrinsics.%IteratorPrototype%.@@dispose
```