Moddable-OpenSource/moddable #524: Object.prototype[Symbol.toPrimitive] needed in string coercion.

erights, 2020-12-23:
Related to #523 

```js
delete Object.prototype[Symbol.toPrimitive];
const o = {
  toString: () => "ok"
};
try {
  console.log(String(o));
} catch (e) {
  console.log('err', e.message);
}
```

Running this program on XS produces `err (host): no function`. But if the first `delete` line is commented out it produces `ok` as it should. The deleted property is non-standard and the behavior of the `String` coercion is specified to work when it does not find such a property. When the program above is run on Node it does produce `ok` as it should.

phoddie, 2020-12-23:
Yup. We know. ;) Looking into it...

erights, 2020-12-23:
See https://github.com/Agoric/SES-shim/pull/549

phoddie, 2020-12-24:
This should now be resolved.

erights, 2020-12-25:
Verified. Closing.

closed 2020-12-25 by erights