Moddable-OpenSource/moddable #527: TypeScript build : Error "Cannot find module 'piu/MC'"

stc1988, 2020-12-28:
**Build environment:** Windows
**Target device:** Windows Simulator
**Version** December 16, 2020

**Description**
Error:"Cannot find module 'piu/MC'" occured when Piu application build with TypeScript.

**Steps to Reproduce**
1. add `$(MODDABLE)/examples/manifest_typings.json` to `manifest.json`
2. build `mcconfig -d -m -p win`
3. I got below error.

> main.ts:1:22 - error TS2792: Cannot find module 'piu/MC'. Did you mean to set the 'moduleResolution' option to 'node', or to add aliases to the 'paths' option?
> 
> 1 import * as Piu from "piu/MC"

**other Information**

I couldn't find  path of `piu/MC` in `moddable\build\tmp\win\mc\debug\<myapplication>\modules\tsconfig.json` .

phoddie, 2020-12-28:
There are a few problems here. Getting the files included is straightforward:

- Rename it to MC.d.ts, because  case matters here
- Add the  piu directory to manifest_typings.json
- In MC.d.ts change the export from "piu/mc" to "piu/MC"

Now it is included in the build. But, it doesn't work because it does not declare the Piu global constructors. That would be something like this...

```ts
  global {
    const Skin: typeof Skin
    const Texture: typeof Texture
  }
```

...though that doesn't work because the  name Skin is a conflict. I think we need to ask @meganetaaan for his guidance since he created the Piu typings.

stc1988, 2020-12-29:
Thank you for confirming.

> - Rename it to MC.d.ts, because case matters here
> - Add the piu directory to manifest_typings.json
> - In MC.d.ts change the export from "piu/mc" to "piu/MC"

I do above steps, then I get no build error and Piu application is launched.

> But, it doesn't work because it does not declare the Piu global constructors. 

I can use `Skin` with no problems.　What kind of case  is problem occured?

phoddie, 2020-12-29:
> I can use Skin with no problems.　What kind of case is problem occurred?

Interesting. I tried with `$MODDABLE/examples/piu/balls` by renaming main.js to main.ts. What did you try? ;)

stc1988, 2020-12-29:
I tried with very small application like this

``` javascript
import {Application, Skin, Style, Label} from "piu/MC"

const BlackSkin = Skin.template({fill: "black"})
const OpenSans42 = Style.template({ font: "OpenSans-Light-42", color: "white"})

let label = new Label(null, {
  Style:OpenSans42,
	string: "Test"
})

export default function() {
	return new Application(null,{
    Skin: BlackSkin,
    top: 0, bottom: 0, left: 0, right: 0,
    contents:[label]
  })
}
```
I also tried with `$MODDABLE/examples/piu/balls`, I got  errors like this.

> main.ts:17:28 - error TS2304: Cannot find name 'Skin'.
> 17 const backgroundSkin = new Skin({ fill:"silver" });

It seems that using `import {} from "piu/MC"`  does not resolve declaration.
To resolve this,  I Import declarations individually like this.
`import {Application, Behavior, Skin, Texture, Content} from "piu/MC";`

phoddie, 2020-12-29:
@stc1988 - Right. What you did will work. But, it should be possible to have the classes stored on globals work as well.

We'll make make the change discussed above. That will allow the explicit imports to work. Solving the globals seems like a separate issue.

stc1988, 2020-12-29:
Yes, this issue is reolved by these fixes.

> * Rename it to MC.d.ts, because case matters here
> * Add the piu directory to manifest_typings.json
> * In MC.d.ts change the export from "piu/mc" to "piu/MC"

While, to get existing example applications worked with TypeScript is needs more fixes :(

phoddie, 2020-12-30:
Good news. I found a way to get the globals to work too. It is a little complicated, but... seems to work well. We'll push that today. (Also, it is experimental... I didn't try to support all globals yet... just enough to compile piu/balls with TypeScript.)

phoddie, 2020-12-30:
I'm closing this out as the changes described above are now available.

closed 2020-12-30 by phoddie