Moddable-OpenSource/moddable #530: Linux aborts with "dead strip" when implicit cast of object to string

cmidgley, 2020-12-29:
**Build environment:** Linux (works correctly on Mac and Windows)
**Target device:** Simulator

**Description**
When casting an object (empty or not) implicitly to a string, the program terminates with `Abort: dead strip`.  Using explicit casting (`obj.toString()`) works correctly.

**Steps to Reproduce**
Compile and run the following on Linux:

```
var x = {};
var y = "hi" + x;
```

In xsbug you will see the `abort: dead strip` message.

Other ways to reproduce:

```
var x = {};
trace(`X is ${x}\n`);
```

and

```
var x = {a: 3, b: 4};
trace("X is " + x);
```

Explicit cast works correctly:

```
var x = {};
trace("X is " + x.toString());
```

phoddie, 2020-12-29:
I can reproduce this problem on macOS by replacing the content of the helloworld example with your two lines. There was some work for Christmas that may have indirectly caused this problem. Perhaps you are running different versions of the Moddable SDK on your different operating systems?

cmidgley, 2021-01-03:
Confirmed fixed.  Thank you!

closed 2021-01-03 by cmidgley