Moddable-OpenSource/moddable #531: dead strip error in template strings

menway, 2020-12-30:
https://github.com/Moddable-OpenSource/moddable/blob/9ed551497e76fc79f3e901d5a73fdd8f509233c3/examples/network/ble/scanner/main.js#L25

When I tried the BLE Scanner example, the debugger prompted an error ```XS abort: dead strip``` on this line.

After checking this document I realized ```device.address``` is not a string:

https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/ble/ble.md#properties

With ```.toString()``` added, the error is gone.
```
trace(`${completeName} - ${device.address.toString()}\n`);
```

So I guess because ```device.address``` is an instance of ```Bytes```, template strings don't support this type of data in XS engine , or the implicit type conversion was automatically removed by feature stripping?

menway, 2020-12-30:
I just confirmed this issue was caused by automatic feature stripping.

As mentioned by the blog post below, I added ```"strip": []``` in ```manifest.json``` and the example worked fine without the explicit ```.toString()```.

https://blog.moddable.com/blog/eval/

But I still don't know why automatic feature stripping determined to remove the type conversion functionality. If this is an intended behavior, with automatic feature stripping enabled (by default), template strings may not work properly for non-string variables.

menway, 2020-12-30:
Update: it seems to be the same issue as https://github.com/Moddable-OpenSource/moddable/issues/530

I was using macOS and an ESP32 board.

phoddie, 2020-12-30:
Thank you for the detailed report. We have a fix pending for #530. We'll check to see if it addresses this one as well

phoddie, 2020-12-30:
Yes. this has the same root cause as #530.

closed 2021-01-15 by phoddie