Moddable-OpenSource/moddable #536: Mods document contains a contradiction

johnwargo, 2020-12-31:
The mods document contains a contradiction; in the document, it says:

> The mod host must be a debug build because mcrun uses the xsbug debugger protocol to install the mod. (Note that mods can run on a release build but the installation method used by mcrun depends on capabilities only present in a debug build.)

but both sentences can't be true. The first sentence says the mod host **must** be a debug build, but the second sentence says it **can** be a release build. Which is it?

phoddie, 2020-12-31:
The full paragraph makes it clear that there is no contradiction here:

> For `mcrun` to install the mod, a mod host must already be installed on the device. The mod host must be a debug build because `mcrun` uses the xsbug debugger protocol to install the mod. (Note that mods can run on a release build but the installation method used by `mcrun` depends on capabilities only present in a debug build.)

The second sentence clearly is related to the first. The first sentence is about installing mods using `mcrun`, and the second sentence provides a requirement for hosts to work with `mcrun`. A host's ability to **run** a mod is distinct from the ability of a host to **install** a mod using `mcrun`, which is what the parenthetical note addresses.

We'll update the text to try to minimize the potential for confusion for future readers.

johnwargo, 2021-01-01:
@phoddie I'm a new learner reading the documentation for the first time and I noticed that the materials were unclear.  I clearly see the contradiction, or at least the lack of clarity on the topic, so I shared the information with you. Thank you very much for immediately discounting my feedback.

I don't understand what you mean when you say "A host's ability to run a mod is distinct from the ability of a host to install a mod using mcrun, which is what the parenthetical note addresses."  I've ready your response 4 times now and I have absolutely no idea what you mean.  Again, I'm looking at this with new eyes - you and your team are so close to the product and have such history with it that I think you're losing the perspective of new users. 

Are you saying you install the host as a debug build, install the mod, then, I don't know, reinstall the host as a release build?

phoddie, 2021-01-02:
> Thank you very much for immediately discounting my feedback.

Your feedback was not discounted. I provided a prompt clarification in this report and made edits to the document to try to address the feedback.

> A host's ability to run a mod is distinct from the ability of a host to install a mod using mcrun

`mcrun` is just one way to install a mod. A mod host may provide others, as @cmidgley explained on Gitter. It is very possible to have a mod host which does not support installing a mod using `mcrun` but does support installation in other ways. Such a mod host is still be able to run mods but does not support installing them using `mcrun`.

paseaf, 2021-01-05:
If I understand correctly, when using MODs, Host and MOD need to be **built**/**compiled** (to XS bytecode) and flashed to your microcontroller **separately** .
- `mcconfig` is used to build and flash firmware.
- `mcrun` is used to build and flash a MOD.

A MOD must run on a host, and a host must be able to load a MOD.
The problem is, currently, only debug build of firmware/host contains the components/features to load MODs, and release build does not.

But a MOD itself can be either a debug or a release build as long as the firmware supports MODs.

Maybe trying on some examples would help:
https://github.com/Moddable-OpenSource/iot-product-dev-book/tree/master/ch1-gettingstarted
`host/` contains settings to build a host. `helloworld/` and `helloworld-gui` are two different MODs (application code).

This will work:
``` bash
# 1. Build and flash the host
cd  ch1-gettingstarted/host
# debug build host
mcconfig -d -m -p esp32

# 2. Build and flash the MOD
cd ../helloworld
# debug build MOD
mcrun -d -m -p esp32
# release build MOD (smaller in size than debug build)
mcrun -i -m -p esp32
```

This won't work:
``` diff
cd  ch1-gettingstarted/host
# release build host
- mcconfig -d -m -p esp32
+ mcconfig -i -m -p esp32

cd ../helloworld
# debug build MOD
mcrun -d -m -p esp32
# release build MOD (smaller in size than debug build)
mcrun -i -m -p esp32
```
Because a release build host doesn't contain the components/features to support/load MODs

cmidgley, 2021-01-05:
The above is correct in that you must have a debug version of the host running in order for the host to be running a protocol (serial or TCP depending on the platform) which `mcrun` can communicate with in order to send the mod.  `mcrun` is basically the same as `mcconfig` except that it doesn't compile a host, it compiles only JS into bytecode, and then attempts to upload it to the host using the debugger protocol.  This is why you must have a debug build on the host in order to send mods.

This does not mean, however, that you cannot run mods on release builds.  It only means you can use the simplicity of the debugger protocol and mcrun to do so.  The ability to load and run a mod is specific to how the platform and startup code is set up.  For example, with the ESP32, there is a flash partition called `xs`.  If you place mod code into that partition, the default esp32 host will find it and run it.  On the simulators, you can use the simulator UI to load a mod (you don't have to use mcrun).  

You can also write your own startup `main.c` for the host, and do anything you want when start the XS virtual machine (where is where the mods are specified and loaded).  I have a pull request that adds a command line (cli) simulator for win, mac and linux that offers the ability to load a mod based on a command line parameter when starting the host - and this does not require any debug code to be enabled.

Hope this helps!

paseaf, 2021-01-05:
> The above is correct in that you must have a debug version of the host running in order for the host to be running a protocol (serial or TCP depending on the platform) which `mcrun` can communicate [...]

Amazing!!! I just tried to flash a MOD to a release version of the host with another tool `esptool.py` and the MOD was able to run perfectly!!
Thanks a lot!

**Steps I did:**
1. build and flash
    ```
    # 1. Build and flash the host
    cd  ch1-gettingstarted/host
    # build the host in release mode and flash
    mcconfig -i -m -p esp32
    
    # 2. Build the mod/application code
    cd ../helloworld
    # release build MOD (debug version should also work)
    # -o . is used to specify the target directory for built binary
    mcrun -i -m -p esp32 -o . 
    
    # 3. flash the built binary file to xs partition
    esptool.py write_flash 0x3A0000 bin/esp32/release/hello_world/hello_world.xsa
    ```
    (`0x3A0000` is the starting address of `xs` partition, assuming Moddable's [default partition table](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/xsProj/partitions.csv) is used)

2. Verify the result:
I used `minicom` to monitor the UART output:
    ``` bash
    minicom -b 115200
    
    # to quit: Ctrl + A, then X
    ```
    Press reboot on the MCU to see the output.
    ```
    ...
    I (263) cpu_start: Pro cpu start user code                
    I (55) esp_core_dump_common: Init core dump to UART       
    E (55) esp_core_dump_common: No core dump partition found!
    I (56) cpu_start: Starting scheduler!on PRO CPU.          
    I (0) cpu_start: Starting scheduler on APP CPU.                                        
    instruments key: Pixels drawn,Frames drawn,Network bytes t
    Hello, World
    instruments: 0,0,0,0,0,0,0,0,0,0,256264,1,0,760,1536,57920
    ...
    ```

phoddie, 2021-01-05:
@paseaf - welcome, and thank you for your great comments!

Yes, for ESP32 you can install a mod by writing directly to the XS partition using `esptool.py` as @cmidgley described That is one way to install  a mod to a release build. Note that doing the same is possible for ESP8266, but more difficult because the address mods are stored is host dependent (to maximize space, as described in the Mods document)

You are also correct that a release build of mod is smaller. That is because it does not include the debugging opcodes and file paths required by the debugger. As a result, a release build of a mod also runs a *little* faster.

phoddie, 2021-01-15:
This report appears to be resolved. Closing.

closed 2021-01-15 by phoddie