Moddable-OpenSource/moddable #537: Simulator-based Worker is not working with debugger or trace

cmidgley, 2020-12-31:
**Build environment:** Windows (perhaps other simulators as well)
**Target device:** simulator (`-p win` is all I have tested, @phoddie says it may be all simulators)

**Description**
When using a Worker thread, calling `debugger` does not cause a breakpoint, and calling `trace` does not output trace messages to xsbug.

**Steps to Reproduce**
For debugger: Using examples\base\worker (using mcconfig -m -d -p win), add a call to debugger and execute:

```
let counter = 0

self.onmessage = function(msg) {
    debugger;
    self.postMessage({hello: "from simple worker", counter: ++counter});
}
```

For trace: Same approach, but try calling trace:

```
let counter = 0

self.onmessage = function(msg) {
    trace("Inside worker\n");
    self.postMessage({hello: "from simple worker", counter: ++counter});
}
```

**Other information**
Accord to @phoddie it works fine on microcontrollers (ESP), but fails on simulator.  I have only tried simulator.

phoddie, 2021-01-02:
I was able to reproduce this. However, it was only because I had been changing file name extensions between .js and .ts, resulting in a build with out of date binaries. Debugging works as expected using $MODDABLE/examples/base/worker on a clean build, both with JavaScript and TypeScript source files.

cmidgley, 2021-01-03:
I continue to have this failure.  I did a `git pull`, removed by `bin` and `tmp` directories, rebuilt everything (`build` on windows), and ran the test above (adding `debugger` and `trace` to the `simpleworker.js` file.  I've only tried on Windows (`mcconfig -d -m`) using the standard Moddable windows simulator, but I do not get an extra tab in xsbug, I do not get into the debugger, and I do not see the trace message.  Just to make sure, I ran this process twice.

cmidgley, 2021-01-04:
I just cleaned up to the original Moddable source tree (and then fixed the bug in xsAtomics with `true` vs. `TRUE` so I could compile) and it works fine.  It's broken in my other branch that I use of Moddable which should just be merging of some simple changes I've made to typings, but obviously something else is wrong!  My bad, sorry to bother you with this.  I'll make sure to test with the original branch next time.

closed 2021-01-04 by cmidgley

cmidgley, 2021-01-04:
It turns out it was a dumb change I made to xsDebug in my `cli-XXX` branch.  The problem is that if you run code that is enabled for debug, yet the debugger is not connected, each `worker` will have a long delay due to a tcp connect timeout. I had incorrectly assumed that the connect would be shared across all instances, but it it is unique per worker VM and so my code broke debugging on workers.  

I've reverted the branch (and therefore the pull request) to the original `public` version of xsDebug, though it would be nice to find a way to avoid these long delays on connect when debug is enabled yet the debugger is not running.

phoddie, 2021-01-04:
Thanks for running this down. The TCP connect timeout can be resolved in the host using a similar approach. It should absolutely not modify XS since that impacts all hosts.