Moddable-OpenSource/moddable #538: Windows io/serial concurrency and parameter issues

cmidgley, 2021-01-03:
**Build environment:** Windows
**Target device:** Windows

**Description**
I have found a concurrency problem in the `io/serial/win/serial.c` implementation.  There are unprotected members in the code that are shared by both the serial thread loop (`xs_serial_loop`) and used by the XS thread (in `xs_serial_read_callback` and `xs_serial_write_callback`), causing communications to randomly fail (in my case, the `onReadable` stops being triggered because `readJob` gets lost due to unprotected concurrency).  I believe this is focused on the `readJob` and `writeJob` members being unprotected when changed across two different threads.  A quick code inspection seems to indicate this may be an issue on Linux as well, though Mac may be ok (not sure).

I also also found that when `onReadable` is triggered, you may not get anything when you call `read` (returns `undefined`) because a prior read may have already read it.  Is this a valid use pattern - that `read` can return more bytes than `onReadable` says and therefore it may also return `undefined`?  

The `count` parameters in `onReadable` and `onWritable` appear meaningless on Windows and Linux (somewhat better on Mac).  For `onWritable` it is always `1` on Windows and Linux (very inefficient) and always `1024` on Mac (unsure if that may cause overflow issues for not).  `onReadable` is always `1` on Windows and Linux (even though more may be available), but code inspection seems to show correct implementation on Mac (I've not tested on anything other than Windows).

cmidgley, 2021-01-03:
I can now confirm that adding critical sections in `xs_serial_loop`, `xs_serial_read_callback` and `xs_serial_write_callback` resolves the problem on Windows.  I did a quick hack, not taking care to minimize the scope of the lock, but basically protected the use of `readJob` and `writeJob`.  My implementation left the critical section in place for the duration of the read/write callbacks - which does also solve the `read` call returning `undefined`, but it starves the serial loop so maybe not the best solution.  But as a hack, it has been running with reliable communications for about two hours now with non-stop transactional (bi-directional) communications (before it only lasted at most a minute before failing on my use case).

See this [gist](https://gist.github.com/cmidgley/c233f7e84467a0f2ebc90d994fca9975) for a copy of the hacked source - just search for "critial" to see where I whacked in the critical sections.

phoddie, 2021-01-05:
Interesting. Completely believable as that module is very fresh. We'll take a look. Thank you!

phoddie, 2021-01-05:
We've pushed a set of changes to address the concurrency issues. They are more minimal, but seem to work. Let us know what you find.

The implementation of the count parameter is a stub on Windows and Linux. This is early code. ;)

cmidgley, 2021-01-06:
Thanks - I have successfully run this code for over 12 hours continuously, with bi-directional transactions and everything seems fine.  This fixes the core issue (failing to work due to concurrency).  The other issue of getting an `onReadable` event with no data also gets addressed by this change, as it is now protected by a critical section.  That final issue, where `count` is always 1, you have confirmed as being "early code" and therefore is not an issue (and also does not affect usage, but does make for reduced performance).  Closing this issue.

closed 2021-01-06 by cmidgley

phoddie, 2021-01-06:
Great that the concurrency issues are resolved. Thank you for confirming that.

FYI - we have changes to get the parameters to `onReadable` and `onWriteable`  to be more correct. They should arrive in the coming days.