Moddable-OpenSource/moddable #539: XS Issue: scope of function declarations

YaoHouyou, 2021-01-08:
**Version**: [9ed5514
](https://github.com/Moddable-OpenSource/moddable/commit/9ed551497e76fc79f3e901d5a73fdd8f509233c3) 

**Description**
The scope of function name `r` of XS is different from other engines.  When running the 5th line of the test case, all engines (such as V8, SpiderMoneky, javascriptCore, chakra) but XS can access `r` normally.  It seems to be an issue of XS.

**Test case**
``` js

1. var NISLFuzzingFunc = function () {
2.     {
3.         function r() {}
4.     }
5.     print(r);
6. };
7. var NISLCallingResult = NISLFuzzingFunc();

```
**Command**:
``` bash
./moddable/build/bin/lin/release/xst testcase.js
```

***Behavior***
```
ReferenceError: NISLFuzzingFunc: get r: undefined variable
```

**Expected behavior**
```
function r (){[native code]}
```

**Build environment:**  Linux

**Build steps**
``` bash
cd $MODDABLE/xs/makefiles/lin
make
```

phoddie, 2021-01-08:
That's subtle. ;) Thank you for the report.

erights, 2021-01-08:
Note that this is a sloppy-mode only issue. It has no effect on strict code. The relevant sloppy spec language is, I believe, in Annex B and so is optional.

phoddie, 2021-01-08:
Aha! Thanks, Mark.

phoddie, 2021-01-09:
Here's a [link](https://tc39.es/ecma262/#sec-block-level-function-declarations-web-legacy-compatibility-semantics) to the relevant in ECMA-262, "Block-Level Function Declarations Web Legacy Compatibility Semantics". Annex B is optional for non-browser engines. XS is not for browsers. ;)

@YaoHouyou, please take a look. This may be a case where there is a valid deviation from the browser behavior.

YaoHouyou, 2021-01-10:
Thank you. I will close this issue.

closed 2021-01-10 by YaoHouyou