Moddable-OpenSource/moddable #540: XS Issue:  Variables declared with let are treated as const

MKGaru, 2021-01-09:
**version:** 4ad0dde1b
**Build environment:** Windows
**Target device:** ESP32

**Description**
Should be possible to declare using const, but the let to be declared next will be treated as const.

**Test case**
```javascript
function func(descriptors) {
  const obj = {}
  
  for (const [key, descriptor] of Object.entries(descriptors)) {
    const type =
      typeof descriptor.type == 'string'
        ? descriptor.type.toLowerCase()
      : typeof descriptor.type.name == 'string'
        ? descriptor.type.name.toLowerCase()
      : ''
    if ((type == 'digital')) {
      if (typeof descriptor.mode == 'string') {
        // It should be possible to declare using const, but the let to be declared next will be treated as const.
        const mode = descriptor.mode.toLowerCase()
        if (mode == 'output') descriptor.mode = 1
      }
      if (descriptor.mode == 1) {
        let value = descriptor.default || false
        //value = !value  // ERROR!! set value: const!
        
        Object.defineProperty(obj, key, {
          get() {
            return value
          },
          set(val) {
            value = val // ERROR!! set value: const!
          },
          enumerable: true,
        })
      }
    }
  }
  
  return obj
}

const obj = func({
  prop: {
    type: 'Digital',
    mode: 'Output',
  }
})

obj.prop = true

```
Sorry for not being able to provide the minimum test case

**Behavior**
`Break: ?.prop: set value: const!`

**Steps to Reproduce**
1. Build and install the app using this build command: `mcconfig -d -m -p esp32`
2. See error

**Expected behavior**
Can update value.

phoddie, 2021-01-09:
That's really strange. I can reproduce it on macOS too. We'll take a look. Thank you for the report.

phoddie, 2021-01-19:
Great find! This was a trivial fix. The problem was a tiny error when initializing the closure that left the state bits mostly uninitialized. This problem has been around for at least a couple years, apparently without causing any trouble.

Thanks again.

MKGaru, 2021-01-20:
Thank you for fixed. I confirmed.

phoddie, 2021-01-20:
Excellent. Thanks for the quick confirmation. Closing.

closed 2021-01-20 by phoddie