Moddable-OpenSource/moddable #541: ESP32: pins/servo/esp32 module wrong parameter.

MKGaru, 2021-01-09:
**Build environment:** Windows
**Target device:** ESP32

**Description**
Servo document wrote:
`min,max: Range of the pulse duration in microseconds.`
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/pins/pins.md#constructordictionary-5

but ESP32 implements:
`[pulse duration in microseconds]  * 1000000 / 32767`
https://github.com/Moddable-OpenSource/moddable/blob/86a98/modules/pins/servo/esp32/modServo.c#L95

andycarle, 2021-01-16:
Thank you for the report! I believe you are correct that there's an issue here.

I think the implementation should actually be `([pulse duration in microseconds] / 20000.0) * 32767`, rather than the current `([pulse duration in microseconds] / 1000000.0) * 32767.0`. (There's a typo in your report above that reverses the `/` and `*`.)

The logic there requires a bit of chasing around the module. But basically, `ms.min` and `ms.max` are the values that will be written to [`ledc_set_duty`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html#_CPPv413ledc_set_duty11ledc_mode_t14ledc_channel_t8uint32_t) as the `duty` value to represent 0 and 180 degrees, respectively. 

The range of `duty` is from 0 - 2^duty_resolution, which in our case is 0 - 32767. To convert from microseconds to a fraction of that maximum duty cycle, we need to know the period of the PWM. That happens to be 20,000 microseconds (50 Hz) with our configuration. 

Put that all together, and you get `minimum_duty = [minimum pulse duration in microseconds] / 20,000 * 32,767`.

I'll commit that fix shortly.

&nbsp;- Andy

MKGaru, 2021-01-20:
Thank you for fix.

I tried SG90 servo.  The servo configure min: 500 max 2400 us pulse.  
My cheap oscilloscope measured correctly 12% duty at 2400 us pulse.

[image]

closed 2021-01-20 by MKGaru