Moddable-OpenSource/moddable #543: Several HTTP Server issues (book, type and potentially implementation)

cmidgley, 2021-01-12:
**Build environment:** Windows
**Target device:** Windows Simulator

**Description**
I think there is a mistake in the book, example 3-21.  The example is a HTTP Server for a PUT request.  It is setting the request type to String by returning `String` from `Server.status` (ID 2) yet the implementation for server requires you return the required type in `Server.headersComplete` (ID 4).   It might also be nice to augment the documentation, under `Class HTTP Server` section, in the `callback` description, for ID 4 (`headersComplete`) to include that it expects a return value of either `String` or `ArrayBuffer`.

Also in the implementation of `http.js` around line 631 the code does not default any return type nor throw an error (perhaps only if there is a payload) nor trace a warning message.  Therefore if you don't return `String` or `ArrayBuffer` from a `Server.headersComplete` message you will not get any request payload to the callback (not as a fragment or as a complete message, instead you just get `undefined`).  If this is desired behavior, perhaps consider updating the documentation to reflect?

Finally, I think the type definition for `HTTPServerCallback` in `http.d.ts` needs to include `object` (or a stronger type definition) in the acceptable return types, as return value from a message of `Server.prepareResponse` is an object containing `headers` and `body`.

phoddie, 2021-01-15:
(Taking a little time to get to this. Apologies. Will take a look soon.)

cmidgley, 2021-01-16:
Peter - no rush at all - sit on this as long as you want.  Not a blocker at all.

phoddie, 2021-01-16:
@cmidgley - the code in the book is indeed incorrect. The [code](https://github.com/Moddable-OpenSource/iot-product-dev-book/blob/master/ch3-network/http-server-put/example.js) in the example repository is correct and hasn't been updated in over a year. I'm not sure how those got out of sync, but that's for noting it. I've updated the [corrigendum](https://github.com/Moddable-OpenSource/iot-product-dev-book/blob/master/corrigendum.md).

> Therefore if you don't return String or ArrayBuffer from a `Server.headersComplete` message you will not get any request payload to the callback

That is intended. If your callback doesn't need the data, it avoids some overhead.

Updating the documentation with the following text. This will be pushed in the coming days.

> All headers received. All HTTP headers have been received. Return `String` or `ArrayBuffer` to receive the complete request body provided to the `requestComplete` message as the corresponding type; return `true` to have `requestFragment` invoked as the fragments arrrive. Return `false` or `undefined` to ignore the request body. The behavior for other return values is undefined

> Finally, I think the type definition for HTTPServerCallback in http.d.ts needs to include object (or a stronger type definition) in the acceptable return types, as return value from a message of Server.prepareResponse is an object containing headers and body.

Will update to the following:

```ts
export type HTTPServerCallback = (this: Request, message: ServerMessages, val1?: any, val2?: any) => void | number | boolean | string | ArrayBuffer |
        {status?: number, reason?: string, headers?: string[], body: true | string | ArrayBuffer};
```

Consistent with the comment by @bmeck ("TODO: better") there should be a way to define this so that the return value is limited by the value of the message parameter, but that is a bigger adventure.

phoddie, 2021-02-10:
Closing as I believe the original issues are resolved.

closed 2021-02-10 by phoddie