Moddable-OpenSource/moddable #544: ESP-IDF-4 not working on DC or mobile battery ( without usb serial) at debug build.

MKGaru, 2021-01-16:
branch: esp-idf-4
version: e01b04ce2

ESP-IDF-4 not working on DC or mobile battery ( without usb serial) at debug build.

Is this behavior correct?

manifest.json
```json
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODULES)/pins/digital/manifest.json"
	],
	"modules": {
		"*": "./main"
	}
}
```

main.js
```js
import Digital from "pins/digital"

Digital.write(PIN_LED, true)
```

andycarle, 2021-01-16:
Thank you for the report and for testing out our ESP-IDF v4.2 preview. 

No, the behavior you're seeing is not expected. It looks like we forgot to pull out a temporary change from the board bring-up process.

I'll get the fix into our next open source push. But in the meantime, it's an easy one to correct locally. 

In [$MODDABLE/xs/platforms/esp/xsPlatform.c](https://github.com/Moddable-OpenSource/moddable/blob/esp-idf-4/xs/platforms/esp/xsPlatform.c#L677) at line 677 you'll see four lines commented out:

```c
//			if (timeout && (timeout < modMilliseconds())) {
//				fxDisconnect(the);
//				break;
//			}
```

Just uncomment those four lines and it should fix the issue.

&nbsp;- Andy

phoddie, 2021-01-18:
FYI - the ability to run a debug build without the debugger attached is provided as a convenience by the ESP8266 and ESP32 hosts. When no debugger is present, the wait for the debugger introduces a delay in start-up.  As a rule, when running without the debugger you should use release build (`mcconfig -m -p...`) or an instrumented build (`mconfig -i -m -p...`).

andycarle, 2021-01-19:
This should be fixed by b9119d2. Sorry, I forgot to tag the issue in that commit!

&nbsp; - Andy

MKGaru, 2021-01-20:
Thank you for fixed. I confirmed.

closed 2021-01-20 by phoddie