Moddable-OpenSource/moddable #545: IRAM_ATTR  Monitor ESP32

louisvangeldrop, 2021-01-17:
Do you have questions about
I am a newbie to the C-language. My question is: does  the ModMonitor.c for ESP32 use IRAM_ATTR  for handling the interrupt in Ram for more accurate timings. Same question for time.microseconds.

I have transpiled rf433-vhduino from c to typescript and am using nodejs the software on a Raspberry-pi with no problems. However If I run the same code - using time.microseconds and the monitor object using an ESP32-board, it does not work.  
My question is: are the microseconds timings accurate enough. A RF433-receiver emits thousands interrupts/second at the 200 microsecond level. So exact timing is needed in order to remove noise.

phoddie, 2021-01-17:
@louisvangeldrop, welcome.

The Digital Monitor implementation for ESP32 does not force the interrupt handler into instruction RAM. You can see the absence of the `IRAM_ATTR` attribute on [the ISR](https://github.com/Moddable-OpenSource/moddable/blob/425f1c756512a964022963d323a24fbddbad19b9/modules/pins/digital/monitor/esp32/modMonitor.c#L202).

You could add it, but I don't think that will get you to accurate time stamps. The ISR itself doesn't capture the time stamp. It posts a message to wake the JavaScript run loop. The timestamp would then need to be taken in JavaScript. That would generally occur a short time later, but the time delta is non-deterministic because it depends on what else the ESP32 is doing.

A Raspberry Pi is many times faster than an ESP32. You may well be able to do this all in JavaScript on the Pi but it wouldn't be practical on an ESP32.
In cases where high performance or very accurate timing is needed, the usual approach is to do that portion of the work in C. Then trigger a JavaScript function with the result.

louisvangeldrop, 2021-01-17:
@phoddie ,
Thanks for your prompt response. The progam runs at  a raspberry pi b3 and uses roughly 6% cpu-time ( 80% at the raspberry pi zero w). Actually I don't know how much faster a Raspberry Pi is compared to an ESP32, but I thought, I give a try, since the cpu-usage in the Instrumention was quite low.

phoddie, 2021-01-17:
@louisvangeldrop - I didn't mean to be entirely discouraging. ;) It might be possible, but it does sound ambitious.

It would be an interesting project to build a high-speed digital interrupt monitor.  It could capture the time stamps in the interrupt (to minimize latency) and then deliver those in batches to the script (to minimize overhead). Depending on how much processing you need to do on that, it could work.

FWIW - an ESP32 runs at 240 MHz. The Pi 3 B+ runs at 1.4 GHz. The Pi CPU has a bunch of architectural advantages as well. Because RAM is very limited (about 300 KB, in practice), the ESP32 majority of code is run from flash, which s much slower than code running from RAM as on the Pi.  In addition, running V8 on Raspberry Pi lets you use V8's native code generation. XS is aways interpreted, and is generally more than fast enough but has its limits. All that adds up. ;)

louisvangeldrop, 2021-01-17:
Is it possible to return a Tick at microsecond level within the Monitor-function- as pigpiod does at the raspberry pi?

phoddie, 2021-01-17:
I'm not familiar with pigpiod or that mechanism.

Just to get a quick estimate of performance, I built a test case to poll microseconds from a script:

```js
import Time from "time";

let data = new Uint32Array(512);

for (let i = 0; i < 512; i++)
	data[i] = Time.microseconds;

let delta = 0;
for (let i = 0; i < 511; i++)
	delta += data[i + 1] - data[i];

trace(`Average us between samples ${delta / 511}\n`);
```

On an ESP32 running an instrumented build (`mcconfig -i -m -p esp32`) it outputs:

```
Average us between samples 13.88454011741683
````

That's about 71,400 samples per second.

(You can run the example in the helloworld project -- just add `"$(MODDABLE)/modules/base/microseconds/manifest.json"` to the `include` array.)

louisvangeldrop, 2021-01-17:
Noise reduction skips everything under 150 us. There error-margin in timings is substantially larger than 14 us. So there is hope for me to get it running on a ESP32

phoddie, 2021-01-18:
Cool.  FYI - I used an instrumented build for the test above because they run scripts faster than a debug build. Since performance matters here you may need to do the same.

phoddie, 2021-01-18:
The original issue has been addressed, so I'm going to close this out.

This project seems interesting. It looks like an optimization problem and those are fun. If you have questions or something to review, please let me know. You can open another issue or post on [our Gitter](https://gitter.im/embedded-javascript/moddable).

closed 2021-01-18 by phoddie