Moddable-OpenSource/moddable #548: avoid allocation for small BigInts?

dckc, 2021-01-21:
**Describe**
Every `BigInt` seems to involve a `fxNewChunk` allocation... even small values that could fit in the `slot` structure. Perhaps the allocation could be avoided for small values.

**Why do you think this feature would be useful?**
In representing money, the Agoric platform is being revised to use `BigInt` to avoid overflow (https://github.com/Agoric/agoric-sdk/issues/2210), but we wonder about the performance impact.

We haven't done performance measurements yet, though.

**Describe alternatives you've considered**
accepting the situation as it is.

cc @dtribble @erights

phoddie, 2021-01-21:
It is fun to think about optimizing `BigInt` for small integers!

The XS chunk allocator is very fast. Since free space is a contiguous within a block, it usually just advances a pointer. There are two costs: garbage collection and calculation. You are going to be running the garbage collector more often. The calculations are going to be much slower, since you'll be bypassing both the integer and floating point paths. The impact, of course, depends on your project's behavior.

The garbage collection load would be reduced with the small integer change you propose. But, each `BigInt` will still require a slot, and those also need to be collected.

The effort to implement what you propose doesn't look trivial. Lots of code would need to be updated: the garbage collector, the BigInt implementation, the bindings used by native code to work with BigInt, the linker which builds ROM images, and the snapshot code. It is surely possible. Before diving into that we should have real-world data that suggests such a change would have a meaningful benefit.

dckc, 2021-01-21:
Thanks for the quick and thorough reply. We'll let you know if we get measurements.

phoddie, 2021-02-23:
Going to close this. Happy to revisit it if there's a need.

closed 2021-02-23 by phoddie