Moddable-OpenSource/moddable #549: XS :   an issue about memory allocation

YaoHouyou, 2021-01-22:
**Version**: [9ed5514
](https://github.com/Moddable-OpenSource/moddable/commit/9ed551497e76fc79f3e901d5a73fdd8f509233c3) 

**Description**
When running the following test case, XS is about 600 times slower embedded JS engine quickJS.  Compared with the resource-constrained JS engine JerryScript, XS is about 170 times slower. It looks like a problem about memory allocation.

Their running time is as follows:
[quickJS](https://bellard.org/quickjs/): `152 ms`
[duktape](https://duktape.org/): `342 ms`
[jerryscript](https://github.com/jerryscript-project/jerryscript): `520 ms`
XS: `91251 ms`

**Test case**
``` js
var NISLFuzzingFunc = function () {
    var row = 1000;

    var arr = [];
    for (var index = 0; index < row; index++)
        arr[index] = new Array();

    for (var i = 0; i < 1024; i++){
        for (var j = 0; j < arr.length; j++){
            arr[j][i] = 0;
        }
    }
};
NISLFuzzingFunc();
```
**Command**:
``` bash
./moddable/build/bin/lin/release/xst testcase.js
```

**Build environment:**  Linux

**CPU:**
Intel(R) Core(TM) i9-9940X CPU @ 3.30GHz

**Build steps**
``` bash
cd $MODDABLE/xs/makefiles/lin
make
```

phoddie, 2021-01-22:
After re-reading the code, I recall the details. The behavior is intentional. It is good for code written for constrained devices. It is can be slow for code written for powerful devices.

XS wants you to allocate large arrays all at once, rather than one element at a time. This avoids  stressing the garbage  collector and  allows them to be stored in contiguous memory for fast read/write access. Since there is no way to do that explicitly in JavaScript, there is a special case on  `Array.prototype.fill` To use, it rework your test script slightly, by changing how the column of arrays is allocated.

```js
		arr[index] = (new Array(1024)).fill(undefined);
```

With that change in place, performance is substantially improved.

But... `xst` is not intended for profiling. It uses a build of XS with debugging support enabled which generates slower code.  To profile, you should rebuild `xst` after removing the following line from the make file:

```
	-DmxDebug=1 \
```

With the change above on a release build, my timing is about 655 ms. That is on a 1.6 GHz Intel Core i5. I  expect your  i9-9940X CPU @ 3.30GHz is quite a bit faster.

YaoHouyou, 2021-01-23:
Is the "655 ms" for case_A?
case_A:
``` js
var NISLFuzzingFunc = function () {
    var row = 1000;

    var arr = [];
    for (var index = 0; index < row; index++)
        arr[index] = (new Array(1024)).fill(0);
};
NISLFuzzingFunc();
```
or the first test case（Test case）. After I delete the 46th line of [xst.mk](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/makefiles/lin/xst.mk)  and recompile, there is no performance improvement when re-executing the first test case. Where did I go wrong?
``` bash
cd $MODDABLE/xs/makefiles/lin
# next, delete 46th line of xst.mk
rm -rf $MODDABLE/build
make
$MODDABLE/build/bin/lin/release/xst testcase.js
```
In addition：
Allocate all memory space at one time with Array.prototype.fill is indeed fast, all engines are fast, there is no doubt. However, once we need to expand the array dynamically, XS will be much slower than other resource-constrained JS engines. Because we cannot guarantee that the JS code running on XS does not have the need to dynamically expand the array. In this respect, other resource-constrained JS engines do a good job. Maybe XS can be improved.

phoddie, 2021-01-23:
I'm not sure what you mean by "case_A". There is one script in the original report. That is what I used.

The shorter script you provided in your comment above will not see much impact from running in release build because it doesn't execute many byte codes. The original does show a benefit because the doubly nested loop executes many more byte codes.

> Maybe XS can be improved.

Certainly. However, this optimization is not our priority as this sort of situation doesn't come in our product work. If someone wants to explore solutions, that is welcome. Unfortunately, no engine can be the best at everything.

YaoHouyou, 2021-01-23:
I delete the 46th line of xst.mk and recompile xst with the following steps. Then, execute the test cases in the original report with xst, the execution time is more than 90000ms. Which step did I go wrong? And what should I do? Thank you.
**build:**
``` bash
cd $MODDABLE/xs/makefiles/lin
# next, delete 46th line of xst.mk
rm -rf $MODDABLE/build
make
```

**run:**
``` bash
$MODDABLE/build/bin/lin/release/xst testcase.js
```
**testcase.js**
``` js
var NISLFuzzingFunc = function () {
    var row = 1000;

    var arr = [];
    for (var index = 0; index < row; index++)
        arr[index] = new Array();

    for (var i = 0; i < 1024; i++){
        for (var j = 0; j < arr.length; j++){
            arr[j][i] = 0;
        }
    }
};
NISLFuzzingFunc();
```

phoddie, 2021-03-04:
This kind of synthetic example is interesting but not relevant to uses of XS in its target embedded environments. It isn't getting in the way of developing embedded systems in JavaScript because embedded developers already understand that this kind of code is inherently inefficient.

It is no unlike [this issue](https://github.com/facebook/hermes/issues/463) you opened against Hermes. Interestingly, while you did not include XS in the performance numbers, it is actually the falsest engine for that particular test case.

closed 2021-03-04 by phoddie

YaoHouyou, 2021-03-05:
Thank you for your reply.