Moddable-OpenSource/moddable #550: exception: sort: no original host function! from xsnap after SES lockdown shim

dckc, 2021-01-26:
**Build environment:** Linux
**Target device:** xsnap

**Description**

After loading a SES shim, [bootstrap-umd.js](https://gist.github.com/dckc/66fcdfc274f13053cfc6a1138c885da7#file-bootstrap-umd-js), calling `.sort()` on an array crashes with `TypeError: sort: no original host function`

**Steps to Reproduce**
1. build xsnap (with larger creation parameters as below)
2. get `bootstrap-umd.js` from above, `nohost.js` below
3. `xsnap -s bootstrap.umd.js nohost.js `
4. out comes `TypeError: sort: no original host function`

`nohost.js` just has one line:

```js
["a", "b"].sort()
```

or two lines:

```js
lockdown();
["a", "b"].sort()
```

```
$ xsnap -s bootstrap.umd.js nohost.js 
Removing intrinsics.ArrayBuffer.fromBigInt
...
bootstrap.umd.js:2606: exception: Object.prototype.__lookupGetter__: cannot coerce undefined to object!
nohost.js:2: exception: sort: no original host function!
TypeError: sort: no original host function
```

**Other information**

```diff
xsnap$ git diff
diff --git a/xsnap/sources/xsnap.c b/xsnap/sources/xsnap.c
index 2a544b2..111bd03 100644
--- a/xsnap/sources/xsnap.c
+++ b/xsnap/sources/xsnap.c
@@ -176,10 +176,10 @@ int main(int argc, char* argv[])
                1 * 1024 * 1024,        /* initialHeapCount */
                1 * 1024 * 1024,        /* incrementalHeapCount */
                4096,                           /* stackCount */
-               4096*3,                         /* keyCount */
+               32000,                  /* keyCount */
                1993,                           /* nameModulo */
                127,                            /* symbolModulo */
-               256 * 1024,                     /* parserBufferSize */
+               1024 * 1024,                    /* parserBufferSize */
                1993,                           /* parserTableModulo */
        };
        xsCreation* creation = &_creation;
```

The only occurrence of this diagnostic in `xs/sources` seems to be:

```
xsFunction.c
779:		mxTypeError("no original host function");
```

I tried turning off `mxMetering` in `xsnap.mk` but then I got compile errors.

cc @warner

phoddie, 2021-01-26:
It may be that the patches for metering are causing trouble. Try reducing `fxPatchBuiltIns` to a stub.

dckc, 2021-01-26:
Yes, that makes these symptoms go away.

Are snapshots with and without the stub compatible? They don't seem to be identical:

```
$ cmp test/fixture-snapshot.xss no-meter/fixture-snapshot.xss 
test/fixture-snapshot.xss no-meter/fixture-snapshot.xss differ: byte 3, line 1

$ ls -l test/fixture-snapshot.xss no-meter/fixture-snapshot.xss 
-rw-rw-r-- 1 connolly connolly 425923 Jan 26 09:04 no-meter/fixture-snapshot.xss
-rw-rw-r-- 1 connolly connolly 425987 Jan 26 09:04 test/fixture-snapshot.xss
```

phoddie, 2021-01-26:
OK, great. Mark bumped into the metering patches in a separate issue. So, it was our mistake to leave the experimental metering patches enabled by default. I suggest you leave them disabled for your mainline work to avoid future unpleasant surprises.

> Are snapshots with and without the stub compatible? 

They should be compatible, as long as the metering patches have no observable side effects. But, if you recall, this is an experimental feature to allow Agoric to explore metering based on the arguments to a built-in method. It isn't a committed feature or a final design.

closed 2021-01-26 by phoddie

dckc, 2021-01-26:
>  I suggest you leave ... the experimental metering patches ... disabled for your mainline work

I tried removing `-DmxMetering=1` in the makefile but then it wouldn't build. So I could use a clue on exactly how to do this.

Is stubbing out `fxPatchBuiltIns` the way to go?

phoddie, 2021-01-26:
It looks like `xssnap.c` assumes that metering is always enabled. Fixing that is pretty easy.

In `fx_print`, add `#if` around these lines: