Moddable-OpenSource/moddable #551: Error while running build on Window 10

MageFeb, 2021-01-28:
Hi,
When I run Build I'm getting these error.
I did the installation and add path to my system and user variable as show in this page for windows
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#win-instructions

Aslo, mcconfig  does not work on my side.

Here are the error i'm getting.
C:\Users\MageFeb\Projects\moddable\build\makefiles\win>build

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f C:\Users\MageFeb\Projects\moddable\xs\makefiles\win\xsc.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f C:\Users\MageFeb\Projects\moddable\xs\makefiles\win\xsid.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f C:\Users\MageFeb\Projects\moddable\xs\makefiles\win\xsl.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f serial2xsbug.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f simulator.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f tools.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>call C:\Users\MageFeb\Projects\moddable\build\bin\win\debug\mcconfig -d -m -p x-win C:\Users\MageFeb\Projects\moddable\tools\xsbug\manifest.json
Le chemin d’accès spécifié est introuvable.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>call C:\Users\MageFeb\Projects\moddable\build\bin\win\debug\mcconfig -d -m -p x-win C:\Users\MageFeb\Projects\moddable\tools\mcsim\manifest.json
Le chemin d’accès spécifié est introuvable.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f C:\Users\MageFeb\Projects\moddable\xs\makefiles\win\xsc.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f C:\Users\MageFeb\Projects\moddable\xs\makefiles\win\xsid.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f C:\Users\MageFeb\Projects\moddable\xs\makefiles\win\xsl.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f serial2xsbug.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f simulator.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f tools.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>call C:\Users\MageFeb\Projects\moddable\build\bin\win\release\mcconfig -m -p x-win C:\Users\MageFeb\Projects\moddable\tools\xsbug\manifest.json
Le chemin d’accès spécifié est introuvable.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>call C:\Users\MageFeb\Projects\moddable\build\bin\win\release\mcconfig -m -p x-win C:\Users\MageFeb\Projects\moddable\tools\mcsim\manifest.json
Le chemin d’accès spécifié est introuvable.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>cd %MODDABLE%\examples\helloworld

C:\Users\MageFeb\Projects\moddable\examples\helloworld>mcconfig -d -m -p win
'mcconfig' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

andycarle, 2021-01-28:
Greetings!

I'm sorry to hear that you're having trouble getting started building the Moddable SDK on Windows. Let's see if we can sort out the issues here.

It appears that you do not have `nmake` in your path. There are two likely possibilities here:

1) You may not have Visual Studio installed. We recommend [Visual Studio 2019 Community Edition](https://www.visualstudio.com/downloads/) on Windows 10.

2) If you have Visual Studio installed, you may not be using the correct Command Prompt. To get `nmake` and other tools in your path, you need to use the `x86 Native Tools Command Prompt for VS 2019` rather than the default Command Prompt.

Please double check those two pieces and let me know how it goes!

&nbsp;- Andy

MageFeb, 2021-01-29:
Hi,

Thanks for coming back to me. I do have Visual Studio 19 installed. However I did not try to run the command from that command prompt.

So I just did it and unfortunately I'm getting the same result.

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd %MODDABLE%\build\makefiles\win

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>build

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f C:\Users\MageFeb\Projects\moddable\xs\makefiles\win\xsc.mak /s
'nmake' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.........

andycarle, 2021-01-29:
Strange.

When you run the "x86 Native Tools Command Prompt for VS 2019," what do you see in the command prompt window at launch?  I get:

```text
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.5.4
** Copyright (c) 2019 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>
```

It's that `vcvarsall.bat` that is doing the important work for getting `nmake` into the path.

andycarle, 2021-01-29:
Oh, one other idea. 

It is possible that when you installed Visual Studio 2019, you did not choose the "Desktop development for C++" workload option in the installer. If that's the case, you just need to run-run the installer and choose that workload this time. That will add the C/C++ command line tools.

MageFeb, 2021-01-29:
I confirm that I did not have the header you have, 
I just started the VS and I do not have the "Desktop development for C++" installed. Nice catch :)
Installing at the moment. I'll keep you advise.

Thanks

MageFeb, 2021-01-29:
it's getting better but still some issue :(

# xsc commodettoBitmap.js
'C:\Users\MageFeb\Projects\moddable\build\bin\win\release\xsc' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.
NMAKE :  U1077: 'C:\Users\MageFeb\Projects\moddable\build\bin\win\release\xsc' : code retour '0x1'
Stop.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>call C:\Users\MageFeb\Projects\moddable\build\bin\win\release\mcconfig -m -p x-win C:\Users\MageFeb\Projects\moddable\tools\xsbug\manifest.json
'C:\Users\MageFeb\Projects\moddable\build\bin\win\release\mcconfig' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>call C:\Users\MageFeb\Projects\moddable\build\bin\win\release\mcconfig -m -p x-win C:\Users\MageFeb\Projects\moddable\tools\mcsim\manifest.json
'C:\Users\MageFeb\Projects\moddable\build\bin\win\release\mcconfig' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>xsbug
'xsbug' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

MageFeb, 2021-01-29:
I think there is something about the version of VS I have.

C:\Users\MageFeb\Projects\moddable\build\tmp\win\release\xsl\xsAll.o : fatal error LNK1112: type d'ordinateur module 'x64' en conflit avec le type d'ordinateur cible 'x86'
NMAKE :  U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x64\link.EXE"' : code retour '0x458'
Stop.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f serial2xsbug.mak /s
serial2xsbug.c
serial2xsbug_win.c
C:\Users\MageFeb\Projects\moddable\build\tmp\win\release\serial2xsbug\serial2xsbug.o : fatal error LNK1112: type d'ordinateur module 'x64' en conflit avec le type d'ordinateur cible 'x86'
NMAKE :  U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x64\link.EXE"' : code retour '0x458'
Stop.

C:\Users\MageFeb\Projects\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\Users\MageFeb\Projects\moddable\build XS_DIR=C:\Users\MageFeb\Projects\moddable\xs /c /f simulator.mak /s
main.cpp
C:\Users\MageFeb\Projects\moddable\build\tmp\win\release\simulator\main.o : fatal error LNK1112: type d'ordinateur module 'x64' en conflit avec le type d'ordinateur cible 'x86'
NMAKE :  U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x64\link.EXE"' : code retour '0x458'
Stop.

andycarle, 2021-01-29:
I believe the results you're seeing there are what would happen if you accidentally build with "x64 Native Tools Command Prompt for VS 2019" instead of "x86 Native Tools Command Prompt for VS 2019". 

Make sure your Command Prompt looks like this:

[image]

Not like this:

[image]

Once you've verified that, delete all build output (`%MODDABLE%\build\tmp` and `%MODDABLE%\build\bin`) and then try building the tools again.

MageFeb, 2021-01-29:
Whoooo thanks a lot,
that did the trick. 

I was opening this one with my search
```
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.8.3
** Copyright (c) 2020 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86_x64'

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>
```
unfortunately it did not show the x86 only, I had to go to `C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Visual Studio 2019\Visual Studio Tools\VC` to see all the other option :) or just type x86 in my search bar. 

I deleted both folder you mentioned and run
`cd %MODDABLE%\build\makefiles\win
build` 
and it look to work great.

Now when I run xsbug I get this trouble from the windows x86 
`C:\Users\MageFeb\Projects\moddable\build\makefiles\win>xsbug
'xsbug' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.`

MageFeb, 2021-01-29:
Nerver mind, I  just realise that path have been created once I run the build `C:\Users\<user>\Projects\moddable\build\bin\win\release`

In the step 4 `Edit the System Variable Path to include the Moddable SDK tools directory. Update the path as necessary for your system by navigating to the correct folder using the "Browse..." button.

Variable Name: Path
Variable Value (add to the existing list): C:\Users\<user>\Projects\moddable\build\**bin\win\release**`
And in the step 5 with the build it create that folder \**bin\win\release** that initially did not exist.

thanks for your help, it is really appreciated

closed 2021-01-29 by MageFeb

MageFeb, 2021-01-29:
I just want to thank you so much for your time and the help.

Now it's time for me to play with the book and make somes test.

Best regards
Manuel Choucino

andycarle, 2021-01-29:
@MageFeb I'm glad we got it sorted out! Don't hesitate to open another issue here if you need additional assistance.

&nbsp;- Andy

asidai, 2021-05-05:
Hi, 

I also have the same problem about "xsc commodettoBitmap.js" that was described as above
----------------------------------------------------------------------------------------------------------------------------
C:\Users\USER\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\USER\Project
s\moddable\build XS_DIR=C:\Users\USER\Projects\moddable\xs /c /f tools.mak /s
win_xs.c
xsAll.c
xsAPI.c
xsArguments.c
xsArray.c
xsAtomics.c
xsBigInt.c
xsBoolean.c
xsCode.c
xsCommon.c
xsDataView.c
xsDate.c
xsDebug.c
xsError.c
xsFunction.c
xsGenerator.c
xsGlobal.c
xsJSON.c
xsLexical.c
xsMapSet.c
xsMarshall.c
xsMath.c
xsMemory.c
xsModule.c
xsNumber.c
xsObject.c
xsPlatforms.c
xsProfile.c
xsPromise.c
xsProperty.c
xsProxy.c
xsRegExp.c
xsRun.c
xsScope.c
xsScript.c
xsSourceMap.c
xsString.c
xsSymbol.c
xsSyntaxical.c
xsTree.c
xsType.c
xsdtoa.c
xsmc.c
xsre.c
xsa.c
xsc.c
xslBase.c
# xsc commodettoBitmap.js
NMAKE :  U1077: 'C:\Users\USER\Projects\moddable\build\bin\win\debug\xsc.EXE' : return code '0xc0000139'
Stop.
----------------------------------------------------------------------------------------------------------------------------

I also use "x86 Native Tools Command Prompt for VS 2019" command line console, like it below
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.9.4
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd %MODDABLE%\build\makefiles\win
C:\Users\USER\Projects\moddable\build\makefiles\win>build

Can you help it? Thanks a lot

MageFeb, 2021-05-16:
Hi, sorry for the delay. I didn't see your email.

I had several issu in my case. One of them what I didn't install all the
correct packet from visual studio.

Alao, at the time I had the issue, the step 4 was not possible as I had to
do the step 5 first then the folder and path was created. From that moment
I was able to update the path variable in Windows system.

And to finish, I didn't use the correct command prompt. Look at the picture
and make sure you use the x86. Do not look at the path that said x86 but at
the top left of the window.

Regards