Moddable-OpenSource/moddable #553: Support for Raspberry Pi Pico and the RP2040

houghtonap, 2021-02-01:
**Describe**
Support for the Raspberry Pi Pico as a target environment.

**Why do you think this feature would be useful?**
This feature seems similar to other microcontroller support and would benefit the Raspberry Pi Pico community by providing an alternative to Micro/Circuit Python.

**Describe alternatives you've considered**
Several alternatives were looked into: muJS and Qjs. MuJS unfortunately is only ES5 compatible and does not have embedded support. Qjs provides ES2020 support, but also does not have embedded support. Moddable has ES2020 support with embedded support and has been implemented for other similar microcontrollers.

devsnek, 2021-02-03:
I'd love to help get this supported in any way I can.

phoddie, 2021-02-03:
@houghtonap - thanks for kicking this off. @devsnek - thanks for jumping in. Help is more than welcome.

I took a quick look when the part was announced. It certainly seems like the chip can handle XS and the Moddable SDK comfortably. We could look at kicking off a public development branch. The biggest challenge in these projects is usually the toolchain -- getting to the point where `mcconfig` "just works" with a new embedded target.

Does anyone here have any experience with the RP2040 build yet? These are some basic questions that need to be answered:

- What's the toolchain? I assume the toolchain is GCC.
- Are builds supported on macOS, Windows, and Linux? How similar/different are those?
- Is the build direct (like our ESP8266 build) or are there other layers of device tools (like the ESP-IDF for ESP32)? Direct is generally easier...
- What's the process for deploying a binary? 
- What's does it take to open a serial connection on the device back to the computer so we can run xsbug and deploy mods?

Once the `helloworld` is working, it is mostly a matter of going through the various IO interfaces one by one to bring up an implementation for the MCU. That part is easily shared.

I (optimistically!) ordered a couple RP2040 boards so I can eventually try this out. Of course, there is a bunch of work before an actual device is useful....

devsnek, 2021-02-03:
The toolchain is GCC (gcc-arm-none-eabi). By default the pico ships with a "boolsel" mode where it appears as a usb mass storage device that you copy a UF2 file into, and you can also use cortex SWD programmers if you have the requisite hardware. The pico sdk comes with some helpful stuff like `stdio_usb`, which sets up the usb port as an 11520 baud serial port which receives stderr/stdout writes, but I'm not sure if that's the sort of thing you'd link into xs. I have one pico I can test this on. I've got some experience with the board from working on getting embedded rust to run on it, but I'm not very familiar with xs. The SDK (https://github.com/raspberrypi/pico-sdk) works everywhere afaik and there are some rather excellent PDFs available as well:
- https://datasheets.raspberrypi.org/rp2040/rp2040_datasheet.pdf
- https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf

phoddie, 2021-02-03:
@devsnek - thanks for the details. It helps to hear from someone with first hand experience.

We have some experience using USB storage devices for flashing with the Moddable SDK. It works, of course, but mounting/unmounting tends to be slower than using a simple serial connection for flashing. The serial USB port you describe likely will work for xsbug (we would want to run it faster than 115200 baud).

I'll find some time to look through the links you provided.

FWIW - bringing up XS itself is pretty easy. It is just standard, portable C. Unfortunately, the result is also pretty inefficient because you don't get our support for precompiled, ROM-resident archives. Most of the work to do that is in the tooling -- hence the reference to `mcconfig` above.

wilberforce, 2021-02-03:
What does this board offer over the ESP32 or ESP8266? 
It looks like it just has general IO and no wifi or BLE - so would require other hardware for projects?

devsnek, 2021-02-03:
@phoddie I've just learned that while in bootsel mode the pico exposes an additional usb interface which can be used to directly write to flash/ram, among other things ("picoboot" in the datasheet)

phoddie, 2021-02-03:
@devsnek - picoboot sounds like what I was hoping for. Thanks.

phoddie, 2021-02-03:
> What does this board offer over the ESP32 or ESP8266?
> It looks like it just has general IO and no wifi or BLE - so would require other hardware for projects?

I more-or-less agree. The standard ARM core is nice for some projects, though the Moddable SDK doesn't have any strong CPU dependency. It sits somewhere between ESP8266 and ESP32 in terms of compute power and memory.

The little I've read about the IO capabilities sounds interesting for some projects. I haven't had a chance to explore the details yet. I could imagine this MCU being a nice host for robotics, like @dtex's [J5e](https://dtex.github.io/j5e/) project

But, mostly I think a lot of people are likely to use this as "my first MCU" because of the brand. It would be great to have real, modern JavaScript and the Moddable SDK available to them.

houghtonap, 2021-02-04:
I agree that a lot of people are likely to use this as "my first MCU" project because of the brand and that is was why I asked about possible support. Having a modern JavaScript available for the board would be fantastic and could really push JavaScript in the IOT market. In looking at JavaScript for embedded use, I stumbled upon moddable. I'm still looking at what it as to offer, but I have to commend the people working on it for the documentation and the SDK install instructions. This is the first project I have used where the instructions were clear and the SDK and/or tools installed without any hitches. Great work and community support!!

> @wilberforce 
> What does this board offer over the ESP32 or ESP8266?
> It looks like it just has general IO and no wifi or BLE - so would require other hardware for projects?

You always have to ask yourself when you start a project, do I really need WIFI or BLE to accomplish the task at hand. If not, then picking a board with it may not be the best choice. However, [Adafruit](https://www.adafruit.com/) is making a Pico board in their feather form factor and that opens the door to adding the exact hardware you need for your project. I do agree that if you need WIFI or BLE, then having it on board and choosing an ESP32 or ESP8266 may be a more cost effective solution.

What I find interesting about the Pico is the PIO state machines which actually may lower the cost for a project by allowing emulation of other protocols. I came across this article today about using the PIO state machines: "[Open hardware Raspberry Pi Pico VGA, SD Card, and Audio demo board to support QVGA video playback](https://www.cnx-software.com/2021/02/03/open-hardware-raspberry-pi-pico-vga-sd-card-and-audio-demo-board-to-support-qvga-video-playback/)". Basically, the PIO state machines are being used to produce a VGA display signal and talk with an SD card. Makes me wonder whether these PIO state machines will be placed on future chips from other manufactures rather than implementing I2C and other protocols on chip...

@phoddie from reading the Pico docs and SDK examples, it appears that you create a single application to be run on the Pico. Press bootsel, plug in the cable and the drive appears in your OS. You then copy it to the drive and the Pico writes it to flash. Basically Micro/Circuit Python is a single application and they just have you drop the application onto the drive, then you open a Thonny (terminal) connection and send your commands. I'm thinking that moddable could be the standalone program and dropped onto the drive instead of Micro/Circuit Python and instead of using Thonny you could use xsbug to talk with moddable over the USB C serial connection. The Pico's unique strategy seems to make the ability of placing embedded interpreters on the chip easy, at least on paper... Possibly another feature that future chips from other manufactures may incorporate...

wilberforce, 2021-02-05:
Looks like arduino is going to release a wifi model version:

https://blog.arduino.cc/2021/01/20/welcome-raspberry-pi-to-the-world-of-microcontrollers/
https://www.hackster.io/news/arduino-announces-raspberry-pi-rp2040-core-port-arduino-nano-rp2040-connect-board-615085ce4791

https://www.u-blox.com/en/product/nina-w10-series-open-cpu

The ironic part is the ublox module contains an Esp32!

phoddie, 2021-02-09:
Good news. My RP2040 board arrived. Now I need to find some time to try it....

barbiani, 2021-02-09:
Ha!

But, the iMX-RT
<https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/i-mx-rt-crossover-mcus:IMX-RT-SERIES>
has
much better performance for the money I guess.

Imagine it with a standard gui toolkit!

I understand that the Pi foundation name will attract much more users to
the sdk. Sad.

phoddie, 2021-02-09:
The NXP parts do look intriguing. But, that's a topic for a different thread.

phoddie, 2021-02-13:
I wanted to share a brief update. We have made some progress in bringing up the Moddable SDK build for the Pico (well, @mkellner has!). We're debugging the output of that now.

The current plan is to get "hello world" working and then post that to a branch here. We are tentatively estimating that to be ready next week. Once that's in place we can work together on creating a real port. There will be plenty to do:

- GPIO/Digital
- Analog input
- PWM output
- I2C
- SPI
- Audio?
- Look at how to integrate PIO. A module like the RMT on ESP32 might be realistic starting point
- Web Workers - looks probably possible but not entirely obvious
- Flash read/write
- Preferences
- Mods
- Bulletproofing builds on macOS, Windows, and Linux
- Documenting set-up
- Testing
- (Probably lots of other things I forgot...)

barbiani, 2021-02-13:
While working on this... how about releasing a port template for new
microcontrollers? Leave empty stubs for all the necessary functions to
bring moddable up.

phoddie, 2021-02-13:
Nice idea. is Perhaps someone will generate that based on the work we do together.

mkellner, 2021-02-20:
`pico` branch is now available.

phoddie, 2021-02-23:
Closing this out as the Pico port is underway and the details are being tracking in [discussion threads](https://github.com/Moddable-OpenSource/moddable/discussions/568). There are lots of opportunities to help out with the port.

closed 2021-02-23 by phoddie