Moddable-OpenSource/moddable #554: esp-idf-4 linux build error

barbiani, 2021-02-02:
Starting from a clean linux VM.

IDF 4.2 and Moddable branch esp-idf-4.

It seems that Bitmap.xsb is never created.

```
> cd build/makefiles/lin
> make
...
...
# tools debug : xsc commodettoBitmap.js
make[1]: ~/moddable/build/bin/lin/debug/xsc: Invalid argument
make[1]: *** [tools.mk:279: ~/moddable/build/tmp/lin/debug/tools/modules/commodetto/Bitmap.xsb] Error 127
make[1]: Leaving directory '~/moddable/build/makefiles/lin'
make: *** [makefile:42: debug] Error 2
$ ls ~/moddable/build/tmp/lin/debug/tools/modules/commodetto/Bitmap.xsb
ls: cannot access '~/moddable/build/tmp/lin/debug/tools/modules/commodetto/Bitmap.xsb': No such file or directory
```

mkellner, 2021-02-02:
Hello, 
Can you try clearing the build previous build results by
`rm -rf $MODDABLE/build/bin $MODDABLE/build/tmp` and then build tools again? Prior build assets would be incompatible with the updated tree.

A number of us are able to build and run on linux without issue.

barbiani, 2021-02-02:
Actually there was no previous build. Started from git clone, checkout esp
idf 4 branch... cd to the makefiles folder and run make.

mkellner, 2021-02-02:
Can you provide the complete output, please? There may be a clue in the omitted information.

barbiani, 2021-02-02:
Here it is:

@kali: /home/user/moddable/build/makefiles/lin$ make
make -f  /home/user/moddable/xs/makefiles/lin/xsc.mk
make[1]: Entering directory ' /home/user/moddable/build/makefiles/lin'
# xsc debug : cc xsBigInt.c
In file included from  /home/user/moddable/xs/sources/xsAll.h:41,
                 from  /home/user/moddable/xs/sources/xsBigInt.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# xsc debug : cc xsCode.c
In file included from  /home/user/moddable/xs/sources/xsScript.h:41,
                 from  /home/user/moddable/xs/sources/xsCode.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# xsc debug : cc xsCommon.c
In file included from  /home/user/moddable/xs/sources/xsCommon.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# xsc debug : cc xsdtoa.c
In file included from  /home/user/moddable/xs/sources/xsAll.h:41,
                 from  /home/user/moddable/xs/sources/xsdtoa.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# xsc debug : cc xsLexical.c
In file included from  /home/user/moddable/xs/sources/xsScript.h:41,
                 from  /home/user/moddable/xs/sources/xsLexical.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# xsc debug : cc xsre.c
In file included from  /home/user/moddable/xs/sources/xsAll.h:41,
                 from  /home/user/moddable/xs/sources/xsre.c:21:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# xsc debug : cc xsScope.c
In file included from  /home/user/moddable/xs/sources/xsScript.h:41,
                 from  /home/user/moddable/xs/sources/xsScope.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# xsc debug : cc xsScript.c
In file included from  /home/user/moddable/xs/sources/xsScript.h:41,
                 from  /home/user/moddable/xs/sources/xsScript.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# xsc debug : cc xsSourceMap.c
In file included from  /home/user/moddable/xs/sources/xsScript.h:41,
                 from  /home/user/moddable/xs/sources/xsSourceMap.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# xsc debug : cc xsSyntaxical.c
In file included from  /home/user/moddable/xs/sources/xsScript.h:41,
                 from  /home/user/moddable/xs/sources/xsSyntaxical.c:38:
[... 825 lines cut ...]
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# tools debug : cc xsc.c
In file included from  /home/user/moddable/xs/sources/xsScript.h:41,
                 from  /home/user/moddable/xs/tools/xsc.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# tools debug : cc xslBase.c
In file included from  /home/user/moddable/xs/sources/xsAll.h:41,
                 from  /home/user/moddable/xs/tools/xsl.h:38,
                 from  /home/user/moddable/xs/tools/xslBase.c:38:
 /home/user/moddable/xs/sources/xsCommon.h:47:9: warning: undefining
"__has_builtin"
   47 |  #undef __has_builtin
      |         ^~~~~~~~~~~~~
# tools debug : xsc commodettoBitmap.js
make[1]:  /home/user/moddable/build/bin/lin/debug/xsc: Invalid argument
make[1]: *** [tools.mk:279:
 /home/user/moddable/build/tmp/lin/debug/tools/modules/commodetto/Bitmap.xsb]
Error 127
make[1]: Leaving directory ' /home/user/moddable/build/makefiles/lin'
make: *** [makefile:42: debug] Error 2
@kali: /home/user/moddable/build/makefiles/lin$

mkellner, 2021-02-02:
You originally mentioned that you are `Starting from a clean linux VM.`

Can you please describe that further? For me, that equates to:
   MacOS running VirtualBox VM (up to date), Ubuntu 20.04.1 (also up to date).

barbiani, 2021-02-02:
It would be:

Oracle virtualbox 6.1
guest: Linux kali 5.10.0-kali2-amd64 #1 SMP Debian 5.10.9-1kali1
(2021-01-22) x86_64 GNU/Linux
host: windows 10
gcc (Debian 10.2.1-6) 10.2.1 20210110

By clean I mean a VM that did compile previous moddable idf 3 versions.

$ git clone https://github.com/Moddable-OpenSource/moddable.git
$ cd moddable/build/makefiles/lin
$make --trace

Point of the failure: make tools.mk

tools.mk:278: update target
'/home/user/moddable/build/tmp/lin/debug/tools/modules/commodetto/Bitmap.xsb'
due to: /home/user/moddable/modules/commodetto/commodettoBitmap.js
echo "#" tools debug ": xsc" commodettoBitmap.js
# tools debug : xsc commodettoBitmap.js
/home/user/moddable/build/bin/lin/debug/xsc
/home/user/moddable/modules/commodetto/commodettoBitmap.js -c -d -e -o
/home/user/moddable/build/tmp/lin/debug/tools/modules/commodetto -r Bitmap
make[1]: /home/user/moddable/build/bin/lin/debug/xsc: Invalid argument
make[1]: *** [tools.mk:279:
/home/user/moddable/build/tmp/lin/debug/tools/modules/commodetto/Bitmap.xsb]
Error 127
make[1]: Leaving directory '/home/user/moddable/build/makefiles/lin'
make: *** [makefile:42: debug] Error 2

At this point,
'/home/user/moddable/build/tmp/lin/debug/tools/modules/commodetto/Bitmap.xsb'
is not in the disk.

mkellner, 2021-02-03:
> By clean I mean a VM that did compile previous moddable idf 3 versions.

Okay. So that means you've been through the complete setup successfully. Good. The host should be ready.

If the following were the commands you used:
```
$ git clone https://github.com/Moddable-OpenSource/moddable.git
$ cd moddable/build/makefiles/lin
$make --trace
```

then there are a few things to check:

The esp-idf-4 support is currently in a branch. We will be making that the mainline branch soon. Until then, you will need to checkout the esp-idf-4 branch.

But that doesn't explain why you can't build the host tools.

If you haven't, clean the build-results directories when switching between branches and re-build the tools.
```
$ cd moddable
$ git checkout esp-idf-4
$ rm -rf $MODDABLE/build/bin $MODDABLE/build/tmp
$ cd build/makefiles/lin
$ make --trace
```

Please give it a try.

barbiani, 2021-02-03:
I´ve tried the other branches also. Same result.

Calling xsc (debug or release) directly gives that invalid argument
response.

barbiani, 2021-02-06:
It did build well in ~/moddable but not in /media/c/moddable that is the VM
host shared folder.

xsbug find field has been weird for a while on top of the file names.

[image: image.png]

phoddie, 2021-03-04:
This thread seems to have gone idle. If there's something new, please update it. Otherwise we should close it out. Thank you.

closed 2021-04-06 by phoddie