Moddable-OpenSource/moddable #555: How to compile XS for formance testing

YaoHouyou, 2021-02-04:
We would like to test the performance of XS, but the execution speed of `xst` is slow. The `xst` file compiled according to the [tutorial](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/xst.md#linux) does not seem to be suitable for performance testing. How can I compile XS?

phoddie, 2021-02-04:
> ...but the execution speed of xst is slow

Please explain this statement.

YaoHouyou, 2021-02-04:
``` js 
var c = 1;
for (var i = 0; i < 100000; i++) {
    c = i;
}
```
[quickJS](https://bellard.org/quickjs/): 24 ms
[JerryScript](https://github.com/jerryscript-project/jerryscript): 35 ms
XS: 70 ms
.
Executing the simple loop statement, the execution time of XS is more than twice that of other resource-constrained JavaScript engines. Generally, XS shouldn't be so slow. Is it because `xst` enables debugging support? if yes,  how can I close it? Thanks.

phoddie, 2021-02-04:
The point of your exercise is unclear.

Your methodology for performance measurements is unstated. You just provide code snippets.

The assertion that quickJS is a resource-constrained JavaScript engine is simply untrue. 

On January 22, I took the time to provide a suggestion regarding debugging support to a separate issue you opened. It is strange to open a new issue to reassert this point without any reference to the original.

This is a poor use of my time. I'll be stepping back from this thread.

closed 2021-02-04 by YaoHouyou