Moddable-OpenSource/moddable #557: detecting unhandled promise rejections

dckc, 2021-02-05:
**Description**

Unhandled promise rejections seem to be silently ignored. I gather this is somewhat as-designed, but suppose I want / need to detect and report them... do I have any options?

**Steps to Reproduce**

Run `xst check-unhandled.js`

```js
(async () => {
    Promise.reject('lose');
    Promise.reject('lose2').catch(print);
    Promise.resolve('win').then(print);
    throw 'lose3';
})();

```

**Expected behavior**
some diagnostic... or some way I can instrument xst or xsnap so that js code can detect unhandled promise rejections

**Build environment:** Linux

cc @michaelfig @erights @dtribble @warner

see also https://github.com/Agoric/agoric-sdk/commit/8bc43261ff31fb2afd54fa043c836a1bf4c99fd3

erights, 2021-02-05:
See https://github.com/Agoric/agoric-sdk/issues/2351

phoddie, 2021-02-05:
Given that there's currently no known desirable behavior, I'm going to close this so it doesn't distract or mislead. Feel free to re-open if the investigation by @erights suggests a useful solution.

closed 2021-02-05 by phoddie

phoddie, 2021-02-13:
FWIW - Patrick was inspired by this discussion to implement debugging support for rejected promises. It is in today's update to the Moddable SDK. It works nicely in my experiments but my perspective on promises is far narrower than yours. Please take a look. We're entirely open to feedback, especially from @erights, on how we could make this more useful.

[image]

dtribble, 2021-02-13:
Sweet!