Moddable-OpenSource/moddable #558: Reference to undefined variable "Host" in "examples/piu/sound"

stc1988, 2021-02-06:
**Build environment:** Windows
**Target device:** M5stack,Desktop simulator

**Description**
I got a reference error when "examples/piu/sound" application launch.

**Steps to Reproduce**
1. Launch "examples/piu/sound" application. 
2. See error.
`examples\piu\sound\main.js (79) # Break: SoundAppBehavior.prototype.onDisplaying: get Host: undefined variable!`

https://github.com/Moddable-OpenSource/moddable/blob/157795cbb7b63f17e1c21e0ab1892442f17d755f/examples/piu/sound/main.js#L79

https://github.com/Moddable-OpenSource/moddable/blob/157795cbb7b63f17e1c21e0ab1892442f17d755f/examples/piu/sound/main.js#L96

You should check if the variable "Host" is defined.

andycarle, 2021-02-06:
Thank you for the report! I'll get a fix for that into our next push.

andycarle, 2021-02-06:
I've committed fixes for this one and something similar in `examples\drivers\neopixel\main.js`. Those fixes will be available in our next open source push.

Thank you again for the report.

&nbsp;- Andy

andycarle, 2021-02-13:
This fix is live. @stc1988 please verify on your build when you get a chance. Thanks!

stc1988, 2021-02-13:
I saw this issue fixed.
Thanks.

closed 2021-02-13 by stc1988