Moddable-OpenSource/moddable #559: Missing file piuPlatform.h

zombeej, 2021-02-08:
Total newb here, so my apologies if I've just missed something.

**Build environment:** macOS
**Target device:** ESP32

**Description**
> .../moddable/modules/piu/All/piuAll.h:28:10: fatal error: piuPlatform.h: No such file or directory

**Steps to Reproduce**
1. Navigate to `.../moddable/examples/drivers/neostrand/docExample`
1. Build and install the app using this build command: `mcconfig -d -m -p esp32`
1. See error

phoddie, 2021-02-08:
Thanks for the report and apologies for the trouble. It looks like that got broken last  year when optimizing the Piu Timeline implementation with some native  code.

A quick workaround is to add the Piu manifest to the neostrand/docExample/manifest.json in the `include` section:

```
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODULES)/pins/digital/monitor/manifest.json",

		"$(MODDABLE)/examples/manifest_net.json",
		"$(MODULES)/network/mdns/manifest.json",

		"$(MODDABLE)/examples/manifest_piu.json"
	],
```

But... Piu requires you to have a screen, so use an ESP32  build target with a screen and disable touch:

```
mcconfig -d -m -p esp32/moddable_two touch=""
```

We'll look into a cleaner solution.

zombeej, 2021-02-08:
Thanks so much for the quick reply! I'll see if I can get it working that way. I'm really excited to tinker with this!

phoddie, 2021-02-09:
It turns out there is an easier solution than the workaround.

Timeline is part of Piu but doesn't have any dependencies on the rest of Piu. So, we can replace the include of piuPlatform.h with something more general:

```c
#include "xsPlatform.h"
#include "mc.xs.h"
```

That works here. With luck, it works for you too. We'll include that in the next Moddable SDK update. Thanks again for the report.

zombeej, 2021-02-10:
Hey, that worked! I can build the project for a generic esp32 controller now. It seems the timer is much less stable this way, though it could just be my cheapo microcontroller board (it's a WeMos running a wroom-32). Anyways, I'll tinker with this. Thanks a lot!

phoddie, 2021-02-13:
This  has been  fixed by the commit above. Thanks again for the report.

closed 2021-02-13 by phoddie