Moddable-OpenSource/moddable #560: Startup sound don't work on M5stack

stc1988, 2021-02-11:
**Build environment:** Windows
**Target device:** M5stack

**Steps to Reproduce**
1. Build any application using this build command: `mcconfig -d -m -p esp32/m5stack`
2. I can't hear the startup sound when application launch,.

wilberforce, 2021-02-11:
@stc1988

There was a change to the speaker setup a while back:

https://github.com/Moddable-OpenSource/moddable/commit/ac0732dbb50e6fcb44320e1d7f4b64f80101d08d

```
"config": {
		"screen": "ili9341",
		"touch": "",
		"autorotate": 300,
		"startupSound": "bflatmajor.maud"
	}
```

Are you overriding this in your config?

stc1988, 2021-02-11:
@wilberforce 
I don't override my config.

https://github.com/Moddable-OpenSource/moddable/blob/157795cbb7b63f17e1c21e0ab1892442f17d755f/build/devices/esp32/targets/m5stack/setup-target.js#L13-L26
I confirmed that `speaker.start()` and `speaker.callback` was called.

And,  I cant'  hear the sound in `example/piu/sound` application,so it seems to `AudioOut` has any problem.

andycarle, 2021-02-11:
Thank you for the report. 

The speaker and startup sound are working correctly for me on the M5Stack Core2. Curious that it isn't working on the M5Stack. I'll look into it.

stc1988, 2021-02-11:
Yes, My M5stack Core2 is working correctly too.
It there any way how  debug it?

andycarle, 2021-02-11:
Do you recall if the startup sound ever worked on your M5Stack? If not, we probably just have a pin number wrong.

I know I've tested the audio on the M5Stack Fire, but I'm not sure I've ever actually used a plain M5Stack.

stc1988, 2021-02-11:
It was definitely working around November last year.
I think this issue occured since update to ESP‑IDF 4.2

andycarle, 2021-02-13:
I've pushed a fix to address this. It's working for me on the M5Go, but it could use verification on the original M5Stack.

stc1988, 2021-02-13:
I saw this issue fixed with my M5stack.
Thank you for your help.

closed 2021-02-13 by stc1988