Moddable-OpenSource/moddable #561: esp32 audioin: I2S_COMM_FORMAT_I2S is deprecated

stc1988, 2021-02-13:
**Version:** https://github.com/Moddable-OpenSource/moddable/commit/4e7225eca2a11d704caecc7d0f467a23db241804
**Build environment:** Windows
**Target:** M5stick_c

**Steps to Reproduce**
Build `examples/pins/audioin/waveserver` with `mcconfig -d -m -p esp32/m5stick_c`

I got a build warning below.

> C:\Users\satt6\Projects\moddable\modules\pins\audioin\esp32\audioin.c: In function 'xs_audioin':
> C:\Users\satt6\Projects\moddable\modules\pins\audioin\esp32\audioin.c:86:5: warning: 'I2S_COMM_FORMAT_I2 ' is deprecated [-Wdeprecated-declarations]
>      communication_format: (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB),
>      ^~~~~~~~~~~~~~~~~~~~
> In file included from C:\Users\satt6\esp32\esp-idf\components\soc\src\esp32\include/hal/i2s_ll.h:30,
>                  from C:\Users\satt6\esp32\esp-idf\components\soc\include/hal/i2s_hal.h:28,
>                  from C:\Users\satt6\esp32\esp-idf\components\driver\include/driver/i2s.h:24,
>                  from C:\Users\satt6\Projects\moddable\modules\pins\audioin\esp32\audioin.c:33:
> C:\Users\satt6\esp32\esp-idf\components\soc\include/hal/i2s_types.h:70:5: note: declared here
>      I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
>      ^~~~~~~~~~~~~~~~~~~
> C:\Users\satt6\Projects\moddable\modules\pins\audioin\esp32\audioin.c:86:5: warning: 'I2S_COMM_FORMAT_I2S_LSB' is deprecated [-Wdeprecated-declarations]
>      communication_format: (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB),
>      ^~~~~~~~~~~~~~~~~~~~
> In file included from C:\Users\satt6\esp32\esp-idf\components\soc\src\esp32\include/hal/i2s_ll.h:30,
>                  from C:\Users\satt6\esp32\esp-idf\components\soc\include/hal/i2s_hal.h:28,
>                  from C:\Users\satt6\esp32\esp-idf\components\driver\include/driver/i2s.h:24,
>                  from C:\Users\satt6\Projects\moddable\modules\pins\audioin\esp32\audioin.c:33:
> C:\Users\satt6\esp32\esp-idf\components\soc\include/hal/i2s_types.h:72:5: note: declared here
>      I2S_COMM_FORMAT_I2S_LSB   __attribute__((deprecated)) = 0x02, /*!< I2S format LSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_LSB) correspond to `I2S_COMM_FORMAT_STAND_MSB`*/
> 

I don't know if this is issues, but this is similar to https://github.com/Moddable-OpenSource/moddable/commit/624057e204c1556a90fddf5acfdf5c036f8a6951

andycarle, 2021-02-13:
Thank you for the report. We took care of those for audioout but not audioin, it seems. I'll get the fix there in for our next push.

stc1988, 2021-03-03:
I confirmed with  Release`March 2, 2021`.

closed 2021-03-03 by stc1988