Moddable-OpenSource/moddable #565: SyntaxError from const { default: greeting } = { default: 1 }

dckc, 2021-02-18:
xst reports this as a syntax error:

```js
const { default: greeting } = { default: 1 }
```

**Steps to Reproduce**

```
xst -e 'try { const { default: greeting } = { default: 1 }; print(1) } catch (e) { print(e.message)}'
SyntaxError: missing identifier
```

**Expected behavior**

not a syntax error.

**Build environment:**

v 4e7225eca2a11 on linux

**Other information**

context: https://github.com/Agoric/SES-shim/blob/master/packages/ses/test/test-compartment-transforms.js#L88
https://github.com/Agoric/SES-shim/issues/508

cc @erights @warner

dckc, 2021-02-22:
just ran into another case of this, I think:

```js
const { in: stuff } = { in: 1 };
```

My linux build of `xst` _hangs_ given that input. Odd.

context:

https://github.com/Agoric/agoric-sdk/blob/master/packages/zoe/test/autoswapJig.js#L160

cc @Chris-Hibbert

dckc, 2021-02-26:
@phoddie any thoughts on this one?

phoddie, 2021-02-26:
Yes, we have a fix. We'll get that in our next SDK push.

dckc, 2021-03-02:
fix confirmed

closed 2021-03-02 by dckc