Moddable-OpenSource/moddable #567: unbounded heap growth crashes xst, xsnap (SIGSEGV)

dckc, 2021-02-19:
**Steps to Reproduce**

1. `xst grow.js` where grow.js is given below
   - whether release or debug build
2. SIGSEGV

```js
let stuff = '1234';
for (;;) {
  stuff += stuff;
}
```

It also crashes xsnap, debug and release.

**Build environment:** Linux
**Target device:** xst

cc @warner

somewhat related: https://github.com/Agoric/agoric-sdk/issues/2276 , https://github.com/Agoric/agoric-sdk/issues/2224

phoddie, 2021-02-20:
Hmmm... yes, that happens. I know it used to work correctly. Interesting, the embedded targets seem fine. On an ESP32 for example, there's no crash: it lands in `fxAbort` with an error value of `XS_NOT_ENOUGH_MEMORY_EXIT`. So, strictly speaking, this looks like a bug in port to certain targets rather than a bug in XS itself.
[image]

dckc, 2021-02-23:
yay! seems to be fixed in c22a50bb7fcb.

phoddie, 2021-02-23:
It turns out to be a more complex issue than it first appeared. On systems with limited memory, it isn't a practical concern. On systems with lots of memory (gigabytes) it can be.

We put in a workaround for the problem you reported. It isn't a proper fix for all situations. We should come back to this at some point.

dckc, 2021-02-23:
yeah... I saw `txSize` is 4 bytes... I can see how that could run into overflow.

I do prefer to track this until a proper fix is available. Memory safety is job 1!

dckc, 2021-04-09:
I'm not sure what to verify; the symptoms I was seeing were fixed Feb 22 by c22a50b.

I don't have any way to provoke the more general problem handy.

phoddie, 2021-04-09:
As the problems you could reproduce are resolved, this can be closed.

closed 2021-04-09 by phoddie