Moddable-OpenSource/moddable #571: Some Commodetto examples don't run in simulator

dbhaig, 2021-02-20:
I'm new to moddable and was running through the examples in the examples/commodetto directory. Most ran successfully but I encountered an error on two of them:

image-frames:
     ReferenceError: module "ili9341" not found!
 
pngdisplay:
     ReferenceError: module "ili9341" not found!

What am I missing?

Thank you, in advance, for your assistance. 

(Ubuntu 20.04, ESP-IDF, release v4.2 branch, up to date pull from moddable github repo)

phoddie, 2021-02-20:
Welcome. And thanks for the report.

Not every example can run on the simulator, since some depend on hardware features. But, the `image-frames` and `pngdisplay` should build and launch.

It looks like they both have small mistakes in the manifest. For `image-frames` just remove these lines from the manifest:

https://github.com/Moddable-OpenSource/moddable/blob/90bf07984bdf56643a39ba6fdc7a8061df5a04c7/examples/commodetto/image-frames/manifest.json#L13-L15

For 'pngdisplay', remove this one:

https://github.com/Moddable-OpenSource/moddable/blob/90bf07984bdf56643a39ba6fdc7a8061df5a04c7/examples/commodetto/pngdisplay/manifest.json#L25

We'll get these changes merged into the next Moddable SDK update.

**FWIW** - I'm not sure `pngdisplay` will work properly in the simulator on all host operating systems, because it opens up an mDNS advertisement which may conflict with services provided by the host operating system.

dbhaig, 2021-02-21:
That resolved the issues. Thanks for the prompt response!

For image-frames/manifest.json, I had to also remove the trailing comma on the line above. Otherwise, an invalid manifest error was generated.

The pngdisplay now runs but nothing much happens, probably for the reasons you outlined above.

closed 2021-02-21 by dbhaig

phoddie, 2021-02-21:
Glad that worked. Thanks for the detail about the comma.

FYI - the `pngdisplay` example doesn't display an image when it starts up. It is waiting for a PNG image to be pushed to it over HTTP. There's an [extensive article](https://blog.moddable.com/blog/pngdisplay/) about how it works on our blog.