Moddable-OpenSource/moddable #577: Map get fails on BigInt key

dckc, 2021-02-23:
**Steps to Reproduce**
1. `xst -e 'const store = new Map([[1n, "abc"]]); print(store.get(1n))'`
3. See `undefined`

**Expected behavior**
print abc

**Other information**

I thought maybe a Map's key had to be an Object, but that's WeakMap. "A Map's keys can be any value (including functions, objects, or any primitive)." -- MDN

I checked the XS conformance docs and didn't find anything about this. That suggests test262 doesn't cover this. odd, that.

node and browser consoles both show...

```
> const store = new Map([[1n, "abc"]]); store.get(1n)
'abc'
```

context:

As part of https://github.com/Agoric/agoric-sdk/pull/2486 , I'm running [zoe/test/unitTests/test-fakePriceAuthority.js](https://github.com/Agoric/agoric-sdk/blob/master/packages/zoe/test/unitTests/test-fakePriceAuthority.js) and the assertion at [store.js line 35](https://github.com/Agoric/agoric-sdk/blob/master/packages/store/src/store.js#L35) is failing.

cc @erights @katelynsills

**Build environment:** linux
**Target device:** xst, xsnap, agoric

phoddie, 2021-02-24:
Nice find. It looks like test262 doesn't have a case for this.

It looks pretty easy to add. In `fxTestEntry` add this after the reference case:

```
		else if ((XS_BIGINT_KIND == a->kind) || (XS_BIGINT_X_KIND == a->kind))
			result = gxTypeBigInt.compare(the, 0, 1, 0, a, b);
```

That's enough to make it work, but all the BigInt values will share a hash table entry which is eventually slow. So, in `fxSumEntry` add this after the reference case:

```
		else if ((XS_BIGINT_KIND == kind) || (XS_BIGINT_X_KIND == kind)) {
			address = (txU1*)slot->value.bigint.data;
			size = slot->value.bigint.size;
		}
```

That seems to work in quick testing. I'll ask Patrick to review and recheck with with test262 (just in case) before we integrate this.

dckc, 2021-03-02:
fix confirmed.

closed 2021-03-02 by dckc