Moddable-OpenSource/moddable #584: memory leaks(i2b)

rain6851, 2021-02-26:
# Enviroment
```
operating system: ubuntu18.04
compile command:  cd /pathto/moddable/xs/makefiles/lin
make
test command: ./xst poc
```
# poc:
```
function opt(a, b, v) {
    if (b.length < 1)
        return;

    for (let i = 0; i < a.length; i++)
        a[i] = v;

    b[0] = 2.3023e-320;
}

function main() {
    for (let i = 0; i < 1000; i++) {
        opt(new Uint8Array(100), [1.1, 2.2, 3.3], {});
    }

    let a = new Uint8Array(100);
    let b = [1.1, 2.2, 3.3];
    opt(a, b, {
        valueOf: ()/=> {
            b[0] = {};
            return 0;
        }
    });

    print(b[0]);
}

main();

```
# description
```
=================================================================
==5959==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 36 byte(s) in 1 object(s) allocated from:
    #0 0x7ff16d692602 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x98602)
    #1 0x6637ba in fxDTOAMalloc /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:6346
    #2 0x659e3e in Balloc /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:1676
    #3 0x65a838 in i2b /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:1893
    #4 0x65aef8 in pow5mult /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:2037
    #5 0x6600c6 in strtod2 /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:4436
    #6 0x6651c2 in fxStringToNumber /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:6640
    #7 0x50cf4e in fxGetNextNumberE /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsLexical.c:424
    #8 0x51128a in fxGetNextTokenAux /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsLexical.c:940
    #9 0x50fe2a in fxGetNextToken /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsLexical.c:819
    #10 0x6332aa in fxAssignmentExpression /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1703
    #11 0x6330a9 in fxCommaExpression /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1674
    #12 0x630eac in fxStatement /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1302
    #13 0x62fe9d in fxBody /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1126
    #14 0x639f00 in fxFunctionExpression /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:2639
    #15 0x630600 in fxStatement /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1222
    #16 0x62f4fe in fxProgram /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1068
    #17 0x645a10 in fxParserTree /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsTree.c:168
    #18 0x57a973 in fxLoadScript /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsPlatforms.c:384
    #19 0x6fa841 in fxRunProgramFile /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:1367
    #20 0x6ed74c in main /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:270
    #21 0x7ff16cd2a82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)

Indirect leak of 388 byte(s) in 5 object(s) allocated from:
    #0 0x7ff16d692602 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x98602)
    #1 0x6637ba in fxDTOAMalloc /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:6346
    #2 0x659e3e in Balloc /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:1676
    #3 0x65aab2 in mult /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:1926
    #4 0x65aff7 in pow5mult /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:2062
    #5 0x6600c6 in strtod2 /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:4436
    #6 0x6651c2 in fxStringToNumber /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:6640
    #7 0x50cf4e in fxGetNextNumberE /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsLexical.c:424
    #8 0x51128a in fxGetNextTokenAux /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsLexical.c:940
    #9 0x50fe2a in fxGetNextToken /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsLexical.c:819
    #10 0x6332aa in fxAssignmentExpression /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1703
    #11 0x6330a9 in fxCommaExpression /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1674
    #12 0x630eac in fxStatement /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1302
    #13 0x62fe9d in fxBody /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1126
    #14 0x639f00 in fxFunctionExpression /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:2639
    #15 0x630600 in fxStatement /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1222
    #16 0x62f4fe in fxProgram /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1068
    #17 0x645a10 in fxParserTree /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsTree.c:168
    #18 0x57a973 in fxLoadScript /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsPlatforms.c:384
    #19 0x6fa841 in fxRunProgramFile /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:1367
    #20 0x6ed74c in main /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:270
    #21 0x7ff16cd2a82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)

Indirect leak of 36 byte(s) in 1 object(s) allocated from:
    #0 0x7ff16d692602 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x98602)
    #1 0x6637ba in fxDTOAMalloc /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:6346
    #2 0x659e3e in Balloc /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:1676
    #3 0x65a838 in i2b /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:1893
    #4 0x65feef in strtod2 /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:4367
    #5 0x6651c2 in fxStringToNumber /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsdtoa.c:6640
    #6 0x50cf4e in fxGetNextNumberE /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsLexical.c:424
    #7 0x51128a in fxGetNextTokenAux /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsLexical.c:940
    #8 0x50fe2a in fxGetNextToken /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsLexical.c:819
    #9 0x6332aa in fxAssignmentExpression /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1703
    #10 0x6330a9 in fxCommaExpression /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1674
    #11 0x630eac in fxStatement /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1302
    #12 0x62fe9d in fxBody /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1126
    #13 0x639f00 in fxFunctionExpression /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:2639
    #14 0x630600 in fxStatement /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1222
    #15 0x62f4fe in fxProgram /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsSyntaxical.c:1068
    #16 0x645a10 in fxParserTree /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsTree.c:168
    #17 0x57a973 in fxLoadScript /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsPlatforms.c:384
    #18 0x6fa841 in fxRunProgramFile /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:1367
    #19 0x6ed74c in main /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:270
    #20 0x7ff16cd2a82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)

SUMMARY: AddressSanitizer: 460 byte(s) leaked in 7 allocation(s).

```

phoddie, 2021-03-17:
Thank you for the report.

XS now reports the syntax error on line 19 and AddressSanitizer does not report any problems, leaks or otherwise.

rain6851, 2021-03-18:
> Thank you for the report.
> 
> XS now reports the syntax error on line 19 and AddressSanitizer does not report any problems, leaks or otherwise.

Please remove code in 19, and  try it again.

phoddie, 2021-03-18:
> Please remove code in 19, and try it again.

Do you mean to remove the extraneous "/" on line 19? That makes line  19:

```
        valueOf: () => {
```

With that change,  it compiles and runs to completion,. The output  is `2.3023e-320`, the same output as V8.

Have I misunderstood?

rain6851, 2021-03-19:
> > Please remove code in 19, and try it again.
> 
> Do you mean to remove the extraneous "/" on line 19? That makes line 19:
> 
> ```
>         valueOf: () => {
> ```
> 
> With that change, it compiles and runs to completion,. The output is `2.3023e-320`, the same output as V8.
> 
> Have I misunderstood?

It may have been repaired sometime in the past.

closed 2021-03-23 by phoddie