Moddable-OpenSource/moddable #590: RMT: tx_config parameter is incomplete

stc1988, 2021-03-02:
I'm trying to make an infrared transmitter with RMT module.
But, I found that setting tx_config parmeter in `rmt.c` is incomple.
Only `idle_output_en` and `idle_level` are set.

https://github.com/Moddable-OpenSource/moddable/blob/8c62c86e6bc7c3defe484c61e1f07d8b4e159241/modules/pins/rmt/esp32/rmt.c#L127-L129

I want to setting `tx_config` parmeter from RMT class constructor.

You can see the default tx_config parameters [here](https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/components/driver/include/driver/rmt.h#L94-L111)

phoddie, 2021-03-02:
@stc1988 -  thanks for the report (and cool demo). I **think** what you want is to be able to set **all** of the fields in `tx_config` from your script. Is that correct?

stc1988, 2021-03-02:
Here is my temporarily fix to send an infrared. The only difference from default value is  setting `carrier_en` to `true`.
```` javascript
	if (transmit){
		// config.tx_config.idle_output_en = 1;
		// config.tx_config.idle_level = 0;
		config.tx_config.loop_en = false;
		config.tx_config.carrier_freq_hz = 38000;
		config.tx_config.carrier_duty_percent = 33;
		config.tx_config.carrier_level =  RMT_CARRIER_LEVEL_HIGH;
		config.tx_config.carrier_en = true;
		config.tx_config.idle_level =  RMT_IDLE_LEVEL_LOW ;
		config.tx_config.idle_output_en = false;
	}else{
````

If you set default values to all of the fields in `tx_config` in `rmt.c`,  my scritpt needs to set only `carrier_en`  to `true`.

phoddie, 2021-03-02:
Thank you for the details. Since this is specific to `tx_config` and ESP32, maybe a dedicated property makes sense:

```js
let rmt = new RMT({
	pin: 17,
	channel: 1,
	divider: 100,
	tx_config: {
		carrier_en: 1,
		idle_output_en: 1
	}
});
```

All values in the native `tx_config` structure are initialized to zero. Only the properties present in the JavaScript options object are set. This gives access to all all native fields, but only those use need to be set. The property names can match the ESP32 C field names so it is straightforward to use.

What do you think?

stc1988, 2021-03-03:
I updated SDK to  `March 2, 2021`.
My script sets `tx_config` fields what I need. It works as I exepected.

```javascript
let rmt = new RMT({pin: IrSendPin, channel: 0, divider: 80, direction: "tx",
					tx_config: {
						loop_en: 0,
						carrier_freq_hz: 38000,
						carrier_duty_percent: 33,
						carrier_level: 1,
						carrier_en: 1,
						idle_level: 0,
						idle_output_en: 0
					}});
```

RMT module simply wraps Esp32 structure,  it is straightforward to use.
I think it is ok.

phoddie, 2021-03-03:
That's excellent news.

Yea, it is a very straightforward approach to supporting the ESP32 feature. It doesn't event attempt to be portable. So, it is not applicable to our Raspberry Pi Pico work, for example. I think that's ok though. MCUs are different and that can't always be hidden. 

Should we close this issue out then?

stc1988, 2021-03-03:
Yes, I close this issue.

closed 2021-03-03 by stc1988