Moddable-OpenSource/moddable #592: SIGSEGV in fxMarkWeakMapTable when running test-loan-e2e.js

dckc, 2021-03-02:
This is a preliminary report in case a stack trace is sufficient for you to address this:
https://gist.github.com/dckc/8f97561e5abbc30909c0ef7f328780e5

**Steps to Reproduce** (IOU)

I'm trying to run https://github.com/Agoric/agoric-sdk/blob/master/packages/zoe/test/unitTests/contracts/loan/test-loan-e2e.js on XS. _IOU details on how I'm doing that..._

**Build environment:** Linux
**Target device:** [Agoric's xsnap](https://github.com/Agoric/agoric-sdk/tree/master/packages/xsnap)

cc @kriskowal @michaelfig

phoddie, 2021-03-02:
We'll need a way to reproduce this.

dckc, 2021-03-03:
Ok... I started from scratch and reproduced it; I hope this session log is clear enough:

[repro.txt](https://github.com/Moddable-OpenSource/moddable/files/6072561/repro.txt)

And here's a full copy of `/tmp/moddable-592` after:

https://drive.google.com/file/d/1_uw-d3-09xcgfntebvm0Hbio_rb1WKKq/view?usp=sharing

dckc, 2021-03-04:
Was the info I gave sufficient to reproduce the problem?

phoddie, 2021-03-04:
Sorry, I haven't had a chance to try yet.

phoddie, 2021-03-10:
This seems better?

```
yarn run v1.22.5
$ ava-xs --debug test/unitTests/contracts/loan/test-loan-e2e.js
[Object: null prototype] {}
# test script: test/unitTests/contracts/loan/test-loan-e2e.js
Removing intrinsics.%ErrorPrototype%.stack
# exception: Object.prototype.__lookupGetter__: cannot coerce undefined to object!
test/unitTests/contracts/loan/test-loan-e2e.js: loan - lend - exit before borrow start
&& task was past its deadline when scheduled: 0 &&
@@ schedule task for:1, currently: 0 @@
test/unitTests/contracts/loan/test-loan-e2e.js: loan - lend - exit before borrow end
. test/unitTests/contracts/loan/test-loan-e2e.js loan - lend - exit before borrow
1 tests passed
✨  Done in 5.24s.
```

dckc, 2021-03-10:
Yes, that looks better... so you diagnosed a problem and found a (tentative) fix? I look forward to it!

phoddie, 2021-03-10:
Yes, there is a bug when a weak map contains a weak map. We should have a fix up later for you  to try.

dckc, 2021-03-12:
`test-loan-e2e.js` passes now (yay!)

But our `test-xsnap.js` hangs now :-/

stay tuned for details...

dckc, 2021-03-12:
another crash... I'll open a separate issue on this one tomorrow...
https://gist.github.com/dckc/f44ee8961b4caca9205b500f0c342810

phoddie, 2021-03-12:
Fun.

If this issue is resolved (it sounds like it), please close it out.

dckc, 2021-03-12:
Will do as soon as I can confirm there isn't a regression that causes `test-xsnap.js` to hang. I'm struggling to figure out what's going on.

closed 2021-03-12 by dckc