Moddable-OpenSource/moddable #598: How do you specify a module location?

dbhaig, 2021-03-09:
Do you have questions about

When I try to run the examples/base/instrumentation/instrumentation xsbug displays an error: module "instrumentation" not found

I'm assuming that there is something missing from the manifest.json file but I am at a loss as to what should be added to the file and the explanation behind it.

Thanks

Feel free to ask us here!

You can also ping us in our [chatroom on Gitter](https://gitter.im/embedded-javascript/moddable). This is a great option if you have a project you'd like to talk about or want to have a longer discussion.

dbhaig, 2021-03-12:
I can confirm that both instrumentation examples now compile and run. Getting output from xsuse is still a bit of a mystery to me but I assume that all will become clear eventually.

closed 2021-03-12 by dbhaig

phoddie, 2021-03-12:
Thanks for confirming. Understanding instrumentation in general, and `xsuse` isn particular, can be a bit challenging as there are providing information about the internal state of the engine.