Moddable-OpenSource/moddable #604: SIGSEGV crash in fxRunID running agoric zoe fakePriceAuthority, coveredCall tests

dckc, 2021-03-15:
**Build environment:** Linux
**Target device:** xsnap

**Description**

A handful of zoe tests are crashing on XS / xsnap.

**Steps to Reproduce**

1. check out https://github.com/Agoric/agoric-sdk/ version 6a3c39c43fc0e2dbe104cd2da4e969cb0643eb9c
2. `cd agoric-sdk; yarn install && yarn build; cd packages/zoe`
3. `yarn test:xs-unit --debug test/unitTests/contracts/test-coveredCall.js
3. See `Error#1: <unnamed xsnap worker> exited due to signal SIGSEGV`

full backtrace for today's `coveredCall` crash: https://gist.github.com/dckc/05b998a67f704f455852ad2de167d70c
I have a core file too.

and earlier `fakePriceAuthority` crash: https://gist.github.com/dckc/f44ee8961b4caca9205b500f0c342810

**Expected behavior**

test passes

**Other information**

cc @phoddie @kriskowal @michaelfig @katelynsills

phoddie, 2021-03-16:
Patrick tried to reproduce this and failed. He tried on macOS and Linux. Below are the logs from Linux.

These tests were run with agoric-sdk master branch top of tree and the latest Moddable SDK.

-----

- For test-coveredCall.js

```zoe$ yarn test:xs-unit --debug test/unitTests/contracts/test-coveredCall.js
yarn run v1.22.5
$ ava-xs --verbose --debug test/unitTests/contracts/test-coveredCall.js
[Object: null prototype] {}
# test script: test/unitTests/contracts/test-coveredCall.js
Removing intrinsics.%ErrorPrototype%.stack
# exception: Object.prototype.__lookupGetter__: cannot coerce undefined to object!
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall start
@@ schedule task for:1, currently: 0 @@
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall end
. test/unitTests/contracts/test-coveredCall.js zoe - coveredCall
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall - alice's deadline expires, cancelling alice and bob start
@@ schedule task for:1, currently: 0 @@
@@ tick:1 @@
&& running a task scheduled for 1. &&
# exception: throw!
# exception: throw!
# exception: throw!
Swap failed. Please make sure your offer has the same underlyingAssets and strikePrice as specified in the invitation details. The keywords should not matter.
# exception: throw!
# exception: throw!
# exception: throw!
# exception: throw!
# exception: throw!
# exception: throw!
# exception: throw!
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall - alice's deadline expires, cancelling alice and bob end
. test/unitTests/contracts/test-coveredCall.js zoe - coveredCall - alice's deadline expires, cancelling alice and bob
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall with swap for invitation start
@@ schedule task for:100, currently: 0 @@
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall with swap for invitation end
. test/unitTests/contracts/test-coveredCall.js zoe - coveredCall with swap for invitation
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall with coveredCall for invitation start
@@ schedule task for:100, currently: 0 @@
@@ schedule task for:100, currently: 0 @@
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall with coveredCall for invitation end
. test/unitTests/contracts/test-coveredCall.js zoe - coveredCall with coveredCall for invitation
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall non-fungible start
@@ schedule task for:1, currently: 0 @@
test/unitTests/contracts/test-coveredCall.js: zoe - coveredCall non-fungible end
. test/unitTests/contracts/test-coveredCall.js zoe - coveredCall non-fungible
5 tests passed
Done in 18.54s.
```

-----

- For test-fakePriceAuthority.js

```zoe$ yarn test:xs-unit --debug test/unitTests/test-fakePriceAuthority.js
yarn run v1.22.5
$ ava-xs --verbose --debug test/unitTests/test-fakePriceAuthority.js
[Object: null prototype] {}
# test script: test/unitTests/test-fakePriceAuthority.js
Removing intrinsics.%ErrorPrototype%.stack
# exception: Object.prototype.__lookupGetter__: cannot coerce undefined to object!
test/unitTests/test-fakePriceAuthority.js: priceAuthority quoteAtTime start
&& task was past its deadline when scheduled: 0 &&
@@ schedule task for:1, currently: 0 @@
@@ tick:1 @@
&& running a task scheduled for 1. &&
@@ schedule task for:3, currently: 1 @@
[... 85 lines cut ...]
@@ schedule task for:4, currently: 3 @@
@@ tick:4 @@
&& running a task scheduled for 4. &&
@@ schedule task for:5, currently: 4 @@
test/unitTests/test-fakePriceAuthority.js: priceAuthority quoteWhenGT end
. test/unitTests/test-fakePriceAuthority.js priceAuthority quoteWhenGT
test/unitTests/test-fakePriceAuthority.js: priceAuthority quoteWhenLTE start
&& task was past its deadline when scheduled: 0 &&
@@ schedule task for:1, currently: 0 @@
@@ tick:1 @@
&& running a task scheduled for 1. &&
@@ schedule task for:2, currently: 1 @@
@@ tick:2 @@
&& running a task scheduled for 2. &&
@@ schedule task for:3, currently: 2 @@
@@ tick:3 @@
&& running a task scheduled for 3. &&
@@ schedule task for:4, currently: 3 @@
@@ tick:4 @@
&& running a task scheduled for 4. &&
@@ schedule task for:5, currently: 4 @@
test/unitTests/test-fakePriceAuthority.js: priceAuthority quoteWhenLTE end
. test/unitTests/test-fakePriceAuthority.js priceAuthority quoteWhenLTE
8 tests passed
Done in 8.47s.```

dckc, 2021-03-16:
broken appliance always works for the repair man :)

These things have been intermittent. My attempt to reproduce it once more just now failed. I'll try once or twice more...

phoddie, 2021-03-17:
> broken appliance always works for the repair man :)

Alas, yes.

Let me know what you find. Some hint would help. Then, we can take another look.

dckc, 2021-03-17:
Currently I have a list of tests to exclude from XS test runs. I tried taking these two and a couple others out of that list (https://github.com/Agoric/agoric-sdk/pull/2668 ) but now another one crashes in ci:

```
# test script: test/unitTests/test-scriptedOracle.js
Removing intrinsics.%ErrorPrototype%.stack
(Error#1)
Error#1: <unnamed xsnap worker> exited due to signal SIGSEGV
```
https://github.com/Agoric/agoric-sdk/pull/2668/checks?check_run_id=2127334385#step:7:765

I've been going round and round with intermittent crashes like this for a week or two. I'm not sure how to approach it.

I don't know whether `test-scriptedOracle.js` crashed in `fxRunID` like these other two did.

phoddie, 2021-03-17:
Ugly. Maybe we should try running with the Address Sanitizer? That sometimes isolates problems that only crash occasionally.

dckc, 2021-03-17:
I don't know how to do that. Is there documentation somewhere? Or is it something you can do? Maybe we can talk tomorrow?

phoddie, 2021-03-17:
(checks notes) To enable the addresss sanitizer on a macOS build of xst, I add these to `xst.mak`:

```
C_OPTIONS += -fsanitize=address -fno-omit-frame-pointer

LINK_OPTIONS += -fsanitize=address -fno-omit-frame-pointer
```

I **think** you can do the same for `xsnap.mk`. It may be different on Linux. I recall it took a little trial and error to find the correct settings on macOS. The address sanitizer is in llvm, so I don't believe it is an option if you build with GCC. Here's the [docs](https://clang.llvm.org/docs/AddressSanitizer.html).

dckc, 2021-03-17:
I don't see address sanitizer complaints as I'm running the zoe tests (still running...).

Running `yarn test` in `agoric-sdk/packages/xsnap` gripes about memory leaks, but those shouldn't lead to a SIGSEGV.

```
Direct leak of 32 byte(s) in 1 object(s) allocated from:
    #0 0x4ebb70  (/home/connolly/projects/agoric/agoric-sdk/packages/xsnap/build/bin/lin/debug/xsnap+0x4ebb70)
    #1 0x7fbe8c8a1514  (/lib/x86_64-linux-gnu/libpthread.so.0+0x9514)
```

phoddie, 2021-03-17:
It does look like you appear to be running with the sanitizer, so that's great. Unless it leads to performance issues you might just keep that enabled.

But... if it doesn't catch a problem, the mystery just deepens.

dckc, 2021-03-19:
These symptoms persist, itermittently, as of 7abc931d7 .

I managed to make a trace with [rr](https://en.wikipedia.org/wiki/Rr_(debugging)), a time-travel debugger. This is supposed to support stepping backwards from the crash... I only explored as far as invoking `rr reply` and getting a gdb prompt, though.

[xsnap-604.zip](https://drive.google.com/file/d/13tzyXIdbsA5BM4PKMuK7i2l6oibedSqA/view?usp=sharing)

cc @warner @michaelfig @kriskowal

dckc, 2021-03-20:
yeah... the `rr` trace isn't as helpful as I had hoped... We tried to `rr replay` it on @warner 's desk without luck. Sigh.

phoddie, 2021-03-23:
From the earlier crash log  you shared (coveredCall-xs-crash.txt), it looks like one off the addresses passed to c_memcpy is bad. That suggests something is wrong with the XS chunk heap. That's a lot of territory to cover.

In thinking about how to track that down, a suggestion from @natashenka a while back. She observed that the blocks in the XS chunk heap are is invisible to the Address Sanitizer because it is a single block with allocations managed internally by XS. She suggested a debug implementation of the chunk allocator that used `malloc` for allocations instead. That would make the blocks visible to the Address Sanitizer so it could detect invalid accesses. It seems possible to implement this, though it is a little tricky because the heap structure is intertwined with the garbage collector. Anyway.. something to consider trying if/when we are out of other ideas to try here.

dckc, 2021-03-31:
FYI: the symptoms of this had subsided a bit, but when I try to un-exclude `test-offerSafety.js`, I get a crash... slightly different from the others, I think: https://gist.github.com/dckc/6326f06d880ece0866e4e798a24ccaa9

dckc, 2021-04-03:
## SEGVs with mxNoChunks + ASAN

I haven't gotten to the zoe tests with ASAN + mxNoChunks yet; I got failures in `packages/xsnap`. With `mxNoChunks` set to 1 and the address sanitizer, I get a few SEGVs; for example in our **write and read snapshot** test in `packages/xsnap`:

```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==21376==ERROR: AddressSanitizer: SEGV on unknown address 0x000000045048 (pc 0x0000006dd72f bp 0x7fff202cbf80 sp 0x7fff202cbe70 T0)
==21376==The signal is caused by a READ memory access.
    #0 0x6dd72e  (/home/connolly/projects/agoric/agoric-sdk/packages/xsnap/build/bin/lin/debug/xsnap+0x6dd72e)
    #1 0x538c24  (/home/connolly/projects/agoric/agoric-sdk/packages/xsnap/build/bin/lin/debug/xsnap+0x538c24)
    #2 0x8cb7cf  (/home/connolly/projects/agoric/agoric-sdk/packages/xsnap/build/bin/lin/debug/xsnap+0x8cb7cf)
    #3 0x7fe6bbe31bf6  (/lib/x86_64-linux-gnu/libc.so.6+0x21bf6)
    #4 0x42b859  (/home/connolly/projects/agoric/agoric-sdk/packages/xsnap/build/bin/lin/debug/xsnap+0x42b859)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/connolly/projects/agoric/agoric-sdk/packages/xsnap/build/bin/lin/debug/xsnap+0x6dd72e) 
==21376==ABORTING
  ✖ xsnap › write and read snapshot Rejected promise returned by test
```

This is in the output of `XSNAP_DEBUG=1 yarn test` in `packages/xsnap` with agoric-sdk version 2605-zoe-xs-tests 35f358c08d48 plus diffs below and moddable version e9742b49d51  Fri Apr 2 11:48:28 2021 -0700 plus diffs below.

The test uses our xsnap.js layer to spawn xsnap:

```
test('write and read snapshot', async t => {
  const messages = [];
  async function handleCommand(message) {
    messages.push(decoder.decode(message));
    return new Uint8Array();
  }

  const snapshot = new URL('fixture-snapshot.xss', importMetaUrl).pathname;

  const vat0 = xsnap({ ...xsnapOptions, handleCommand });
  await vat0.evaluate(`
    globalThis.hello = "Hello, World!";
  `);
  await vat0.snapshot(snapshot);
  await vat0.close();

  const vat1 = xsnap({ ...xsnapOptions, handleCommand, snapshot });
  await vat1.evaluate(`
    issueCommand(ArrayBuffer.fromString(hello));
  `);
  await vat1.close();

  t.deepEqual(['Hello, World!'], messages);
});
```

Full output from `yarn test` in `packages/xsnap` with and without `XSNAP_DEBUG=1`:

[,test-results-debug.txt](https://github.com/Moddable-OpenSource/moddable/files/6253131/test-results-debug.txt)
[,test-results.txt](https://github.com/Moddable-OpenSource/moddable/files/6253130/test-results.txt)

### agoric-sdk diff to turn on ASAN and xsnap debug

```diff
12:33 connolly@jambox$ git diff
diff --git a/packages/xsnap/makefiles/lin/xsnap.mk b/packages/xsnap/makefiles/lin/xsnap.mk
index 5168b8538..be97c3978 100644
--- a/packages/xsnap/makefiles/lin/xsnap.mk
+++ b/packages/xsnap/makefiles/lin/xsnap.mk
@@ -38,7 +38,6 @@ C_OPTIONS = \
        -I$(TLS_DIR) \
        -I$(TMP_DIR)
 C_OPTIONS += \
-       -Wno-misleading-indentation \
        -Wno-implicit-fallthrough
 ifeq ($(GOAL),debug)
        C_OPTIONS += -DmxDebug=1 -g -O0 -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter
@@ -50,6 +49,13 @@ LIBRARIES = -ldl -lm -lpthread
 
 LINK_OPTIONS = -rdynamic
 
+# Address Sanitizer
+# https://github.com/Moddable-OpenSource/moddable/issues/604#issuecomment-800771700
+C_OPTIONS += -fsanitize=address -fno-omit-frame-pointer
+CC=clang
+
+LINK_OPTIONS += -fsanitize=address -fno-omit-frame-pointer
+
 OBJECTS = \
        $(TMP_DIR)/xsAll.o \
        $(TMP_DIR)/xsAPI.o \
diff --git a/packages/xsnap/moddable b/packages/xsnap/moddable
--- a/packages/xsnap/moddable
+++ b/packages/xsnap/moddable
@@ -1 +1 @@
-Subproject commit e9742b49d5132f2723d3a5fca45b65cf3818d01a
+Subproject commit e9742b49d5132f2723d3a5fca45b65cf3818d01a-dirty
diff --git a/packages/xsnap/test/test-xsnap.js b/packages/xsnap/test/test-xsnap.js
index 8758e54fc..ca8a25fbf 100644
--- a/packages/xsnap/test/test-xsnap.js
+++ b/packages/xsnap/test/test-xsnap.js
@@ -1,4 +1,4 @@
-/* global setTimeout, __filename */
+/* global setTimeout, __filename, process */
 // eslint-disable-next-line import/no-extraneous-dependencies
 import test from 'ava';
 import * as childProcess from 'child_process';
@@ -16,6 +16,7 @@ const xsnapOptions = {
   os: os.type(),
   stderr: 'inherit',
   stdout: 'inherit',
+  debug: process.env.XSNAP_DEBUG,
 };
 
 function options() {
```

### XS diff to turn on `mxNoChunks`

```diff
12:33 connolly@jambox$ git diff
diff --git a/xs/sources/xsMemory.c b/xs/sources/xsMemory.c
index 70a9288b..57212aa2 100644
--- a/xs/sources/xsMemory.c
+++ b/xs/sources/xsMemory.c
@@ -47,7 +47,7 @@
 #define mxFill 0
 #endif
 #ifndef mxNoChunks
-#define mxNoChunks 0
+#define mxNoChunks 1
 #endif
 
 #if mxStress
~/projects/agoric/agoric-sdk/packages/xsnap/moddable
```

dckc, 2021-04-03:
## zoe tests pass with flying colors under ASAN

not a peep. :-/

```
connolly@jambox:~/projects/agoric/agoric-sdk/packages/zoe$ yarn test:xs-unit
...
317 tests passed
Done in 429.03s.
```

dckc, 2021-04-05:
## SEGV in a SES test

I ran into a SEGV trying to run the SES tests on XS; not sure whether this is the same issue...

https://gist.github.com/dckc/29486d322a46618ef191408179a0d189

test/test-global-object-mutability.js

SES vesion: IOU (locally modified)

dckc, 2021-04-06:
## What JS code was running? `mxTrace`

Yesterday, @phoddie and I looked at the crash from [test-testHelpers.js](https://github.com/Agoric/agoric-sdk/blob/master/packages/zoe/test/unitTests/test-testHelpers.js) ( https://github.com/Agoric/agoric-sdk/issues/2605#issuecomment-812614258 ).

Using [mxTrace](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/sources/xsRun.c#L40) we found that the last few bytecodes executed before the crash were...

```
479: run_1 3
480: begin_strict
480: reserve 8
488: retrieve 4 [freeze] [declassifiers] [quote] [hiddenDetailsMap]
488: retrieve_target
488: retrieve_this
488: new_local [template] 4
488: new_local [args] 5
488: argument 0
489: var_local_1 4
489: pop
488: arguments 1
489: branch_else_4
488: branch_else_4(Error#1)
Error#1: test-testHelpers.js exited due to signal SIGSEGV
```

The identifier `hiddenDetailsMap` helped me find `assert.js`; then `quote`, `freeze`, and `declassifiers` allowed us to zoom in on:

```js
const hiddenDetailsMap = new WeakMap();
...
const unredactedDetails = (template, ...args) => {
  const detailsToken = freeze({ __proto__: null });
  args = args.map(arg => (declassifiers.has(arg) ? arg : quote(arg)));
  hiddenDetailsMap.set(detailsToken, { template, args });
  return detailsToken;
};
freeze(unredactedDetails);
export { unredactedDetails };
```
 -- https://github.com/endojs/endo/blob/master/packages/ses/src/error/assert.js#L54-L98

### Something wrong with `XS_CODE_BRANCH_ELSE_4`?

We looked at `branch_else_4` for a bit... that's a fairly rare opcode for when a relative branch needs a full 32 bit offset. Reverse-stepping from the crash on line 1431 of xsRun.c ( ef5216cc8bf ) had led us to the same opcode. But the `offset` seemed to be complete garbage, suggesting some form of corruption. To check for a logic bug in `XS_CODE_BRANCH_ELSE_4`, we set a breakpoint at 1402, but the crash didn't happen on the first time thru that opcode.

### A closer look at `XS_CODE_ARGUMENTS`

@phoddie looked at other relevant parts of the interpreter. The `XS_CODE_ARGUMENTS` opcode would be used for the spread operator, to turn the arguments into an array. There he saw a call to `mxBehaviorSetProperty`, which may allocate and trigger gc. Any such calls from the bytecode loop are usually surrounded by `mxSaveState` / `mxRestoreState` to restore state such as `code`, a register-allocated pointer to the beginning of the chunk of bytecode being executed, in case the chunk is moved.

After adding save / restore around this call to `mxBehaviorSetProperty`, the SEGV symptoms went away -- at least in this case.

### Intermittent failure due to addresses in WeakMap hashes?

We're still a bit uneasy with the intermittent nature of the symptoms. `hiddenDetailsMap` is a `WeakMap`... perhaps `WeakMap` hashing includes addresses as inputs?

I'll leave this open until Moddable completes their review of #624 .

cc @warner @kriskowal @michaelfig @erights @dtribble

dckc, 2021-04-09:
verified; this crash is fixed by 5dab47a

closed 2021-04-09 by dckc