Moddable-OpenSource/moddable #605: Partitions

mauroForlimpopoli, 2021-03-15:
I  changed the partitions in my Partitions.csv.
In my manifest I added the root "mod" :
		"esp32": {
			"config": {
				"rotation": 90,
				"brightness": 0,
				"file": {
					"root": "/mod/"
				}
			},
			"modules": {
				"*": "$(MODULES)/pins/pwm/esp32/*"
			}
		}

and :
	"build": {
		"PARTITIONS_FILE": "./partitions.csv"
	},

but in the main the instrunction :
let file = new File(root + "data.txt", true);
gives the following error in the XSBUG:
Failed to allocate memory for filesystem
C:\Users\fly3\projects\moddable\modules\files\file\esp32\modFile.c (111) # Break: File: file not found!

	xsUnknownError("file not found");

arg(0) ='/mod/data.txt'

This is the definition of root:
const root = config.file.root;

What I need to

mauroForlimpopoli, 2021-03-15:
do?

mauroForlimpopoli, 2021-03-15:
....and this error in the debugger:
Failed to allocate memory for filesystem
C:\Users\fly3\projects\moddable\modules\files\file\esp32\modFile.c (392) # Break: (host): system info failed!

xsUnknownError("system info failed");

if in the main there is this instruction:
	let info = System.info();
Thank you

andycarle, 2021-03-16:
Hi @mauroForlimpopoli,

Sorry to hear that you're having trouble with the Files module. Let's get that sorted out.

I'd like to be able to reproduce as much of what you're seeing as I can. Could you please share with me:
 
1. Your full `manifest.json` file.
2. Your custom `partitions.csv` file.

I believe that the error that you're seeing (`Failed to allocate memory for filesystem`) should only be printed in situations where the ESP-IDF failed to allocate memory for the filesystem-related structures. If you set a breakpoint in xsbug just before you call `new File`, what does your "System bytes free" look like?

&nbsp;- Andy

mauroForlimpopoli, 2021-03-16:
system byte free = 1996 byte
before 
	let file = new File(root + "data.txt", true);

[partitions.zip](https://github.com/Moddable-OpenSource/moddable/files/6145370/partitions.zip)

the partition file must be changed again to implement OTA function

andycarle, 2021-03-16:
> system byte free = 1996 byte

Yeah, that's not quite enough free memory, I think. 

I just set breakpoints in our [Files example](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/files/files/main.js) before and after Line 23 (`file = new File(root + "test.txt", true);`). In that example, running on an ESP32-based Moddable Two, I'm showing:

System bytes free before Line 23: 138,856
System bytes free after Line 23: 135,940

So, creating the file system is using about 2,916 bytes of system memory. 

You'll need to use a bit less memory elsewhere if you want to use the filesystem. Or use a device with additional external RAM.

&nbsp;- Andy

andycarle, 2021-03-16:
This is somewhat off-topic, but I did just review your `manifest.json`. The number of keys that you're allocating in your manifest is substantial and contributes a lot to your memory constraints:

```json
"creation":{
		"stack":400,
		 "static": 98304,
		 "keys": {
			"available":16386
		}
	}
```

That 16,386 keys is taking up about 65 KB of RAM and is *quite* unusual for apps in the Moddable SDK.

mauroForlimpopoli, 2021-03-16:
I changed the number of keys, and now it works correctly.
Thank you Andy,

andycarle, 2021-03-16:
Great! Glad to hear that it's working for you. 

In case you need it, there is a bit of information about the `creation` `keys` property in the [Manifest documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/manifest.md#creation) (which itself links to a relevant section in the [XS in C documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md#machine-allocation)) and a relevant note in the [Mods documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/mods.md#keys).

andycarle, 2021-03-16:
I'm going to go ahead and close this out. Please feel free to re-open this if there are related problems or to open up a new issue in the future.

Thanks!
&nbsp;- Andy

closed 2021-03-16 by andycarle