Moddable-OpenSource/moddable #606: Error when building command line tools, simulator and debugger on Raspbian Buster

dbhaig, 2021-03-16:
cd $MODDABLE/build/makefiles/lin
make

make -f /home/don/moddable/xs/makefiles/lin/xsc.mk
make[1]: Entering directory '/home/don/moddable/build/makefiles/lin'
# xsc debug : cc xsBigInt.c
# xsc debug : cc xsCode.c
# xsc debug : cc xsCommon.c
# xsc debug : cc xsdtoa.c
# xsc debug : cc xsLexical.c
# xsc debug : cc xsre.c
# xsc debug : cc xsScope.c
# xsc debug : cc xsScript.c
# xsc debug : cc xsSourceMap.c
# xsc debug : cc xsSyntaxical.c
# xsc debug : cc xsTree.c
# xsc debug : cc xsc.c
# xsc debug : cc xsc
/usr/bin/ld: /home/don/moddable/build/tmp/lin/debug/xsc/xsScript.o: undefined reference to symbol 'pthread_getattr_np@@GLIBC_2.4'
/usr/bin/ld: //lib/arm-linux-gnueabihf/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[1]: *** [/home/don/moddable/xs/makefiles/lin/xsc.mk:105: /home/don/moddable/build/bin/lin/debug/xsc] Error 1
make[1]: Leaving directory '/home/don/moddable/build/makefiles/lin'
make: *** [makefile:37: debug] Error 2

more /proc/version
Linux version 5.10.11-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1399 SMP Thu Jan 28 12:09:48 GMT 2021

mkellner, 2021-03-16:
Thank you for the report. 
The tree has been updated to fix this.

dbhaig, 2021-03-16:
Fix confirmed. 
Thanks

closed 2021-03-16 by dbhaig