Moddable-OpenSource/moddable #608: WeakMap loses an entry in GC? Agoric zoe test-zcf.js failing

dckc, 2021-03-17:
**Build environment:** Linux
**Target device:** xsnap

**Description**

One of our tests is failing; if we replace a `WeakMap` with a regular `Map` it doesn't fail, so we suspect a bug around GC and WeakMaps . In https://github.com/Agoric/agoric-sdk/issues/2670 , @warner reduced it to 24 lines (inluding a couple imports):

```js
// eslint-disable-next-line import/no-extraneous-dependencies
import '@agoric/zoe/tools/prepare-test-env';
// eslint-disable-next-line import/no-extraneous-dependencies
import test from 'ava';

function setup(wm1, wm2) {
  const key1 = harden({});
  const key2 = harden({});
  wm1.set(key1, key2);
  wm2.set(key2, 'target');
  return key1;
}

test(`weakmap`, async t => {
  const wm1 = new WeakMap();
  const wm2 = new WeakMap();
  const key1 = setup(wm1, wm2);

  // try to force a GC
  for (let i=0; i < 10000; i++) {
    const o = harden({});
  }
  t.is(wm2.get(wm1.get(key1)), 'target');
});
```

**Steps to Reproduce**

These are automated in a Makefile in a gist. Just grab the 2 files here and run `make` on a linux box.
https://gist.github.com/dckc/7aaa585903730744bb92a64950116fd8

1. clone and build the agoric-sdk
2. copy `test-weakmap.js` into `packages/zoe/test`
3. run that test $(YARN) test:xs-unit test/test-weakmap.js
4. watch it fail

**Expected behavior**
test passes

warner, 2021-03-17:
Note that you can omit all three `harden` calls if you increase the loop count to a million.

The `prepare-test-env` is using the SES shim to create a SES environment, and adding a few objects to it (named `makeKind` and `makeWeakStore`).. I would be surprised if those added objects are significant.

The next step would be to reduce this to a program that runs directly under `xsnap`, rather than including our "run AVA-style unit tests in an `xsnap` child process" framework, but we didn't get that far.

phoddie, 2021-03-17:
Interesting. We'll take a look.

FWIW - the unmodified xsnap sources have a `gc` global function which will avoid the need to coax the collector into running.

phoddie, 2021-03-18:
We expect to have a solution for this ready for you to try Friday.

dckc, 2021-03-18:
thanks for the update!

dckc, 2021-03-19:
> FWIW - the unmodified xsnap sources have a `gc` global function which will avoid the need to coax the collector into running.

right; we turned off some things until we need them... @warner is re-enabling it in https://github.com/Agoric/agoric-sdk/pull/2683

dckc, 2021-03-19:
ooh... new bits... gotta try it out...

dckc, 2021-03-19:
confirmed: fixed!

```
~/projects/agoric/agoric-sdk/packages/zoe
14:44 connolly@jambox$ yarn test:xs-unit test/unitTests/test-weakmap.js 
yarn run v1.22.5
$ ava-xs --verbose test/unitTests/test-weakmap.js
[Object: null prototype] {}
# test script: test/unitTests/test-weakmap.js
Removing intrinsics.%ErrorPrototype%.stack
. test/unitTests/test-weakmap.js weakmap
1 tests passed
Done in 1.59s.
```

closed 2021-03-19 by dckc