Moddable-OpenSource/moddable #621: Symbol.for('bar').toString() missing 'bar' (missing [[Description]] value)

dckc, 2021-04-02:
**Steps to Reproduce**
```
xst -e "print(Symbol.for('bar').toString())"
Symbol()
```

**Expected behavior**

`Symbol(bar)`

**Other information**

ref https://tc39.es/ecma262/#sec-symbol.prototype.tostring

context: `q as best efforts stringify` test
https://github.com/endojs/endo/blob/master/packages/ses/test/error/test-assert-log.js#L421
https://github.com/endojs/endo/issues/508

cc @erights @kriskowal

phoddie, 2021-04-02:
Interesting find. A quick fix is to modify `fxSymbolToString`:

```c
void fxSymbolToString(txMachine* the, txSlot* slot)
{
	txSlot* key = fxGetKey(the, slot->value.symbol);
	fxStringX(the, slot, "Symbol(");
	if (key) {
		if ((key->kind == XS_STRING_KIND) || (key->kind == XS_STRING_X_KIND))
			fxConcatStringC(the, slot, key->value.string);
		else if ((key->kind == XS_KEY_KIND) || (key->kind == XS_KEY_X_KIND))
			fxConcatStringC(the, slot, key->value.key.string);
	}
	fxConcatStringC(the, slot, ")");
}
```

A more careful review and a run through test262 will be needed for a real fix.

phoddie, 2021-04-02:
(Apparently test262 didn't report this? If that's true, we should let the maintainers of teset262 know.)

erights, 2021-04-02:
Report it at https://github.com/tc39/test262/issues ?

dckc, 2021-04-14:
fix confirmed.

```
connolly@jambox:~/projects/agoric/endo-mod-up/packages/ses$ yarn test:xs test/error/test-assert-log.js 
yarn run v1.22.0
$ ava-xs test/error/test-assert-log.js
12 tests passed
Done in 1.40s.
```

closed 2021-04-14 by phoddie