Moddable-OpenSource/moddable #622: Question about custom serialization formats

danfinlay, 2021-04-03:
Hi, I'm a friend of Agoric, playing with some of their xs-interfacing tools like xsnap, and I had a question about the serialization blob output by xs today.

I don't really know anything about how it's done, I just know it seems to be an `.xss` file format, and I have no idea how rewindable or forkable it is, but I am interested in situations where a machine's state might be rewound and resumed with different inputs.

Where in the codebase would I look to learn more about the serialization format, and where one would go to customize it if needed?

phoddie, 2021-04-04:
Hey, @danfinlay. I think you are asking about the snapshot of the virtual machine state written out by `xsnap`? The reading and writing is done by [xsSnapshot.c](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/sources/xsSnapshot.c). The data isn't a serialization that is intended to be useful independent of XS. The data is more-or-less the internal state of XS which is made up of fixed size [slots](https://github.com/Moddable-OpenSource/moddable/blob/e9742b49d5132f2723d3a5fca45b65cf3818d01a/xs/sources/xsAll.h#L314-L340) and variable size chunks. The biggest difference is that pointers are stored as offsets in the snapshot, because the snapshot may be reloaded at a different address. Understanding the snapshot data begins with understanding the internals of XS itself.

The snapshot file organizes the different sections of data using atoms. Here's the atom types inside the root `XS_M` atom:

- `VERS`: The version of XS and the architecture. 
- `SIGN`: The runtime signature. 
- `CREA`: The parameters to create the machine: block initial and incremental size, heap initial and incremental size, stack size, etc.
- `BLOC`: The chunks in the blocks.
- `HEAP`: The slots in the heaps.
- `STAC`: The slots in the stack.
- `KEYS`: The keys table.
- `NAME`: The names table.
- `SYMB`: The symbols table.

danfinlay, 2021-04-04:
Thanks Peter! Yes, you're right, I was asking about the snapshot format. In particular, preparing for scenarios (which I would expect Agoric to also care about), making a blockchain which means many leaders might propose different "next batches of transactions", and then all clients needing to adopt one.

This can mean wanting a lightest-weight-possible "snapshot update" format, like a diff instead of a full state dump each time.

This looks like a good place to start, I'll let you know if (or inevitably when?) I have more questions, thanks! I'll bookmark this as a starting point, you can close this for now if you'd like to keep your inbox tidy.

phoddie, 2021-04-04:
If I understand correctly, you want to keep several branches of patches from a given snapshot. That sounds complicated. ;) If it works, it would be interesting. Is the motivation behind that to conserve storage space that is generated by the snapshots or to perform some kind of analysis on the differences?

danfinlay, 2021-04-04:
There are a couple scenarios I can think of, I think a common theme between them is "when an XS environment is extended to many parties".

- When running a command from an untrusted party that may include an infinite loop, the hypervisor process (like xsnap or swingset) needs to be able to restore the machine state to right before the transaction, maybe recording a penalty for that transaction's proposer.
- If an XS engine were being used to maintain the state of a shared virtual machine where many people might add "blocks", like in the case of the Agoric blockchain, each of those validators needs to be able to receive proposed blocks from other validators, run them on their existing "latest state", to conclude whether they result in the claimed state changes or not, discarding them if they don't, and committing to them of they do. I need to get more acquainted with Agoric's plans, but my guess is their plan is to have occasional XS snapshots, and then otherwise rely on replaying transactions to the machine to wind the machine to the desired step.

You'll probably be seeing some similar "what if XS is run on a public computer" questions from me, as I experiment with how we might use xsnap/swingset at MetaMask.

Maybe worthy of another issue, but kinda thinking about the Agoric blockchain case here, I also suspect at some point there will be a need/benefit to keep the XS heap at least partially on disk (eventually mostly on disk for them), since "the current state" might come to include an entire economy. For example, as of about a year ago [the Ethereum blockchain was about 50 GB](https://blog.ethereum.org/2019/12/10/eth1x-files-fast-sync/).

I suspect these two challenges might have related solutions, but I agree it is likely to be complicated, so that's why I was eager to open the discussion.

danfinlay, 2021-04-04:
@erights just letting you know I found myself asking these questions, maybe Agoric already has ways of approaching these outside of XS, and you could direct me to the right person to discuss how?

erights, 2021-04-04:
Problem foreseen and (mostly) solved: so-called virtual objects, hierarchical c-lists, and virtual collections. Already in use where they are most important: the purses of issuers. The point of these abstractions:
   * The programmer can use them almost as-if these were normal language objects and language collections
   * The vast majority of these objects are idle at any given time. The vast majority of those idle objects it represented in external storage, outside the language's object heap.
   * The size of the language heap is proportional to the number of non-virtual objects + the number of active virtual objects. This should normally remain very small for the uses Agoric has in mind.

@fudco @warner what's the best place to start reading about these abstractions?

phoddie, 2021-04-06:
Closing this for now. @danfinlay, please re-open if/when there is more to cover on this topic,

closed 2021-04-06 by phoddie