Moddable-OpenSource/moddable #626: keyboard

mauroForlimpopoli, 2021-04-07:
This is my keyboard:

const KeyboardContainer = Column.template($ => ({
	left: 0, right: 0, top: 0, bottom: 0, active: true, Skin: WhiteSkin,
	contents:[
		Label($, {
			anchor: "LABEL", left: 0, right: 100, top: 0, height: 48, 
			string: "", Style: orange28 
		}),
		Label($, {
			anchor: "LABELESC", left: 250, right: 0, top: -48, height: 48, 
			string: "esc", Style: black28 , skin: butBackground, active: true, Behavior: escBehavior
		}),
		Container($, {
			anchor: "KEYBOARD", left: 0, right: 0, top: 0, bottom: 0, 
			contents: [
				Keyboard($, {style: new OpenSans20(), doTransition: false})
			]
		}),
	],
	Behavior: KeyboardContainerBehavior
}));

class KeyboardContainerBehavior extends Behavior {
		onCreate(application, data) {
			this.data = data;
		}
		onKeyUp(application, key,container,data) {	
			
			if (key == BACKSPACE) {				
				theString = theString.slice(0, -1);
			} else if (key == SUBMIT) {
				kbValue = theString
				menuContainer.content(0).content(1).string = theString
				theString = "";
				
			keyboardUp = false;
			} else {
				theString += key;
			}
			if (PASSWORDMODE && theString.length > 0) {
				if (undefined !== timerID) {
					Timer.clear(timerID);
					timerID = undefined;
				}
				if (key != BACKSPACE) {
					application.first.string = "*".repeat(theString.length - 1) + theString.charAt(theString.length - 1);
					timerID = Timer.set(id => {timerID = undefined; application.first.first.string = "*".repeat(theString.length);}, 500);
				} else {
					application.first.string = "*".repeat(theString.length);
				}
			} else {
				application.first.string = theString;
			}
		}
		onKeyboardTransitionFinished(application) {
		}
}
Object.freeze(KeyboardContainerBehavior.prototype);

class escBehavior extends Behavior {
 	onTouchBegan(label) {
		application.remove(kbContainer);	
 	}
}
Object.freeze(escBehavior.prototype);

Sometimes, it gives this error:
"C:\Users\fly3\projects\moddable\modules\piu\MC\piuView.c (1067) # Break: (host): display list overflowed!
It happen only if I type about 17 characters and they are completly in the label. I some of them go outside (es: ejewr....) no error is showed.
Furthermore the compiler says:
### warning: "keyboard": Keyboard() f() KeyboardBehavior.prototype.onCreate() ShiftTexture() it: not frozen
### warning: "keyboard": Keyboard() f() KeyboardBehavior.prototype.onCreate() BackspaceTexture() it: not frozen
### warning: "keyboard": Keyboard() f() KeyboardBehavior.prototype.onCreate() NumKeysTexture() it: not frozen
What can I do to freeze them ?

lprader, 2021-04-07:
Regarding this:

```
C:\Users\fly3\projects\moddable\modules\piu\MC\piuView.c (1067) # Break: (host): display list overflowed!
```

The default display list is 1024 bytes. You can specify the display list length in the dictionary of properties passed into the constructor of the Piu `Application` object. There a few examples that do this, including [balls](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/piu/balls/main.js#L58) and [cards](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/piu/cards/main.js#L159). Here's an example that changes it to be 4096 bytes:

```js
new Application(null, { 
    displayListLength: 4096
});
```

And regarding the warnings about objects not being frozen, you can ignore those. But if you'd like to get rid of them, you just need to change [lines 62-64 of `keyboard.js`](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/input/keyboard/keyboard.js#L62) to the following:

```js
const ShiftTexture = Texture.template(Object.freeze({ path:"shift-arrow.png" }));
const BackspaceTexture = Texture.template(Object.freeze({path:"delete-arrow.png"}));
const NumKeysTexture = Texture.template(Object.freeze({path:"num-keys.png"}));
```

FWIW, there's lots of information about freezing objects in the [Freezing section of `preload.md`](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/preload.md#freezing).

mauroForlimpopoli, 2021-04-07:
...perfect, it works correctly. thank you

closed 2021-04-07 by phoddie