Moddable-OpenSource/moddable #627: SIGSEGV in fxRunID in xsnap when running ses test-global-object-mutability.js test

dckc, 2021-04-09:
**Steps to Reproduce**

https://gist.github.com/dckc/19451649697bb7cb231ab5a776fde0d1 has a `Makefile` that automates these steps as well as a transcript of a run and a backtrace:

1. clone https://github.com/endojs/endo and check out v f84387457a35; cd endo
2. `yarn; yarn build`
3. cd packages/ses; `yarn test:xs --debug test/test-global-object-mutability.js`
   - the `--debug` is optional; it crashes either way. But `--debug` gives a useful backtrace.
4. see `Error#1: test-global-object-mutability.js exited due to signal SIGSEGV`
5. gdb on the resulting `core` file shows that it crashed at xsRun.c:4065 in fxRunID

**Build environment:** Linux
**Target device:** agoric xsnap in the endo repo

**Other information**

This is using the current moddable SDK 0b451c98f92c1637835f6a079bf5b6e41210d438

I noted this in https://github.com/Moddable-OpenSource/moddable/issues/604#issuecomment-813494765 but this looks like a separate issue from #604 .

The ses test is [test-global-object-mutability.js](https://github.com/endojs/endo/blob/master/packages/ses/test/test-global-object-mutability.js).

Getting the SES test suite to run on XS is https://github.com/endojs/endo/issues/508 .

phoddie, 2021-04-09:
@dckc - thanks for writing this up separately. That backtrace has nearly all of our favorite things: Promise, Reflect, async functions, eval, and Proxy. ;) Do you happen to know if this issue happen reliably across your development systems?

dckc, 2021-04-09:
Yes, I just tried it on one other machine (AMD Opteron(tm) Processor 6128 HE w 32GB RAM) and it fails there too.

dckc, 2021-04-11:
Trace info (using `mxTrace`) shows it died in [createScopeHandler](https://github.com/endojs/endo/blob/e9199f41c511b5df10593d931febdd90693b011a/packages/ses/src/scope-handler.js#L47) around one of the calls to `reflectGet`. The `if (prop === 'eval')` test evidently failed, but I'm not sure whether the `if (prop in localObject)` test passed or failed.

1032135 is a line number from `less`; yes, it was >1M bytecodes into the execution when it crashed.

```
1032135 341: retrieve 3 [sloppyGlobalsMode] [localObject] [globalObject]
1032136 341: new_local [_shadow] 3
1032137 341: new_local [prop] 4
1032138 341: argument 0
1032139 342: var_local_1 3
1032140 342: pop
1032141 341: argument 1
1032142 342: var_local_1 4
1032143 342: pop
1032144 341: get_closure 0
1032145 342: dub
1032146 343: branch_if_2
1032147 342: pop
1032148 341: get_local 4
1032149 342: string "eval"
1032150 343: strict_equal
1032151 342: dub
1032152 343: branch_if_2
1032153 342: pop
1032154 341: get_local 4
1032155 342: get_closure 1
1032156 343: in
1032157 342: dub
1032158 343: branch_if_2
1032159 342: pop
1032160 341: get_local 4
1032161 342: get_closure 2
1032162 343: in
1032163 342: dub
1032164 343: branch_if_2
1032165 342: branch_else_2
1032166 341: true
1032167 342: result
1032168 341: end
1032169 328: function_environment(Error#1)
1032170 Error#1: test-global-object-mutability.js exited due to signal SIGSEGV
```

dckc, 2021-04-14:
oh! nice to see you've got a fix I can try.

Unfortunately, adding a label doesn't notify me. In the future, please add a comment and/or tag @dckc when you have something for me to verify.

dckc, 2021-04-14:
fix confirmed. Thanks!

```
connolly@jambox:~/projects/agoric/endo-mod-up/packages/ses$ yarn test:xs test/test-global-object-mutability.js 
yarn run v1.22.0
$ ava-xs test/test-global-object-mutability.js
2 tests passed
Done in 1.76s.
```

closed 2021-04-14 by dckc