Moddable-OpenSource/moddable #633: Windows SDK won't execute module

johnwargo, 2021-04-18:
**Build environment:** Windows 10, fully updated with all latest patches.

Processor	Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz   3.60 GHz
Installed RAM	4.00 GB
System type	64-bit operating system, x64-based processor
Edition	Windows 10 Pro
Version	20H2
Installed on	‎4/‎18/‎2021
OS build	19042.928
Experience	Windows Feature Experience Pack 120.2212.551.0

**Target device:** Moddable Two

**Description**

I recently reinstalled the latest Moddable SDK on my Windows system, but in my testing, I could install a host, but not a module on my device. When I install the host using `mcconfig` the console in `xsbug` shows the output from the host. When I install the module using `mcrun` the `xsbug` console shows that the device rebooted after installing the mod (the host content appears in the console a second time) but no output from the module appears in the console.

Since its possible my development environment has issues, I created a new Windows 10 VM, installed the Moddable SDK and the ESP32 SDK and tested it again with the exact same results.

**Steps to Reproduce**

1. Build and install the Moddable SDK example HelloWorld app using `mcconfig`
2. Build and install the IOT Book Chapter 1 host using `mcconfig`
3. Build and install the IO book Chapter 1 HelloWorld module using `mcrun`

**Expected behavior**

1. hello world app output in console - success
2. host output in console - success
3. module output in console - failure

**Images**
Screenshots or photos to help explain the problem

[image]

**Other information**
Any other details you believe may be helpful in understanding the issue, reproducing the unexpected behavior, or fixing the software

I've confirmed the behavior on two systems, one a complete installation of Windows and all associated tools from scratch.

andycarle, 2021-04-18:
@johnwargo Thank you for all the details; they are appreciated.

We have seen something similar to this before, but only with third-party development boards (mostly the M5Stack devices). In those cases, the issue was a mismatch between the `DEBUGGER_SPEED` that was used to build the host and the `DEBUGGER_SPEED` used when deploying the mod. 

That mismatch should not be an issue with Moddable Two unless the `DEBUGGER_SPEED` environment variable has been manually set. But it's worth checking, as that would definitely produce this result.

Could you do the `mcconfig` and `mcrun` commands each with the `-v` flag to get verbose output? Then post the logs here and I'll take a look. What we're trying to verify is that the `DEBUGGER_SPEED` used in the build looks the same as the `serial2xsbug` command that gets issued by `mcrun`. 

For instance, one line in my host build looks like this, showing the `-DDEBUGGER_SPEED=460800`:

```
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DmxDebug=1 -DSDKCONFIG_HEADER=C:\Users\andyc\Projects\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=VERBOSE -DDEBUGGER_SPEED=460800 -DESP32_SUBCLASS=esp32 -DCCACHE_ENABLE=1 c:\users\andyc\projects\moddable\build\tmp\esp32\moddable_two\debug\xsproj-esp32"...
```

And when I run `mcrun -d -m -p esp32/moddable_two -v` on the `helloworld` book example, the `serial2xsbug` command is:

```
C:\Users\andyc\Projects\moddable\build\bin\win\release\serial2xsbug  460800 8N1 -install C:\Users\andyc\Projects\moddable\build\bin\esp32\debug\helloworld\helloworld.xsa
```

The matching `460800` speeds are what we're looking for. There also may be other hints in your build output, even if those match as expected.

andycarle, 2021-04-18:
Oh, one other idea. I have very occasionally seen issues with mods that can be resolved by simply erasing the flash on the device and starting over. 

That's worth trying:

```
python %IDF_PATH%\components\esptool_py\esptool\esptool.py erase_flash
```

After doing that you will have to install the host again with `mcconfig` before trying `mcrun` on the example.

johnwargo, 2021-04-18:
@andycarle thanks, I'll try both and get back to you, hopefully tonight.

johnwargo, 2021-04-18:
@andycarle erased the flash and re-tested, no change. No workie.

johnwargo, 2021-04-18:
@andycarle `DEBUGGER_SPEED` looks to be the same to me:

## Host

```text
	D:\dev\hardware\moddable\moddable\build\bin\win\release\bles2gatt -s D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\config-esp32\sdkconfig.mc -o D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\ch1host
	tasklist /nh /fi "imagename eq serial2xsbug.exe" | (find /i "serial2xsbug.exe" > nul) && taskkill /f /t /im "serial2xsbug.exe" >nul 2>&1
	tasklist /nh /fi "imagename eq xsbug.exe" | find /i "xsbug.exe" > nul || (start D:\dev\hardware\moddable\moddable\build\bin\win\release\xsbug.exe)
	if exist D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32\xs_esp32.elf del D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32\xs_esp32.elf
	if not exist D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32 mkdir D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32
	copy D:\dev\hardware\moddable\moddable\build\bin\esp32\moddable_two\debug\ch1host\xs_esp32.a D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32\.
        1 file(s) copied.
	cd D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\xsProj-esp32 & python %IDF_PATH%\tools\idf.py -v  -b 921600 -B D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32 build flash -D mxDebug=1 -D SDKCONFIG_HEADER="D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32\config\sdkconfig.h" -D CMAKE_MESSAGE_LOG_LEVEL=VERBOSE -D DEBUGGER_SPEED=460800 -D ESP32_SUBCLASS=esp32
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file D:/dev/hardware/moddable/moddable/build/tmp/esp32/moddable_two/debug/xsProj-esp32/sdkconfig
Loading defaults file D:/dev/hardware/moddable/moddable/build/tmp/esp32/moddable_two/debug/config-esp32/sdkconfig.mc...
-- App "xs_esp32" version: OS210406-13-g15026147
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script D:/dev/hardware/moddable/moddable/build/tmp/esp32/moddable_two/debug/idf-esp32/esp-idf/esp32/esp32_out.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: D:/dev/hardware/esp32/esp-idf/components/app_trace D:/dev/hardware/esp32/esp-idf/components/app_update D:/dev/hardware/esp32/esp-idf/components/asio D:/dev/hardware/esp32/esp-idf/components/bootloader D:/dev/hardware/esp32/esp-idf/components/bootloader_support D:/dev/hardware/esp32/esp-idf/components/bt D:/dev/hardware/esp32/esp-idf/components/cbor D:/dev/hardware/esp32/esp-idf/components/coap D:/dev/hardware/esp32/esp-idf/components/console D:/dev/hardware/esp32/esp-idf/components/cxx D:/dev/hardware/esp32/esp-idf/components/driver D:/dev/hardware/esp32/esp-idf/components/efuse D:/dev/hardware/esp32/esp-idf/components/esp-tls D:/dev/hardware/esp32/esp-idf/components/esp32 D:/dev/hardware/esp32/esp-idf/components/esp_adc_cal D:/dev/hardware/esp32/esp-idf/components/esp_common D:/dev/hardware/esp32/esp-idf/components/esp_eth D:/dev/hardware/esp32/esp-idf/components/esp_event D:/dev/hardware/esp32/esp-idf/components/esp_gdbstub D:/dev/hardware/esp32/esp-idf/components/esp_hid D:/dev/hardware/esp32/esp-idf/components/esp_http_client D:/dev/hardware/esp32/esp-idf/components/esp_http_server D:/dev/hardware/esp32/esp-idf/components/esp_https_ota D:/dev/hardware/esp32/esp-idf/components/esp_https_server D:/dev/hardware/esp32/esp-idf/components/esp_ipc D:/dev/hardware/esp32/esp-idf/components/esp_local_ctrl D:/dev/hardware/esp32/esp-idf/components/esp_netif D:/dev/hardware/esp32/esp-idf/components/esp_ringbuf D:/dev/hardware/esp32/esp-idf/components/esp_rom D:/dev/hardware/esp32/esp-idf/components/esp_serial_slave_link D:/dev/hardware/esp32/esp-idf/components/esp_system D:/dev/hardware/esp32/esp-idf/components/esp_timer D:/dev/hardware/esp32/esp-idf/components/esp_websocket_client D:/dev/hardware/esp32/esp-idf/components/esp_wifi D:/dev/hardware/esp32/esp-idf/components/espcoredump D:/dev/hardwa [... cut ...]
	tasklist /nh /fi "imagename eq serial2xsbug.exe" | (find /i "serial2xsbug.exe" > nul) && taskkill /f /t /im "serial2xsbug.exe" >nul 2>&1
	tasklist /nh /fi "imagename eq xsbug.exe" | find /i "xsbug.exe" > nul || (start D:\dev\hardware\moddable\moddable\build\bin\win\release\xsbug.exe && echo Starting xsbug... && timeout /nobreak /t 7 > nul)
	D:\dev\hardware\moddable\moddable\build\bin\win\release\serial2xsbug COM6 460800 8N1 -install D:\dev\hardware\moddable\moddable\build\bin\esp32\debug\helloworld\helloworld.xsa
Terminate batch job (Y/N)? 
```

johnwargo, 2021-04-18:
Just for grins, I tried this connection on a completely different USB controller. No change. 
I've tried two cables and three USB Ports. I'm trying USB2 ports, I didn't try a USB 3 port

andycarle, 2021-04-19:
Thank you for the additional information. I agree that everything looks like it should be working based on that output. 

Let me do a bit of experimentation to see if I can find any set of conditions that trigger mcrun to fail in this way. And I can put together some changes to mcrun to get some extra debugging information off of your system.

johnwargo, 2021-04-19:
@andycarle OK, thanks, that makes sense. Just let me know what I can do to help.

johnwargo, 2021-04-20:
I just remembered something - don't forget, this is my Windows box that has stuff installed on the C drive but my working development directory is on D. That caused the previous problem I had.

andycarle, 2021-04-21:
Understood; thank you for the reminder. I don't _think_ that's a factor here, but I'll keep an eye on it.

johnwargo, 2021-04-27:
@andycarle sorry to be a pest, but any progress? Anything I can test for you?

andycarle, 2021-04-27:
@johnwargo Apologies, I've been head-down trying to sort out a different problem. I'll put together a couple of test changes for you to try out later this afternoon.

andycarle, 2021-04-28:
Hi @johnwargo,

I've spent some time this evening trying out different sequences and combinations of builds, trying to find a way to get into the state you're seeing. I have gotten into the bad state  a couple of times. But, in each case I was able to fix it by reinstalling the host and the running `mcrun` on the example mod again. 

But, it is suspicious that it happens at all. It makes me wonder if there is some edge case that I'm seeing some small % of the time but that always happens on your PC. More to investigate...

In the meantime, I have one thing you can change on your side to get some additional diagnostic information.

In `%MODDABLE%\tools\serial2xsbug\serial2xsbug.h` [line 91 and 92](https://github.com/Moddable-OpenSource/moddable/blob/4ae91266ad4a9a4370e335d3904834ccccc3c5dc/tools/serial2xsbug/serial2xsbug.h#L91) there are two defines:

```c
#define mxTrace 0
#define mxTraceCommands 0
```

Please change those both to 1:

```c
#define mxTrace 1
#define mxTraceCommands 1
```

And then **rebuild the Moddable SDK tools** before proceeding (an incremental build is sufficient, so just `cd %MODDABLE%\build\makefiles\win` then `build`).

This change will provide quite a lot of additional information at the command line, dumping out all the internal messages between xsbug and the device. What I'm curious to see is the output when running `mcrun -d -m -p esp32/moddable_two` in the `iot-product-dev-book\ch1-gettingstarted\helloworld` directory. 

On a working deployment on my machine, that looks like this:

```
C:\Users\andyc\Projects\iot-product-dev-book\ch1-gettingstarted\helloworld>mcrun -d -m -p esp32/moddable_two
### fxRestart
xÇx°xÇxÇÇxxxx°xÇ<?xs-00000000?>

<?xs.3FFB4748?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB4748?>### install
### install fragment @ 0 size 4

<?xs#3FFB4748?>
<?xs#3FFB4748?>
### fxCommandReceived
### install fragment @ 4 size 512

<?xs#3FFB4748?>☻♥αα
<?xs#3FFB4748?>
### fxCommandReceived
### install fragment @ 516 size 170

<?xs#3FFB4748?>
<?xs#3FFB4748?>
### fxCommandReceived
### update install header
### install fragment @ 4 size 4

<?xs#3FFB4748?>
<?xs#3FFB4748?>
### fxCommandReceived
### install complete
### fxRestart

<?xs#3FFB4748?>xÇx°xÇxÇÇxxxx°xÇ<?xs-00000000?>

<?xs.3FFB4748?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB4748?>
<?xs.3FFB4748?>
<set-all-breakpoints><breakpoint path="exceptions" line="0"/></set-all-breakpoints>

<?xs.3FFB4748?>
<xsbug><instruments><instrument name="Pixels drawn" value=" pixels"/><instrument name="Frames drawn" value=" frames"/><instrument name="Network bytes read" value=" bytes"/><instrument name="Network bytes written" value=" bytes"/><instrument name="Network sockets" value=" sockets"/><instrument name="Timers" value=" timers"/><instrument name="Files" value=" files"/><instrument name="Poco display list used" value=" bytes"/><instrument name="Piu command List used" value=" bytes"/><instrument name="SPI flash erases" value=" sectors"/><instrument name="System bytes free" value=" bytes"/><instrument name="CPU 0" value=" percent"/><instrument name="CPU 1" value=" percent"/><instrument name="Chunk used" value=" / "/><instrument name="Chunk available" value=" bytes"/><instrument name="Slot used" value=" / "/><instrument name="Slot available" value=" bytes"/><instrument name="Stack used" value=" / "/><instrument name="Stack available" value=" bytes"/><instrument name="Garbage collections" value=" times"/><instrument name="Keys used" value=" keys"/><instrument name="Modules loaded" value=" modules"/><instrument name="Parser used" value=" bytes"/><instrument name="Floating Point" value=" operations"/></instruments></xsbug>

<?xs.3FFB4748?>
<xsbug><samples>0,0,0,0,0,0,0,0,0,0,256456,0,10,760,1536,4640,8192,752,4096,1,8,0,0,0</samples></xsbug>

<?xs.3FFB4748?>
<xsbug><frames><frame name="?" value="@3FFC0E70" path="C:\Users\andyc\Projects\iot-product-dev-book\ch1-gettingstarted\helloworld\example.js" line="15"/><frame name="(host)" value="@3FFC0F00"/><frame name="default" value="@3FFC0F90" path="C:\Users\andyc\Projects\iot-product-dev-book\ch1-gettingstarted\host\main.js" line="22"/><frame name="(host)" value="@3FFC1000"/></frames><local name="?" value="@3FFC0E70" path="C:\Users\andyc\Projects\iot-product-dev-book\ch1-gettingstarted\helloworld\example.js" line="15"><property flags=" cew_" name="(return)" value="undefined"/><property flags=" cew_" name="new.target" value="undefined"/><property flags="+cew_" name="(function)" value="@3FFBF7F0"/><property flags=" cew_" name="this" value="undefined"/><property flags=" cEw_" name="message"/></local><global><property flags="+cEw_" name="AggregateError" value="@3F4119E0"/><property flags="+cEw_" name="Array" value="@3F40FBE0"/><property flags="+cEw_" name="ArrayBuffer" value="@3F40FC60"/><property flags="+cEw_" name="Atomics" value="@3F40F290"/><property flags="+cEw_" name="BigInt" value="@3F40FD60"/><property flags="+cEw_" name="BigInt64Array" value="@3F410A40"/><property flags="+cEw_" name="BigUint64Array" value="@3F4108E0"/><property flags="+cEw_" name="Boolean" value="@3F411A60"/><property flags="+cEw_" name="DataView" value="@3F4117E0"/><property flags="+cEw_" name="Date" value="@3F410010"/><property flags="+cEw_" name="Error" value="@3F411A20"/><property flags="+cEw_" name="EvalError" value="@3F4119A0"/><property flags="+cEw_" name="FinalizationRegistry" value="@3F4115A0"/><property flags="+cEw_" name="Float32Array" value="@3F410840"/ [... cut ...] "/><node flags="+CEWM" name="/time.xsb" value="@3F40ED30"/><node flags="+CEWM" name="/timer.xsb" value="@3F40ED20"/><node flags="+CEWM" name="/Resource.xsb" value="@3F40ED00"/></grammar><break path="C:\Users\andyc\Projects\iot-product-dev-book\ch1-gettingstarted\helloworld\example.js" line="15"># Break: debugger!
</break></xsbug>
```

There you can see the mod installing, the device restarting, and the new boot making it to the `helloworld` `debugger` statement.

What do you see on your machine?

johnwargo, 2021-04-28:
@andycarle mine looks pretty similar to yours:

```text
mcrun -d -m -p esp32/moddable_two
### fxRestart
xÇx°xÇxÇÇxxxx°xÇ<?xs-00000000?>

<?xs.3FFB4740?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB4740?>### install
### install fragment @ 0 size 4

<?xs#3FFB4740?>
<?xs#3FFB4740?>
### fxCommandReceived
### install fragment @ 4 size 512

<?xs#3FFB4740?>☻♥αα
<?xs#3FFB4740?>
### fxCommandReceived
### install fragment @ 516 size 214

<?xs#3FFB4740?>
<?xs#3FFB4740?>
### fxCommandReceived
### update install header
### install fragment @ 4 size 4

<?xs#3FFB4740?>
<?xs#3FFB4740?>
### fxCommandReceived
### install complete
### fxRestart

<?xs#3FFB4740?>xÇx°xÇxÇÇxxxx°xÇ<?xs-00000000?>

<?xs.3FFB4740?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB4740?>
<?xs.3FFB4740?>
<set-all-breakpoints><breakpoint path="exceptions" line="0"/></set-all-breakpoints>

<?xs.3FFB4740?>
<xsbug><instruments><instrument name="Pixels drawn" value=" pixels"/><instrument name="Frames drawn" value=" frames"/><instrument name="Network bytes read" value=" bytes"/><instrument name="Network bytes written" value=" bytes"/><instrument name="Network sockets" value=" sockets"/><instrument name="Timers" value=" timers"/><instrument name="Files" value=" files"/><instrument name="Poco display list used" value=" bytes"/><instrument name="Piu command List used" value=" bytes"/><instrument name="SPI flash erases" value=" sectors"/><instrument name="System bytes free" value=" bytes"/><instrument name="CPU 0" value=" percent"/><instrument name="CPU 1" value=" percent"/><instrument name="Chunk used" value=" / "/><instrument name="Chunk available" value=" bytes"/><instrument name="Slot used" value=" / "/><instrument name="Slot available" value=" bytes"/><instrument name="Stack used" value=" / "/><instrument name="Stack available" value=" bytes"/><instrument name="Garbage collections" value=" times"/><instrument name="Keys used" value=" keys"/><instrument name="Modules loaded" value=" modules"/><instrument name="Parser used" value=" bytes"/><instrument name="Floating Point" value=" operations"/></instruments></xsbug>

<?xs.3FFB4740?>
<xsbug><log>Device flashed. Ready to install apps.&#10;</log></xsbug>

<?xs.3FFB4740?>
<xsbug><samples>0,0,0,0,0,0,0,0,0,0,256484,0,3,728,1536,4064,8192,672,4096,0,0,0,0,0</samples></xsbug>

<?xs.3FFB4740?>
<xsbug><samples>0,0,0,0,0,0,0,0,0,0,256484,0,0,728,1536,4064,8192,208,4096,0,0,0,0,0</samples></xsbug>

<?xs.3FFB4740?>
<xsbug><samples>0,0,0,0,0,0,0,0,0,0,256484,0,0,728,1536,4064,8192,208,4096,0,0,0,0,0</samples></xsbug>
NMAKE : fatal error U1058: terminated by user
Stop.^
CNMAKE : fatal erTerminate batch job (Y/N)? ror U1077: 'D:\dev\hardware\moddable\moddable\build\binm
```

andycarle, 2021-04-28:
Yeah, what we're seeing there is that everything looks right during the mod installation, but then it just isn't found for execution.

There would seem to be two possibilities here. Either the installation is not actually succeeding or the detection and mapping of the mod archive is failing on the reboot. 

Let's investigate option 1 first. Could you send me a binary with the contents of the Moddable Two flash after running `mcrun`? 

To get this, just do: `python %IDF_PATH%\components\esptool_py\esptool\esptool.py -b 921800 read_flash 0 0x400000 flash.bin` . That will dump out the full contents of flash to `flash.bin` in the directory where you run this. Then you can zip it up and post it to this issue. 

(This will contain any data you've flashed to the device. If you're just running examples that's presumably okay, but you could always erase before installing the host and mod, just to be sure.)

So, to be clear:

1. Erase flash (optional)
2. Install the host with `mcconfig`
3. Install the example mod with `mcrun`
4. Dump the flash using the above command.

johnwargo, 2021-04-29:
Followed steps 1 through 4, here's the `.bin` file. 
[flash.zip](https://github.com/Moddable-OpenSource/moddable/files/6398084/flash.zip)

johnwargo, 2021-05-05:
@andycarle this one's a stumper, eh?

andycarle, 2021-05-05:
So, funny story: the other day I was investigating this specific issue on my Windows desktop when the video card died, trapping my work in progress. 

[image]

I'll try to get back to the same place on my laptop this evening to get you unstuck.

johnwargo, 2021-05-10:
@andycarle that really sucks about your video card - all good now?

johnwargo, 2021-05-13:
@andycarle any progress?

johnwargo, 2021-05-17:
@andycarle are you still there? Everything OK?

phoddie, 2021-05-18:
@andycarle is alive and well and very busy with a project for a client.

johnwargo, 2021-05-18:
@phoddie thank you very much for letting me know; I figured either a secret project or COVID. Glad to hear he's just busy.

johnwargo, 2021-06-07:
@andycarle any progress on this?

andycarle, 2021-06-17:
@johnwargo Thank you for your concern; I'm alive and well!

@phoddie recently checked in some changes to provide additional debugging information during the mod installation process. Could you please give this a try again and let us know what the new tooling reports back?

To get the expected results, you'll need to fully delete `%MODDABLE%\build\bin` and `%MODDABLE%\build\tmp` and then rebuild the Moddable SDK tools. It probably would also be a good idea to erase your ESP32 first as well.

johnwargo, 2021-06-25:
I deleted the bin and tmp folders as you suggested, built the SDK, then updated the ESP SDK using the instructions on https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md.

Received a new error when I run `mcconfig`:

```text
D:\dev\hardware\moddable\iot-product-dev-book\ch1-gettingstarted\host>mcconfig -d -m -p esp32/moddable_two
FC: cannot open D:\DEV\HARDWARE\MODDABLE\MODDABLE\BUILD\TMP\ESP32\MODDABLE_TWO\DEBUG\XSPROJ-ESP32\PARTITIONS.CSV - No such file or folder
```

It goes through all of its other stuff (build and deploy) but I've never seen that message before. 

I also get an interesting error in xsbug:

```text
Mod failed: fatal
Device flashed. Ready to install apps.
Mod failed: not enough keys
Device flashed. Ready to install apps
```

Here's console output:

```text
D:\dev\hardware\moddable\iot-product-dev-book\ch1-gettingstarted\host>cd ..
D:\dev\hardware\moddable\iot-product-dev-book\ch1-gettingstarted>cd helloworld-gui
D:\dev\hardware\moddable\iot-product-dev-book\ch1-gettingstarted\helloworld-gui>mcrun -d -m -p esp32/moddable_two
# xsc pins\digital.xsb
# xsc ili9341.xsb
# xsc preference.xsb
# xsc ft6206.xsb
# xsc time.xsb
# xsc timer.xsb
# xsc Resource.xsb
# xsc main.xsb
# xsc commodetto\Bitmap.xsb
# xsc pins\i2c.xsb
# xsc pins\pwm.xsb
# xsc pins\digital\monitor.xsb
# xsc button.xsb
# xsc led.xsb
# xsc setup\target.xsb
# xsc commodetto\parseBMF.xsb
# xsc commodetto\parseBMP.xsb
# xsc commodetto\parseRLE.xsb
# xsc commodetto\Poco.xsb
# xsc setup\commodetto.xsb
# xsc check.xsb
# xsc mod\config.xsb
# png2bmp OpenSans-Regular-24-alpha.bmp
# compressbmf OpenSans-Regular-24.bf4
# xsl helloworld-gui.xsa
Installing mod...........complete
```

johnwargo, 2021-06-25:
Here's the complete output:

```text

D:\dev\hardware\moddable\iot-product-dev-book\ch1-gettingstarted\host>mcconfig -d -m -p esp32/moddable_two
FC: cannot open D:\DEV\HARDWARE\MODDABLE\MODDABLE\BUILD\TMP\ESP32\MODDABLE_TWO\DEBUG\XSPROJ-ESP32\PARTITIONS.CSV - No such file or folder

Does D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\xsProj-esp32 specify a file name
or directory name on the target
(F = file, D = directory)? d
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\CMakeLists.txt
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\getPort.py
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\Makefile
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\partitions.csv
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\sdkconfig.defaults
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\sdkconfig.defaults.release
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\sdkconfig.inst
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\sdkconfig.release.part
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\main\CMakeLists.txt
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\main\component.mk
D:\dev\hardware\moddable\moddable\build\devices\esp32\xsProj-esp32\main\main.c
11 File(s) copied
        1 file(s) copied.
Reconfiguring ESP-IDF...
Executing action: reconfigure
Running cmake in directory d:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=D:/dev/hardware/moddable/moddable/build/tmp/esp32/moddable_two/debug/config-esp32/sdkconfig.mc -DSDKCONFIG_HEADER=D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=460800 -DESP32_SUBCLASS=esp32 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 d:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\xsproj-esp32"...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.32.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/john/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/john/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/john/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/john/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/john/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file D:/dev/hardware/moddable/moddable/build/tmp/esp32/moddable_two/debug/xsProj-esp32/sdkconfig
Loading defaults file D:/dev/hardware/moddable/moddable/build/tmp/esp32/moddable_two/debug/config-esp32/sdkconfig.mc...
-- Found PythonInterp: C:/Users/john/.espressif/python_env/idf4.2_py3.9_env/Scripts/python.exe (found version "3.9.5")
-- Found Perl: C:/Program Files/Git/usr/bin/perl.exe (found version "5.32.1")
-- App "xs_esp32" version: OS210603-63-g4e27f761
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script D:/dev/hardware/moddable/moddable/build/tmp/esp32/moddable_two/debug/idf-esp32/esp-idf/esp32/esp32_out.ld
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script D:/dev/hardware/esp32/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs [... cut ...]
# xsc Resource.xsb
# xsc main.xsb
# xsc commodetto\Bitmap.xsb
# xsc pins\i2c.xsb
# xsc pins\pwm.xsb
# xsc pins\digital\monitor.xsb
# xsc button.xsb
# xsc led.xsb
# xsc setup\target.xsb
# xsc commodetto\parseBMF.xsb
# xsc commodetto\parseBMP.xsb
# xsc commodetto\parseRLE.xsb
# xsc commodetto\Poco.xsb
# xsc setup\commodetto.xsb
# xsc check.xsb
# xsc mod\config.xsb
# png2bmp OpenSans-Regular-24-alpha.bmp
# compressbmf OpenSans-Regular-24.bf4
# xsl helloworld-gui.xsa
Installing mod...........complete
```

johnwargo, 2021-07-18:
@andycarle did you learn anything from what I posted?

andycarle, 2021-07-28:
Hi @johnwargo! I'm swinging back around to GitHub duty for a bit. Let's see if we can get this closed out.

The issue in the most recent build output you've posted here is the result a rather different problem than we were looking at before. The `iot-product-dev-book\ch1-gettingstarted\helloworld-gui` example is _not_ a mod and thus should not be installed with `mcrun`. It is just a regular application and should be installed with `mcconfig` like any other.

We're talking a bit at the office today about whether there could be a more descriptive error message in that situation. But fundamentally, the error you got is true. `Mod failed: not enough keys` was displayed because the host was trying to process the large app (large because it brings in the graphics framework) as a mod, but exhausting keys while doing so.

Please give this a shot again with one of the examples that is intended to be run as a mod and see if you get different results. My guess is you'll still see issues (we haven't necessarily resolved whatever was going on up-thread, after all), but hopefully the new logging output will point us in the right direction.

andycarle, 2021-07-28:
Oh, also, on this error: `FC: cannot open D:\DEV\HARDWARE\MODDABLE\MODDABLE\BUILD\TMP\ESP32\MODDABLE_TWO\DEBUG\XSPROJ-ESP32\PARTITIONS.CSV - No such file or folder`

This one is annoying but harmless. The build does a file comparison between two files, one of which doesn't exist yet on a clean build. I'll look into suppressing this message, but you can safely ignore it.

johnwargo, 2021-07-29:
@andycarle ugh, sorry about that. I'll do it again with `helloworld` which is a module.

johnwargo, 2021-07-29:
hmmm, it worked. Worked the first time. Interesting. Just tested it again with the Colored Squares mod and it worked as well. I think you fixed it. 

I did encounter a problem (which is likely not related, but I thought I'd let you know) when I installed the host from the ch10 folder.  

[image]

closed 2021-10-04 by phoddie