Moddable-OpenSource/moddable #636: fxGrowChunks making very small allocations

warner, 2021-05-03:
I've been investigating performance problems in our `xsnap` usage of XS, where we're seeing GC happen more frequently than expected, and the allocations being made to `fxAllocateChunks` are surprisingly small (32-4000 bytes at a time), even though our `xsCreation` settings do `incrementalHeapCount = 256 * 1024`.

I'm suspicious of the following code in `fxGrowChunks`:

https://github.com/Moddable-OpenSource/moddable/blob/a2a38f4625371bbce3e9993a07c4e9bab3d2b8f7/xs/sources/xsMemory.c#L423-L434

It looks like the return value from `fxAddChunkSizes()` is being ignored, leaving `theSize` at its original value (e.g. 28 when allocating a new Function). On 17-Mar-2021, in 3070f249dec49d3bc62dc4f4d3dd6453444ba96a, this was changed:

https://github.com/Moddable-OpenSource/moddable/commit/3070f249dec49d3bc62dc4f4d3dd6453444ba96a#diff-5458ac58321f0e9dcd9ea60d1b7e2b87ea74e00cb70c5da666d1eed076c14258R380

I think the original intention was to increase `theSize` to `the->minimumChunksSize`, give or take some rounding and headers. In the new version, I think that increase got lost, resulting in small allocations, and therefore more frequent GC calls.

The fix is probably just:

```c
	theSize = fxAddChunkSizes(the, theSize, the->minimumChunksSize - modulo);
```

We're building against a derivative of commit 0b451c98f92c1637835f6a079bf5b6e41210d438 (aka tag `OS210406`), on Linux.

thanks!
 -Brian

phoddie, 2021-05-03:
Good find. That's our bug. We'll apply that to our next SDK update. Thank you for the help. Apologies for the trouble.

phoddie, 2021-07-28:
Closing this issue, as I believe it has been resolved.

closed 2021-07-28 by phoddie