Moddable-OpenSource/moddable #638: Error happens when building tools for Moddable SDK

asidai, 2021-05-05:
Hi,

I have a problem about "xsc commodettoBitmap.js" for building Moddable SDK tools on Windows
The error message happened in step5 and list below.

------------------------------------------------------------------------------------------------------------------------
C:\Users\USER\Projects\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\USER\Project
s\moddable\build XS_DIR=C:\Users\USER\Projects\moddable\xs /c /f tools.mak /s
win_xs.c
xsAll.c
xsAPI.c
xsArguments.c
xsArray.c
xsAtomics.c
xsBigInt.c
xsBoolean.c
xsCode.c
xsCommon.c
xsDataView.c
xsDate.c
xsDebug.c
xsError.c
xsFunction.c
xsGenerator.c
xsGlobal.c
xsJSON.c
xsLexical.c
xsMapSet.c
xsMarshall.c
xsMath.c
xsMemory.c
xsModule.c
xsNumber.c
xsObject.c
xsPlatforms.c
xsProfile.c
xsPromise.c
xsProperty.c
xsProxy.c
xsRegExp.c
xsRun.c
xsScope.c
xsScript.c
xsSourceMap.c
xsString.c
xsSymbol.c
xsSyntaxical.c
xsTree.c
xsType.c
xsdtoa.c
xsmc.c
xsre.c
xsa.c
xsc.c
xslBase.c

xsc commodettoBitmap.js
NMAKE : U1077: 'C:\Users\USER\Projects\moddable\build\bin\win\debug\xsc.EXE' : return code '0xc0000139'
Stop.
------------------------------------------------------------------------------------------------------------------------

I also used "x86 Native Tools Command Prompt for VS 2019" command line console, like it below

**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.9.4
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd %MODDABLE%\build\makefiles\win
C:\Users\USER\Projects\moddable\build\makefiles\win>build

Can you help it? 
Thanks a lot

andycarle, 2021-05-05:
Greetings! I'm sorry to hear that you're having trouble building the Moddable SDK tools. Let's see if we can sort this out.

I don't believe we've seen return code `0xc0000139` before. My understanding from some Googling is that that error comes up when trying to load a library that can't be found. But that's a bit odd; the Moddable SDK tools do not have any outside dependencies beyond what should have been installed with Visual Studio.

A couple of questions:

1) Is this a Windows 10 computer? Anything particularly unusual about your setup? We've seen some issues in the past when building on network drives or Google Drive mount points.

2) Is there anything unusual about your username (`USER` in your output above)? For instance, we've seen some issues with spaces and special characters in the username portion of paths.

And one thing to try:

I suggest re-running the [Visual Studio installer](https://www.visualstudio.com/downloads/) and having it re-install the Desktop development for C++" workload option. It is plausible that something is missing there. 

&nbsp;- Andy

asidai, 2021-05-06:
Hi Andy,

Thanks for your kindly reply in detail.

Yes, I built the Moddable SDK tools in Windows 7 environment. This is because my PC can't be upgraded to Windows 10 due to unsupportable internal display card driver, but I have been built it on the other PC installed Windows 10 and no error message happens. 

Thanks a lot.

andycarle, 2021-05-07:
I'm glad to hear that you got your build working on a Windows 10 PC. 

We've heard some reports of successful builds on Windows 7, but we do not actively support Windows 7 anymore since Microsoft has EOLed it. 

I'm going to go ahead and close this issue out. Please feel free to reopen it or to open a new issue if you have any additional questions.

&nbsp;- Andy

closed 2021-05-07 by andycarle