Moddable-OpenSource/moddable #639: Emulating a BLE HID Device using the Moddable SDK?

dbhaig, 2021-05-06:
The ESP-IDF has a BLE HID device demo. (esp-idf/examples/bluetooth/bluedroid/ble/ble_hid_device_demo)

Are there any plans to provide a similar Moddable example?
 
If not, do you have any advice on how to go about using the Moddable SDK to emulate a BLE HID device?

Thanks

andycarle, 2021-05-06:
I've done this a couple of times for personal projects. It's pretty straightforward to implement, depending on how complicated of a device you're wanting to emulate. I've done keyboard, mouse, and generic gamepad for various projects.

What type of HID device are you wanting to do? I'd be happy to share some code if what I have is relevant.

dbhaig, 2021-05-06:
Initially,  would like to emulate a composite mouse and keyboard but I would also be interested the gamepad solution as well. Any code that you are willing to share would be gratefully received.

Thanks

andycarle, 2021-05-07:
I've uploaded a couple of BLE HID examples to a [new repository here](https://github.com/andycarle/moddable-ble-hid-examples). For now I've included a very basic touchpad and keyboard.

A couple of caveats on these:

1. These are decidedly not fully-worked examples. Nor have they been polished up to the extent that we normally would before including them in the Moddable SDK examples. These are just quick demos I put together for a meetup presentation.
2. The Moddable SDK supports two different BLE stacks on ESP32: Bluedroid and NimBLE. We use NimBLE by default. However, in quick testing tonight I could only get these examples to work reliably using Bluedroid. To build with Bluedroid, you need to set an environment variable `ESP32_BLUEDROID` to `1` before building. When I get some time next week I'll see if I can figure out why these examples aren't playing nicely with NimBLE.

You should be able to run these examples on a Moddable Two and pair them with a PC (I've tested on both Windows and macOS) to emulate a touchpad and keyboard, respectively. I have not put in the pieces to make these reconnect gracefully if power-cycled, but that should be pretty straightforward.

The magic happens in [`mouseService.js`](https://github.com/andycarle/moddable-ble-hid-examples/blob/main/touchpad/mouseService.js) and [`keyboardService.js`](https://github.com/andycarle/moddable-ble-hid-examples/blob/main/keyboard/keyboardService.js). Those show how to populate and send the HID reports for the emulated devices.

The HID device descriptor goes into `hid.json` ([touch pad](https://github.com/andycarle/moddable-ble-hid-examples/blob/main/touchpad/bleservices/hid.json) and [keyboard](https://github.com/andycarle/moddable-ble-hid-examples/blob/main/keyboard/bleservices/hid.json)) in the `report_map` property. If you have experience with USB or BLE HID, that should look familiar, along with everything else in `hid.json`.

Modifying those is a matter of generating the right device descriptor and propagating your changes through the report characteristic. It's finicky, but no more so than HID anywhere else. :)

Good luck! I'll try to get the basic gamepad example posted soon. Let me know if I can offer any further assistance and I look forward to hearing what you end up building!

&nbsp;- Andy

dbhaig, 2021-05-07:
Thank you very much for this! I will let you know how I make out.

- Don

andycarle, 2021-07-28:
I'm going to convert this issue into a [discussion](https://github.com/Moddable-OpenSource/moddable/discussions) to clear it out of the issue queue. I would be delighted to continue the conversation in the discussion topic if anyone gives this a shot.

closed 2021-07-28 by andycarle