Moddable-OpenSource/moddable #640: Entering gdb stub now

mauroForlimpopoli, 2021-05-09:
Compiling my project I get this error: 
////////////
C:\moddable\examples\piu\k7>mcconfig -d -m -r 90 -p esp32/moddable_two
# bles2gatt bleservices
# xsc main.xsb
# xsl modules
### 1306 instances, 1865 keys, 95 colors, 0 holes
### warning: "keyboard": Keyboard() f() KeyboardBehavior.prototype.onCreate() ShiftTexture() it: not frozen
### warning: "keyboard": Keyboard() f() KeyboardBehavior.prototype.onCreate() BackspaceTexture() it: not frozen
### warning: "keyboard": Keyboard() f() KeyboardBehavior.prototype.onCreate() NumKeysTexture() it: not frozen
# cc mc.xs.o (slots in flash)
# ld xs_esp32.bin
        1 file copiati.
Executing action: all (aliases: build)
Running ninja in directory c:\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32
Executing "ninja all"...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[3/3] Generating binary image from built executable
esptool.py v3.0
Generated C:/moddable/build/tmp/esp32/moddable_two/debug/idf-esp32/xs_esp32.bin
Executing action: flash
Choosing default port b'COM9' (use '-p PORT' option to set a specific serial port)
Running ninja in directory c:\moddable\build\tmp\esp32\moddable_two\debug\idf-esp32
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\esp32\esp-idf\components\esptoo... C:/esp32/esp-idf/components/esptool_py/run_esptool.cmake"
esptool.py --chip esp32 -p COM9 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0x1000 bootloader/bootloader.bin 0x10000 xs_esp32.bin
esptool.py v3.0
Serial port COM9
Connecting.....
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: fc:f5:c4:61:a5:b8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 3072 bytes to 157...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (157 compressed) at 0x00008000 in 0.0 seconds...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x0000d000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.0 seconds...
Hash of data verified.
Compressed 17728 bytes to 12034...
Writing at 0x00001000... (100 %)
Wrote 17728 bytes (12034 compressed) at 0x00001000 in 0.2 seconds (effective 756.6 kbit/s)...
Hash of data verified.
Compressed 1673376 bytes to 989368...
Writing at 0x00010000... (1 %)
Writing at 0x00014000... (3 %)
Writing at 0x00018000... (4 %)
Writing at 0x0001c000... (6 %)
Writing at 0x00020000... (8 %)
Writing at 0x00024000... (9 %)
Writing at 0x00028000... (11 %)
Writing at 0x0002c000... (13 %)
Writing at 0x00030000... (14 %)
Writing at 0x00034000... (16 %)
Writing at 0x00038000... (18 %)
Writing at 0x0003c000... (19 %)
Writing at 0x00040000... (21 %)
Writing at 0x00044000... (22 %)
Writing at 0x00048000... (24 %)
Writing at 0x0004c000... (26 %)
Writing at 0x00050000... (27 %)
Writing at 0x00054000... (29 %)
Writing at 0x00058000... (31 %)
Writing at 0x0005c000... (32 %)
Writing at 0x00060000... (34 %)
Writing at 0x00064000... (36 %)
Writing at 0x00068000... (37 %)
Writing at 0x0006c000... (39 %)
Writing at 0x00070000... (40 %)
Writing at 0x00074000... (42 %)
Writing at 0x00078000... (44 %)
Writing at 0x0007c000... (45 %)
Writing at 0x00080000... (47 %)
Writing at 0x00084000... (49 %)
Writing at 0x00088000... (50 %)
Writing at 0x0008c000... (52 %)
Writing at 0x00090000... (54 %)
Writing at 0x00094000... (55 %)
Writing at 0x00098000... (57 %)
Writing at 0x0009c000... (59 %)
Writing at 0x000a0000... (60 %)
Writing at 0x000a4000... (62 %)
Writing at 0x000a8000... (63 %)
Writing at 0x000ac000... (65 %)
Writing at 0x000b0000... (67 %)
Writing at 0x000b4000... (68 %)
Writing at 0x000b8000... (70 %)
Writing at 0x000bc000... (72 %)
Writing at 0x000c0000... (73 %)
Writing at 0x000c4000... (75 %)
Writing at 0x000c8000... (77 %)
Writing at 0x000cc000... (78 %)
Writing at 0x000d0000... (80 %)
Writing at 0x000d4000... (81 %)
Writing at 0x000d8000... (83 %)
Writing at 0x000dc000... (85 %)
Writing at 0x000e0000... (86 %)
Writing at 0x000e4000... (88 %)
Writing at 0x000e8000... (90 %)
Writing at 0x000ec000... (91 %)
Writing at 0x000f0000... (93 %)
Writing at 0x000f4000... (95 %)
Writing at 0x000f8000... (96 %)
Writing at 0x000fc000... (98 %)
Writing at 0x00100000... (100 %)
Wrote 1673376 bytes (989368 compressed) at 0x00010000 in 16.4 seconds (effective 818.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
        1 file copiati.
        1 file copiati.
        1 file copiati.
        1 file copiati.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x40091317  PS      : 0x00060b33  A0      : 0x800945e2  A1      : 0x3ffb4520
A2      : 0x800e690d  A3      : 0x3ffb4660  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3ffbcf98  A8      : 0x0000abab  A9      : 0x0000abab
A10     : 0x0000cdcd  A11     : 0xb33fffff  A12     : 0x00060b20  A13     : 0x00060b23
A14     : 0x3aff090c  A15     : 0x3ffb4430  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x800e690d  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000
Backtrace:0x40091314:0x3ffb4520 0x400945df:0x3ffb4540 0x40094614:0x3ffb4560 0x40081325:0x3ffb4590 0x40081354:0x3ffb45c0 0x4009791e:0x3ffb45e0 0x4009793c:0x3ffb4600 0x4016ef12:0x3ffb4620 0x4012d0a5:0x3ffb4640 0x400fd06b:0x3ffb4660 0x400eb3c1:0x3ffb46c0 0x400d4aa1:0x3ffb4730 0x400fdd25:0x3ffb4750 0x400fe161:0x3ffb47e0 0x400d28ee:0x3ffb4800
ELF file SHA256: 1545c2039bc6cae1
Entering gdb stub now.
//////////
1) If can help, if I put a "debugger" on the code main.js, and then I click "play", the error is not present, and the debugger run correctly.

phoddie, 2021-05-11:
There's not enough information here to do more than make wild guesses.

1. Before you install and run your project with `mcconfig` clear all breakpoints in xsbug by pressing the trash icon in the Breakpoints pane. Then, run as usual.
2. If (1) doesn't help, try running a release build. In a release build, you should see a native stack trace in addition to the register dump. That may provide a clue about where it is failing. To run a release build remove `-d` from your usual `mcconfig` command

wilberforce, 2021-05-12:
```
/////  This won't comment the block...
```

```
/*
This won't compile
*/
```

phoddie, 2021-05-12:
In general, we've seen that using BLE on ESP32 with NimBLE uses about 32 KB of RAM. That's consistent with your report above of memory going from about 45 KB to about 11 KB.

phoddie, 2021-06-17:
Closing as this appears to be resolved.

closed 2021-06-17 by phoddie